espressif/esp-idf #8959: When installing with encrypted merge bin, format is required to the web server partition

sengulhamza, 2022-05-16:
Hi,

We use encrypted in the project. When I convert the build artifacts to merge.bin and load them with esptool.py, the 'www' flash partition cannot be read without formatting.
But if I flash the script we wrote in ./make_flash.sh, spifs does not want to format while init and my web files can be read properly by the web server.

All this happens when I use encrypted.

Demo Project for that issue-> [restful_server.zip](https://github.com/espressif/esp-idf/files/8700304/restful_server.zip)

partition table
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data,   nvs,        0x11000,    16K     ,
nvs_key,    data,   nvs_keys,   ,           4K      , encrypted,
phy_init,   data,   phy,        ,           4K      ,
factory,    app,    factory,    0x20000,    2M      , encrypted,
sem,        data,   spiffs,     ,           512K    ,
www,        data,   spiffs,     ,           1M      ,
log,        data,   spiffs,     ,           2M      ,

```

sdkconfig.defaults

```
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="8MB"
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_example.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_example.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x10000
CONFIG_SPIFFS_MAX_PARTITIONS=10
CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM=y
CONFIG_UART_ISR_IN_IRAM=y
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=4096
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FMB_TIMER_ISR_IN_IRAM=y
CONFIG_FMB_PORT_TASK_AFFINITY_CPU1=y
CONFIG_FMB_PORT_TASK_AFFINITY=0x1
CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=500
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
CONFIG_LWIP_TCPIP_TASK_AFFINITY=0x0
CONFIG_LWIP_MAX_SOCKETS=16
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
CONFIG_FMB_COMM_MODE_RTU_EN=y
CONFIG_FMB_COMM_MODE_ASCII_EN=n
CONFIG_FMB_COMM_MODE_TCP_EN=n

#
# Flash encryption 
#
CONFIG_SECURE_FLASH_ENC_ENABLED=y

# less secure option (uart enabled, jtag disabled)
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
CONFIG_SECURE_BOOT_ALLOW_JTAG=n
CONFIG_SECURE_FLASH_UART_BOOTLOADER_ALLOW_ENC=y

# full secure option (not used)
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y

```

./make_flash.sh
```

#!/bin/bash

. $HOME/esp/esp-idf/export.sh

declare -r dec_dir="build_bins"
declare -r proj_bin="restful_server.bin"

if rm -r ${dec_dir}
then
echo "-- build_bins dir removed!"
fi
if mkdir ${dec_dir} 
then
echo "-- build_bins dir created."
else 
exit 0
fi

if ! cp build/www.bin ${dec_dir} ;
then 
exit 0 
fi

if ! cp build/bootloader/bootloader.bin ${dec_dir};
then 
exit 0 
fi

if ! cp build/${proj_bin} ${dec_dir};
then 
exit 0 
fi

if ! cp build/partition_table/partition-table.bin ${dec_dir};
then 
exit 0 
fi

echo "-- All .bin files copied to ${dec_dir} dir"

ls -c ${dec_dir}

if [ "$1" = "encrypt" ]; then
    echo "-- Encrypt flash enabled!"
    if esptool.py --chip esp32 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x2a0000 ${dec_dir}/www.bin --encrypt-files 0x1000 ${dec_dir}/bootloader.bin 0x10000 ${dec_dir}/partition-table.bin 0x20000 ${dec_dir}/${proj_bin}; 
    then
    echo "Program flash successed."
    else 
    echo "Program flash FAILED!"
    exit 0
    fi
elif [ "$1" = "" ]; then
    echo "-- Encrypt flash disabled!"
    if esptool.py --chip esp32 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 ${dec_dir}/bootloader.bin 0x10000 ${dec_dir}/partition-table.bin 0x20000 ${dec_dir}/${proj_bin} 0x2a0000 ${dec_dir}/www.bin ;
    then
    echo "Program flash successed."
    else 
    echo "Program flash FAILED!"
    exit 0
    fi
f
```i

I hope you can help. Thanks

sengulhamza, 2022-05-17:
Thank you @igrr for your quick reply.
On the CI side, I create merged.bin with prod build and release.
But since I can't use merged.bin anymore, I need to make changes in CI yaml as well. Do you have an example or documentation for this?
Thank you.

igrr, 2022-05-17:
I think you can still create the merged binary, just don't include the `www` partition when merging. In CI you can store two binaries as artifacts, the merged binary and the `www` partition binary. Then in the production environment, flash these 2 binaries as separate files, as you are doing this in your make_flash script.

sengulhamza, 2022-05-17:
Thank you @igrr for your reply :)
I'm closing this issue.

closed 2022-05-17 by sengulhamza