espressif/esp-idf #8960: mbedTLS example build fails with TLS1.3 only

ddomnik, 2022-05-16:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1
- IDF version // Master Branch - ad747b237a (HEAD -> master, origin/master, origin/HEAD) Merge branch 'toos/add_s3_ut_based_on_quad_octal_psram' into 'master'
- Build System:         CMake|idf.py
- Operating System:     Windows
- Eclipse IDE

## Problem Description
https_mbedtls build fails when only "Support TLS 1.3 protocol" (CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y) is selected and not 1.2.
When both, TLS 1.2 and 1.3 is selected, the example uses 1.2 but 1.3 should be preferred.

//Detailed problem description goes here.

The first error `error: #error "MBEDTLS_SSL_TLS_C defined, but no protocols are active"` should not appear when i take a closer look at check_config.h and when my sdkconfig says `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y`:

```
 #if defined(MBEDTLS_SSL_TLS_C) && \
!( defined(MBEDTLS_SSL_PROTO_TLS1_2) || defined(MBEDTLS_SSL_PROTO_TLS1_3) )
#error "MBEDTLS_SSL_TLS_C defined, but no protocols are active"
#endif 
```

The migration guide found here (https://github.com/Mbed-TLS/mbedtls/blob/development/docs/3.0-migration-guide.md) states, that `check_config.h` should not be included if a custom configuration file is used. What exactly is meant by a custom configuration file? I assume the example does not use of such a file?

## Debug Logs

```
Building in: C:\Users\ddufner\Documents\esp\workspace_test1\https_mbedtls\build
cmake --build . -- -v
[0/1] C:\Espressif\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SC:\Users\ddufner\Documents\esp\workspace_test1\https_mbedtls -BC:\Users\ddufner\Documents\esp\workspace_test1\https_mbedtls\build
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Checking Python dependencies...
Python requirements are satisfied.
Skipping the download of C:\Espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Constraint file: C:\Espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe
-- Project sdkconfig file C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/sdkconfig
-- App "https_mbedtls" version: 1
-- Adding linker script C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mdns mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf/components/app_trace C:/Espressif/frameworks/esp-idf/components/app_update C:/Espressif/frameworks/esp-idf/components/asio C:/Espressif/frameworks/esp-idf/components/bootloader C:/Espressif/frameworks/esp-idf/components/bootloader_support C:/Espressif/frameworks/esp-idf/components/bt C:/Espressif/frameworks/esp-idf/components/cmock C:/Espressif/frameworks/esp-idf/components/console C:/Espressif/frameworks/esp-idf/components/cxx C:/Espressif/frameworks/esp-idf/components/driver C:/Espressif/frameworks/esp-idf/components/efuse C:/Espressif/frameworks/esp-idf/components/esp-tls C:/Espressif/frameworks/esp-idf/components/esp32s3 C:/Espressif/frameworks/esp-idf/components/esp_adc_cal C:/Espressif/frameworks/esp-idf/components/esp_common C:/Espressif/frameworks/esp-idf/components/esp_eth C:/Espressif/frameworks/esp-idf/components/esp_event C:/Espressif/frameworks/esp-idf/components/esp_gdbstub C:/Espressif/frameworks/esp-idf/components/esp_hid C:/Espressif/frameworks/esp-idf/components/esp_http_client C:/Espressif/frameworks/esp-idf/components/esp_http_server C:/Espressif/frameworks/esp-idf/components/esp_https_ota C:/Espressif/frameworks/esp-idf/components/esp_https_server C:/Espressif/frameworks/esp-idf/components/esp_hw_support C:/Espressif/frameworks/esp-idf/components/esp_lcd C:/Espressif/frameworks/esp-idf/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf/components/esp_netif C:/Espressif/frameworks/esp-idf/components/esp_phy C:/Espressif/frameworks/esp-idf/components/esp_pm C:/Espressif/frameworks/esp-idf/components/esp_ringbuf C:/Espressif/frameworks/esp-idf/components/esp_rom C:/Espressif/frameworks/esp-idf/components/esp_serial_slave_link C:/Espressif/frameworks/esp-idf/components/esp_system C:/Espressif/frameworks/esp-idf/components/esp_timer C:/Espressif/frameworks/esp-idf/components/esp_wifi C:/Espressif/frameworks/esp-idf/components/espcoredump C:/Espressif/frameworks/esp-idf/components/esptool_py C:/Espressif/frameworks/esp-idf/components/fatfs C:/Espressif/frameworks/esp-idf/components/freertos C:/Espressif/frameworks/esp-idf/components/hal C:/Espressif/frameworks/esp-idf/components/heap C:/Espressif/frameworks/esp-idf/components/http_parser C:/Espressif/frameworks/esp-idf/components/idf_test C:/Espressif/frameworks/esp-idf/components/ieee802154 C:/Espressif/frameworks/esp-idf/components/json C:/Espressif/frameworks/esp-idf/components/log C:/Espressif/frameworks/esp-idf/components/lwip C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/main C:/Espressif/frameworks/esp-idf/components/mbedtls C:/Espressif/frameworks/esp-idf/components/mdns C:/Espressif/frameworks/esp-idf/components/mqtt C:/Espressif/frameworks/esp-idf/components/newlib C:/Espressif/frameworks/esp-idf/components/nvs_flash C:/Espressif/frameworks/esp-idf/components/openthread C:/Espressif/frameworks/esp-idf/components/partition_table C:/Espressif/frameworks/esp-idf/components/perfmon C:/Espressif/frameworks/esp-idf/components/protobuf-c C:/Espressif/frameworks/esp-idf/examples/common_components/protocol_examples_common C:/Espressif/frameworks/esp-idf/components/protocomm C:/Espressif/frameworks/esp-idf/components/pthread C:/Espressif/frameworks/esp-idf/components/sdmmc C:/Espressif/frameworks/esp-idf/components/soc C:/Espressif/frameworks/esp-idf/components/spi_flash C:/Espressif/frameworks/esp-idf/components/spiffs C:/Espressif/frameworks/esp-idf/components/tcp_transport C:/Espressif/frameworks/esp-idf/components/tinyusb C:/Espressif/frameworks/esp-idf/components/ulp C:/Espressif/frameworks/esp-idf/components/unity C:/Espressif/frameworks/esp-idf/components/usb C:/Espressif/frameworks/esp-idf/components/vfs C:/Espressif/frameworks/esp-idf/components/wear_levelling C:/Espressif/frameworks/esp-idf/components/wifi_provisioning C:/Espressif/frameworks/esp-idf/components/wpa_supplicant C:/Espressif/frameworks/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/build
[1/846] ccache C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/Espressif/framewor [... cut ...]
C:\Espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf\tools\idf_size.py C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/build/https_mbedtls.map
C:\Espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf/components/partition_table/gen_esp32part.py C:/Users/ddufner/Documents/esp/workspace_test1/https_mbedtls/build/partition_table/partition-table.bin
Total time taken to build the project: 25.254 ms
```

## Other items if possible

[sdkconfig-copy.txt](https://github.com/espressif/esp-idf/files/8700740/sdkconfig-copy.txt)

laukik-hase, 2022-05-17:
Hello, @ddomnik!
Thank you for the issue report.

> https_mbedtls build fails when only "Support TLS 1.3 protocol" (CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y) is selected and not 1.2.

As per `Mbed TLS v3.1.0`, TLS 1.3 cannot be enabled in the build (`CONFIG_MBEDTLS_SSL_PROTO_TLS1_3` option) without TLS 1.2 (`CONFIG_MBEDTLS_SSL_PROTO_TLS1_2` option). The build error in your case arises from the check [here](https://github.com/espressif/mbedtls/blob/9bb5effc3298265f829878825d9bd38478e67514/include/mbedtls/check_config.h#L647).

> When both, TLS 1.2 and 1.3 is selected, the example uses 1.2 but 1.3 should be preferred.

Could you try out this patch - [mbedtls_tls1_3.zip](https://github.com/espressif/esp-idf/files/8706647/mbedtls_tls1_3.zip) and let us know if it works for you?

### Notes
1. The ESP-IDF `master` branch currently uses `Mbed TLS v3.1.0`, the `check_config.h` file you looked at seems to be from the current [`development`](https://github.com/Mbed-TLS/mbedtls) branch of `Mbed-TLS/mbedtls`.
2. The `Mbed TLS` support for TLS 1.3 is still developing - see [here](https://github.com/ARMmbed/mbedtls/blob/development/docs/architecture/tls13-support.md).

ddomnik, 2022-05-17:
@laukik-hase thanks for the reply. My mistake for checking the wrong line and file version regarding the check_config.h
After applying the patch and enabling TLS 1.2 & 1.3 the example ends with the SSL Handshake error **0x7780**

```
I (98327) example: Starting again!
I (98327) example: Connecting to www.howsmyssl.com:443...
I (98557) example: Connected.
I (98567) example: Performing the SSL/TLS handshake...
E (98977) example: mbedtls_ssl_handshake returned -0x7780
E (98977) example: Last error was: -0x7780 - SSL - A fatal alert message was received from our peer
```

I also changed the TLS Protocol Role to Client and tested again, but resulted in the same error.

laukik-hase, 2022-05-17:
Hello, @ddomnik!

The support for TLS 1.3 in Mbed TLS is still developing - see [here](https://github.com/ARMmbed/mbedtls/blob/development/docs/architecture/tls13-support.md). 

We are exploring more on TLS 1.3 support, and it has been difficult due to very few test endpoints being able to connect to `mbedtls` TLS 1.3 clients. An example endpoint we found was `google.com` (which gives the `HTTP/1.1 404 Not Found` response). The next release of Mbed TLS will overcome the current shortcomings.

closed 2022-05-24 by espressif-bot

AxelLin, 2024-11-20:
@laukik-hase 

With below settings: (build with v5.2.3-431-gda635ecd9d)
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=n
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
I got:
```

/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:240:5: error: implicit declaration of function 'MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG' [-Werror=implicit-function-declaration]
  240 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_ECDSA, MBEDTLS_SSL_HASH_SHA512 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:240:5: error: initializer element is not constant
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:240:5: note: (near initialization for 'tls_sig_algs_for_eap[0]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:241:5: error: initializer element is not constant
  241 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_ECDSA, MBEDTLS_SSL_HASH_SHA384 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:241:5: note: (near initialization for 'tls_sig_algs_for_eap[1]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:244:5: error: initializer element is not constant
  244 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_RSA, MBEDTLS_SSL_HASH_SHA512 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:244:5: note: (near initialization for 'tls_sig_algs_for_eap[2]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:245:5: error: initializer element is not constant
  245 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_RSA, MBEDTLS_SSL_HASH_SHA384 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:245:5: note: (near initialization for 'tls_sig_algs_for_eap[3]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:250:5: error: initializer element is not constant
  250 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_ECDSA, MBEDTLS_SSL_HASH_SHA256 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:250:5: note: (near initialization for 'tls_sig_algs_for_eap[4]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:251:5: error: initializer element is not constant
  251 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_ECDSA, MBEDTLS_SSL_HASH_SHA224 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:251:5: note: (near initialization for 'tls_sig_algs_for_eap[5]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:254:5: error: initializer element is not constant
  254 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_RSA, MBEDTLS_SSL_HASH_SHA256 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:254:5: note: (near initialization for 'tls_sig_algs_for_eap[6]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:255:5: error: initializer element is not constant
  255 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_RSA, MBEDTLS_SSL_HASH_SHA224 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:255:5: note: (near initialization for 'tls_sig_algs_for_eap[7]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:260:5: error: initializer element is not constant
  260 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_ECDSA, MBEDTLS_SSL_HASH_SHA1 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:260:5: note: (near initialization for 'tls_sig_algs_for_eap[8]')
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:263:5: error: initializer element is not constant
  263 |     MBEDTLS_SSL_TLS12_SIG_AND_HASH_ALG( MBEDTLS_SSL_SIG_RSA, MBEDTLS_SSL_HASH_SHA1 ),
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c:263:5: note: (near initialization for 'tls_sig_algs_for_eap[9]')
cc1: all warnings being treated as errors

```