espressif/esp-idf #8961: ESP32 PPPOS support EC20

smartmos, 2022-05-16:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

david-cermak, 2022-05-17:
Hi @smartmos 

Could you please raise this feature request in [esp-protocols](https://github.com/espressif/esp-protocols/issues) repository? The PPPoS example was removed from IDF on latest branches; while it is still supported, new features would only be added to `esp_modem` component.

Have you tried to run the PPPoS example with Quectel's EC20 already? After checking the docs, it seems to me that it should still work and provide the basic connection functionally that the IDF example demonstrates. It supports traditional set of AT commands and PPP mode, as well. Which interface you're using? It supports UART, but also USB. Please note that the [USB-DTE](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console#usb-dte-support) support has been recently added to `esp_modem`

Alvin1Zhang, 2022-08-02:
Thanks for reporting, feel free to reopen with more updates.

closed 2023-03-27 by david-cermak