espressif/esp-idf #8963: ESP32-C3 Deep Sleep doesn't work with enabled DFS light sleep

ermacv, 2022-05-16:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): v4.4.1-160-g3806b240cf
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:        USB

## Problem Description

When DFS auto light sleep is enabled deep sleep doesn't work. There are actually few problems please see the code gists below.

### Expected Behavior
Deep Sleep should work with DFS auto light sleep.
### Actual Behavior
Deep Sleep doesn't work with DFS auto light sleep.

### Steps to reproduce

Use the code below.

### Code to reproduce this issue
Here MCU will never return from a deep sleep with enabled 1 second timer value:
https://gist.github.com/ermacv/6946e0165006a1c640064f3901341c4b
Here MCU will perform an infinite sleep loop without any set wakeup source:
https://gist.github.com/ermacv/a746dce0c788e4c332af953a4dd62b19
For the second example if you delete `vTaskDelay(pdMS_TO_TICKS(100));` line then MCU enters the sleep mode and there no wakeup loop.

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ermacv, 2022-05-16:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8702201/sdkconfig.zip)

ermacv, 2022-05-16:
It seems that using the code like this solves the issue:
```c
  esp_pm_lock_handle_t out_handle;
  esp_pm_lock_create(ESP_PM_NO_LIGHT_SLEEP, 0, "sleep_lock", &out_handle);
  esp_pm_lock_acquire(out_handle);
  esp_deep_sleep(1 * 1000000);
```

But I could be wrong so please check it.

ermacv, 2022-05-17:
I also tested the following and it doesn't work:
```c
vTaskDelay(pdMS_TO_TICKS(100));
esp_sleep_enable_timer_wakeup(1 * 1000000);
esp_pm_lock_handle_t out_handle;
esp_pm_lock_create(ESP_PM_NO_LIGHT_SLEEP, 0, "btLS", &out_handle);
esp_pm_lock_acquire(out_handle);
esp_deep_sleep_start();
```

So it seems that both calls to deep sleep API (`esp_sleep_enable_timer_wakeup` and `esp_deep_sleep_start`) should be protected by PM lock. I guess light sleep and deep sleep both use the same API/RTC registers/etc. internally.

ermacv, 2022-05-19:
Even more. I want to have auto light sleep for RTOS and deep sleep with only wakeup pin source enabled I should have the following code:
```c
esp_pm_lock_handle_t out_handle;
esp_pm_lock_create(ESP_PM_NO_LIGHT_SLEEP, 0, "btLS", &out_handle);
esp_pm_lock_acquire(out_handle);
esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_TIMER);
esp_deep_sleep_start();
```
So `esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_TIMER);` extra line that resets the possible enabled values by auto light sleep.

xiaojie76, 2022-05-20:
Hi @ermacv, In your first case, the phenomenon on your side is that it will enter deep sleep mode, and will not wake up after 1s, then enter deep sleep mode again after using pm lock, it can wake up normally after 1s, right?

xiaojie76, 2022-05-20:
Hi @ermacv  A few notes:
1. Whether it is light sleep or deep sleep, the sleep configuration is shared, so the wake up source will not be distinguished separately. The main differences between the two modes are:
1) The power down domains  are different.
2) After light sleep wakes up, it will continue to execute, and after deep sleep wakes up, it will be re-executed from the boot phase.

2. After configuring `esp_pm_configure `to enable light sleep, rtos will execute `vApplicationSleep`(pm_impl.c) when it schedule to idle task.
```
IDLE_TASK
    vApplicationSleep()
        if(!should_skip_light_sleep(core_id))
            esp_sleep_enable_timer_wakeup(...);
            esp_light_sleep_start();
```
For your second case,` vTaskDelay(pdMS_TO_TICKS(100))`-> idle task-> `vApplicationSleep`-> `esp_sleep_enable_timer_wakeup(<100ms)`->` esp_light_sleep_start-`> wake up from light sleep ->schedule  main task-> `esp_deep_sleep_start`-> enter deep sleep-> wake up from deep sleep (<100ms), so there is the infinite loop wakeup you said.

3. After using pm lock`(esp_pm_lock_create(ESP_PM_NO_LIGHT_SLEEP...)`, `should_skip_light_sleep` is always true in the previous process, and the rtc timer will not be set to wake up.  So you can this pm lock to disable automatic light sleep.

ermacv, 2022-05-20:
Hi @xiaojie76 !
For the first case: yes this is the issue.
For the second case: yes the sequence is that.

So is it OK to use a sleep lock and `esp_sleep_disable_wakeup_source` function call explicitly by application to be able to use deep sleep properly if I need GPIO as the only wake up source for deep sleep?

xiaojie76, 2022-05-23:
@ermacv  I didn't reproduce your first issue, in my side it works fine to wake up from deep sleep. Even if light_sleep_enable = true, it should not have any effect on esp_deep_sleep, which is a simple timer wake up.
Unless there is such a weird phenomenon in your environment:
Scheduled to idle task before entering deep sleep, and set the wakeup time to a large value in vApplicationSleep. You can check if this phenomenon exists.

For the second case, I think your solution is ok .

ermacv, 2022-05-23:
@xiaojie76 Just a side question. Will it possible to add information about auto light sleep and deep sleep using together into the documentation that there could be found such issues and some "tweaks" are possibly needed? For me it was not so obvious that I need to call an extra sleep/PM API functions to make it work properly.

xiaojie76, 2022-05-23:
@ermacv You can get a more detailed information here: [power_management](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.0/api-reference/system/power_management.html) and [sleep_modes](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.0/api-reference/system/sleep_modes.html). If there is anything unclear, please feel free to ask.

ermacv, 2022-07-29:
The issue is resolved.

closed 2022-07-29 by ermacv

Alvin1Zhang, 2022-08-01:
Thanks for sharing the updates.