espressif/esp-idf #8967: Panic if setting CONFIG_WPA_11KV_SUPPORT=y

AxelLin, 2022-05-17:
## Environment

- Module or chip used: ESP32-C3-DevKitM-1
- IDF version: v5.0-dev-2959-g31b7694551
- Build System: idf.py
- Compiler version: riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

it below panic if setting CONFIG_WPA_11KV_SUPPORT=y.
Note, my test access point does not support 802.11kv.

I (1036) pp: pp rom version: 9387209
I (1037) net80211: net80211 rom version: 9387209
I (1038) wifi:wifi driver task: 3fca77c4, prio:23, stack:6656, core=0
I (1045) wifi:wifi firmware version: cad211c
I (1046) wifi:wifi certification version: v7.0
I (1046) wifi:config NVS flash: enabled
I (1046) wifi:config nano formating: disabled
I (1046) wifi:Init data frame dynamic rx buffer num: 32
I (1046) wifi:Init management frame dynamic rx buffer num: 32
I (1047) wifi:Init management short buffer num: 32
I (1047) wifi:Init dynamic tx buffer num: 32
I (1047) wifi:Init static tx FG buffer num: 2
I (1047) wifi:Init static rx buffer size: 1600
I (1048) wifi:Init static rx buffer num: 10
I (1048) wifi:Init dynamic rx buffer num: 32
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
Stack dump detected
Core  0 register dump:
MEPC    : 0x420333d2  RA      : 0x42031876  SP      : 0x3fc9ab30  GP      : 0x3fc90c00
0x420333d2: eloop_cancel_timeout at /home/axel/esp/esp-idf/components/wpa_supplicant/port/eloop.c:136

0x42031876: wpas_rrm_reset at /home/axel/esp/esp-idf/components/wpa_supplicant/src/common/rrm.c:53

TP      : 0x3fc69ff0  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000000
S0/FP   : 0x3fc978f8  S1      : 0x3fc979a0  A0      : 0x00000000  A1      : 0x3fc979a0
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3fc97274  A5      : 0x3fc95000
A6      : 0x00000000  A7      : 0x43500000  S2      : 0x3fcac994  S3      : 0x00000000
S4      : 0x42030ff8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
0x42030ff8: wpas_rrm_neighbor_rep_timeout_handler at /home/axel/esp/esp-idf/components/wpa_supplicant/src/common/rrm.c:29

S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x40140000  T4      : 0x00000000  T5      : 0x401c0000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Backtrace:

0x420333d2 in eloop_cancel_timeout (handler=handler@entry=0x42030ff8 <wpas_rrm_neighbor_rep_timeout_handler>, eloop_data=eloop_data@entry=0x3fc979a0 <g_wpa_supp+168>, user_data=user_data@entry=0x0) at /home/axel/esp/esp-idf/components/wpa_supplicant/port/eloop.c:136
136             dl_list_for_each_safe(timeout, prev, &eloop.timeout,
#0  0x420333d2 in eloop_cancel_timeout (handler=handler@entry=0x42030ff8 <wpas_rrm_neighbor_rep_timeout_handler>, eloop_data=eloop_data@entry=0x3fc979a0 <g_wpa_supp+168>, user_data=user_data@entry=0x0) at /home/axel/esp/esp-idf/components/wpa_supplicant/port/eloop.c:136
#1  0x42031876 in wpas_rrm_reset (wpa_s=0x3fc978f8 <g_wpa_supp>) at /home/axel/esp/esp-idf/components/wpa_supplicant/src/common/rrm.c:51
#2  0x4202dfec in esp_supplicant_common_init (wpa_cb=wpa_cb@entry=0x3fcac994) at /home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_common.c:371
#3  0x4209e902 in esp_supplicant_init () at /home/axel/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:263
#4  0x4203fc70 in esp_wifi_init (config=config@entry=0x3fc9ad28) at /home/axel/esp/esp-idf/components/esp_wifi/src/wifi_init.c:245
#5  0x42005646 in start_wifi () at /home/axel/esp/esp-idf-dev/apps/myproj/main/main.c:399
#6  0x42006ca8 in app_main () at /home/axel/esp/esp-idf-dev/apps/myproj/main/main.c:907
#7  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 5942b3671ee5dd72

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038065e
0x4038065e: esp_restart_noos_dig at /home/axel/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)

### Expected Behavior
It should not hit panic.

### Actual Behavior

### Steps to reproduce

My application seems hit this issue soon after calling esp_wifi_init().
ps. it works fine if CONFIG_WPA_11KV_SUPPORT=n.

kapilkedawat, 2022-05-17:
Fix for this is already merged internally and should be available soon on github.

AxelLin, 2022-05-20:
Fixed by ee33c6ae05d55acbc462afd8595b432d9002e62d.

BTW, mixing a bug fix in a commit adding new APIs is not good.
Such bug fix should be in a commit for bug fix only.

closed 2022-05-20 by AxelLin