espressif/esp-idf #8969: Canno't run idf.py after updating cmake

EdMlt, 2022-05-17:
I ran _brew update_ and now it updated my cmake.
Now when I'm trying to run _idf.py menuconfig_, and the ninja lib seems to look for the previous cmake version:

```
Executing action: menuconfig
Running ninja in directory /Users/edouardmallecourt/Library/CloudStorage/OneDrive-PaloIT/Drop/drop/build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
/bin/sh: /usr/local/Cellar/cmake/3.23.1/bin/cmake: No such file or directory
FAILED: build.ninja
/usr/local/Cellar/cmake/3.23.1/bin/cmake --regenerate-during-build -S/Users/edouardmallecourt/Library/CloudStorage/OneDrive-PaloIT/Drop/drop -B/Users/edouardmallecourt/Library/CloudStorage/OneDrive-PaloIT/Drop/drop/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```

Need help to solve this or a fix, thank you

EdMlt, 2022-05-17:
I run on Mac Monterey 12.3.1

dobairoland, 2022-05-17:
Hi @EdMlt.

Please try the following:
```sh
ccache --clear
idf.py fullclean   # in your project's directory
```

EdMlt, 2022-05-17:
It's working,  thank you !

closed 2022-05-17 by dobairoland