espressif/esp-idf #8970: CONFIG_LWIP_ETHARP_TRUST_IP_MAC doesn't do anything

szmodz, 2022-05-17:
ETHARP_TRUST_IP_MAC  was removed form LwIP some 6 years ago. The define is not used anywhere in LwIP source code.

https://github.com/espressif/esp-lwip/blob/475d658a47b63223a50fd61f4032476e99d0885a/CHANGELOG#L750

```
grep -r ETHARP_TRUST_IP_MAC ../esp-idf/components
../esp-idf/components/lwip/port/esp32/include/lwipopts.h: * ETHARP_TRUST_IP_MAC==1: Incoming IP packets cause the ARP table to be
../esp-idf/components/lwip/port/esp32/include/lwipopts.h:#define ETHARP_TRUST_IP_MAC             CONFIG_LWIP_ETHARP_TRUST_IP_MAC
../esp-idf/components/lwip/Kconfig:    config LWIP_ETHARP_TRUST_IP_MAC
../esp-idf/components/lwip/lwip/CHANGELOG:  * etharp: removed ETHARP_TRUST_IP_MAC since it is insecure and we don't need
../esp-idf/components/lwip/lwip/CHANGELOG:  * opt.h: Disable ETHARP_TRUST_IP_MAC by default since it slows down packet
../esp-idf/components/lwip/lwip/CHANGELOG:    ETHARP_TRUST_IP_MAC, review SO_REUSE)
```

david-cermak, 2022-05-17:
Hi @szmodz 

I'm sorry, IDF is still a little behind... The option has only been removed recently from IDF in cfc52bacbc3ecc25c94157b37f2ef766fb8503ce

szmodz, 2022-05-17:
No problem, I didn't notice It's been fixed in master already. Thanks for the quick reply.

closed 2022-05-18 by Alvin1Zhang