espressif/esp-idf #8973: NIMBLE Mem allocation incorrect port for external RAM

gydronix, 2022-05-17:
Impossible to use PSRAM in NimBLE
As I can malloc/calloc functions in the In the port [esp_nimble_mem.c](https://github.com/espressif/esp-idf/blob/master/components/bt/host/nimble/port/src/esp_nimble_mem.c)
use defines which absent in project settings.

ocESP, 2022-05-20:
Hello @gydronix, 

Could you please mention the details of the issue you are currently facing? 

Please use the below provided template for editing the issue - 

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ocESP, 2022-05-20:
Moreover, in order to use [`CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL`](https://github.com/espressif/esp-idf/blob/31b7694551620522cc3fde0623321a38bfea762f/components/bt/host/nimble/port/src/esp_nimble_mem.c#L16) configuration, you need to do the following - 

In your project, run `idf.py menuconfig`. 
In the menuconfig -> Component config -> ESP32 (or your device) Specific -> enable "Support for external, SPI-connected RAM" option. 

Also, 
In the menuconfig -> Component config -> Bluetooth -> Nimble options -> Memory allocation strategy -> select "External SPIRAM"

Please let us know if you are still facing the issue, and add the details in the above format, mentioned in the previous comment.

gydronix, 2022-05-20:
Thank you, friend
It my mistake, I didn't attached PSRAM to system in menu config.
Solved!

closed 2022-05-20 by gydronix