espressif/esp-idf #8974: port_common.c:123: undefined reference to `app_main'

cyberumut, 2022-05-17:
hello all. 

Earlier i using esp-idf 4.1 now upgrated to 5.0 but not compiled my program.

where is the problem . 

my app_main:

void app_main(void)
{ bla bla bla 
	}

error: 

/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x14): undefined reference to `app_main'
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
/mnt/y/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:123: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
CMakeFiles/x-master.elf.dir/build.make:451: recipe for target 'x-master.elf' failed
make[2]: *** [xt-master.elf] Error 1
CMakeFiles/Makefile2:521: recipe for target 'CMakeFiles/xmaster.elf.dir/all' failed
make[1]: *** [CMakeFiles/x-master.elf.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

ESP-Marius, 2022-05-18:
Hi @cyberumut 

I think we'll need some more info to help you with this one. Could you please also provide the following if possible"

1.  Full log from the build process (including the commands you ran to build it)
3.  The projects `CMakeLists.txt` which adds the file with `app_main` -function to the build (usually in the same folder as the source file where `app_main` is defined)

cyberumut, 2022-05-18:
> Merhaba@cyberumut
> 
> Sanırım bu konuda size yardımcı olmak için biraz daha bilgiye ihtiyacımız olacak. Mümkünse aşağıdakileri de sağlar mısınız"
> 
> 1. Derleme sürecinden tam günlük (oluşturmak için çalıştırdığınız komutlar dahil)
> 2. `CMakeLists.txt`-function ile dosyayı yapıya ekleyen projeler (genellikle tanımlandığı `app_main`kaynak dosya ile aynı klasörde )`app_main`

my cmakeoutput: 
`The target system is: Generic -  - 
The host system is: Linux - 5.4.72-microsoft-standard-WSL2 - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/3.10.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/3.10.2/CompilerIdCXX/a.out"

Checking whether the ASM compiler is GNU using "--version" matched "(GNU assembler)|(GCC)|(Free Software Foundation)":
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Determining if the C compiler works passed with the following output:
Change Dir: /mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_4fc54/fast"
/usr/bin/make -f CMakeFiles/cmTC_4fc54.dir/build.make CMakeFiles/cmTC_4fc54.dir/build
make[1]: Entering directory '/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_4fc54.dir/testCCompiler.c.obj
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address     -o CMakeFiles/cmTC_4fc54.dir/testCCompiler.c.obj   -c /mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_4fc54
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_4fc54.dir/link.txt --verbose=1
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address      CMakeFiles/cmTC_4fc54.dir/testCCompiler.c.obj  -o cmTC_4fc54 
make[1]: Leaving directory '/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_99615/fast"
/usr/bin/make -f CMakeFiles/cmTC_99615.dir/build.make CMakeFiles/cmTC_99615.dir/build
make[1]: Entering directory '/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_99615.dir/CMakeCCompilerABI.c.obj
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address     -o CMakeFiles/cmTC_99615.dir/CMakeCCompilerABI.c.obj   -c /usr/share/cmake-3.10/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_99615
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_99615.dir/link.txt --verbose=1
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address     -v CMakeFiles/cmTC_99615.dir/CMakeCCompilerABI.c.obj  -o cmTC_99615 
Using built-in specs.
COLLECT_GCC=/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2021r2-patch3' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp3 [... cut ...]
/home/maniac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address  -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -DC_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS    CMakeFiles/cmTC_3536c.dir/src.c.obj  -o cmTC_3536c 
make[1]: Leaving directory '/mnt/c/Users/root/Desktop/espproject/5.0/test-master/NOMBRE_PROYECTO/build/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
`

ESP-Marius, 2022-05-18:
Thanks! Just to make sure:

Did you define your main function as `void app_main(void)` or `int main(void) ` as you mentioned in your last reply? Because it needs to be defined the first way.

cyberumut, 2022-05-18:
> Teşekkürler! Sadece emin olmak için:
> 
> Ana işlevinizi son cevabınızda belirttiğiniz gibi `void app_main(void)`mi tanımladınız mı ? `int main(void) `Çünkü ilk yoldan tanımlanması gerekiyor.

yes but there are different c files in the same main directory.

cyberumut, 2022-05-18:
> If you are using `main.cpp` (or similarly named file containing `app_main`) try:
> 
> ```
> extern "C" void app_main(void)
> {
> ...
> }
> ```
> 
> The C++ bindings are different than the C bindings which ld is looking for when linking your final elf file.

no cpp all files .c

cyberumut, 2022-05-18:
no cmakelist.txt under main/ folder

my cmakelist : 

# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(MAIN_SRCS main/main.c)

# This component includes modbus example common definitions
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/protocols/modbus/mb_example_common)

# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(main)

my main/ folder :
[image]

my master folder : 
[image]

ESP-Marius, 2022-05-19:
I think I see your issue. It seems that you havent updated to the CMake based build-system yet.

Your `main`-folder only has a `component.mk` which is for the old makefile based system, this is no longer supported in IDF 5.0 (see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/build-system.html#migrating-from-gnu-make-build-system)

Since there is no `CMakeLists.txt` in your main folder nothing from it actually gets built (which results in IDF not finding `app_main`)

Please update your main folder to use CMake (you can e.g. start by using the simple one from https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/CMakeLists.txt as an example).

ESP-Marius, 2022-06-09:
Closing this issue as the initial request seems to have been resolved.

closed 2022-06-09 by ESP-Marius

wangyiwu, 2023-05-04:
Im getting same error, even I already used cmake
```
FAILED: esp32_nat_router.elf 
cmd.exe /C "cd . && C:\Users\vuong\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\esp32_nat_router.elf.rsp -o esp32_nat_router.elf  && cd ."
c:/users/vuong/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x18): undefined reference to `app_main'
c:/users/vuong/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Users/vuong/eclipse-esp32/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/port_common.c:128: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

ESP-Marius, 2023-05-05:
> Im getting same error, even I already used cmake

If you are using c++ then please check https://github.com/espressif/esp-idf/issues/8974#issuecomment-1129947588

If that is not the issue then feel free to post your cmake files and the file that contains `app_main` somewhere and I can help you take a look.