espressif/esp-idf #8978: How to set custom component name?

stkw0, 2022-05-18:
I have a shared code, this code lives in a directory named "src" which contains a CMakeLists.txt for building a component out of this shared code, but obviously I don't want this component to be called "src" and I am not able to find a way of setting a custom name for this component.

It would be cool if something like set(COMPONENT_NAME whatever) worked, but it doesn't.

igrr, 2022-05-18:
Hi @stkw0, for components on the file system the name is always same as the directory name. There is no feature to change this. Component CMakeLists.txt file expansion happens too late to get the component name from component CMakeLists, and there isn't a simple way to change this (short of a pretty big overhaul in the build system.)

You can place the component CMakeLists.txt file into the directory `components/desired_component_name` and then use `target_sources` referencing the `src` directory of your project.

stkw0, 2022-05-18:
Having an (almost) empty folder just for this feels like a workaround. I wish there were a cleaner way to do this.

igrr, 2022-05-18:
It is indeed a workaround given your requirements (keep the code in `src` directory and not use `src` as the component name.)
However I suspect that you might also have some platform-specific code in your project, in addition to the shared code. You can place the platform (IDF) specific code into the component as well, so it won't be so empty.

stkw0, 2022-05-20:
Okay, thanks.

closed 2022-05-20 by stkw0