espressif/esp-idf #8979: module 'esptool' has no attribute '_main'

xiaguangbo, 2022-05-18:
## Environment

- Development Kit: nodemcu-32-s2
- Kit version (for WroverKit/PicoKit/DevKitC): master
- Module or chip used:  esp32s2
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-2959-g31b7694551
- Build System: vscode 'Espressif IDF' plugin the 'ESP-IDF Build Project' button
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32s2-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: vscode 'Espressif IDF' plugin the PowerShell terminal
- Using an IDE?: vscode 'Espressif IDF' plugin
- Power Supply: USB

## Problem Description

vscode 'Espressif IDF' plugin the 'ESP-IDF Flash device' button：
```
> Executing task in folder esp32s2: d:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe d:\software\esp\esp-idf\components\esptool_py\esptool\esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32s2 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 main.bin 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

Traceback (most recent call last):
  File "d:\software\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'
终端进程“d:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe 'd:\software\esp\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM6', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s2', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x10000', 'main.bin', '0x1000', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'”已终止，退出代码: 1。
```

do it manually：
can see that the 'esptool' module exists
```
PS D:\xiaguangbo\project\dap\project\esp32s2> d:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python 3.8.7 (tags/v3.8.7:6503f05, Dec 21 2020, 17:59:51) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import esptool
>>> esptool._main()
esptool.py v3.3.1
usage: esptool [-h] [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2}] [--port PORT] [--baud BAUD]
               [--before {default_...
...
```

I tried adding 'peintf('\n aaa \n')' to 'D:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Scripts\esptool.py' and 'D:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Lib\site-packages\esptool.py', but I don't see the output, but I can see two outputs when I add it to 'd:\software\esp\esp-idf\components\esptool_py\esptool\esptool.py', apparently 'ccc' is importing itself

If 'idf.py' is used, it is normal

radimkarnis, 2022-05-18:
Hello @xiaguangbo,
thank you for the detailed report, this will help us a lot!
There are reports of this issue in the esptool repository: https://github.com/espressif/esptool/issues/740
We will fix this ASAP.

radimkarnis, 2022-05-19:
@dotlambda This issue seems to be a result of [this wrapper script](https://github.com/espressif/esp-idf/blob/master/components/esptool_py/esptool/esptool.py) importing itself. This wrapper is a part of ESP-IDF and serves as a compatibility layer. You can see that in this one, we try to remove the path to the script itself:

```python
    sys.path.remove(os.path.dirname(os.path.realpath(__file__)))  # do not import this script
```

https://github.com/espressif/esptool/pull/742 concerns a wrapper script, which is directly in the esptool repository. The issue is similar, but I don't think both of these can be solved in the same place.

RavioliSauce, 2022-05-22:
@dotlambda I was getting the same "module 'esptool' has no attribute '_main'" error. Your fix worked for me! I am now able to communicate with my ESP8266!

xiaguangbo, 2022-05-24:
@radimkarnis Today I reinstalled the whole environment and it works fine, I don't know if this is a code fix or something is wrong with my previous environment

closed 2022-05-27 by xiaguangbo

wreyford, 2022-06-11:
I used ESP-IDF setup in Visual Studio Code insiders, and installed Master and tools using the interface, on a Windows10 PC, and got the same issue now. I cannot flash my cod to ESP32-S3-DevKitC-1

> Traceback (most recent call last):
  File "c:\esp\esp-idf-master\esp-idf\components\esptool_py\esptool\esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'

xiaguangbo, 2022-06-11:
@wreyford Delete the `esptool_py` folder and use `git pull`, delete `./espressif/python_env` and reinstall using the `install` script, try

LAestafa, 2024-09-19:
hola cuando intento flashear un esp32 me sale este error 
Traceback (most recent call last):
  File "C:\Users\Osniel\AppData\Local\Programs\Python\Python312\Scripts\esptool.py", line 37, in <module>
    esptool._main()
    ^^^^^^^^^^^^^
AttributeError: module 'esptool' has no attribute '_main'

como pudiera solucionar esto

LAestafa, 2024-09-19:
porque me sale este error 

Traceback (most recent call last):
  File "C:\Users\Osniel\AppData\Local\Programs\Python\Python312\Scripts\esptool.py", line 37, in <module>
    esptool._main()
    ^^^^^^^^^^^^^
AttributeError: module 'esptool' has no attribute '_main'

si alguien pudiera ayudar

invictus-anic3tus, 2024-09-25:
I have the exact same problem! I don't know what to do