espressif/esp-idf #8985: SPI Flash chip on ESP32-C3-32S module not recognised in boot code

DaveBerkeley, 2022-05-19:
During boot the JEDEC ID is read from the SPI Flash. There are a set of supported devices in components/spi_flash/.

My module returns the id 0x5e6015. I'm not sure which manufacturer this is so can't find the chip data. But it is not a supported device so the boot code selects a "generic" device. This then fails the chip size test.

My work around is to comment out the line "return ESP_ERR_FLASH_SIZE_NOT_MATCH;" in esp_flash_init_default_chip();

The device 0x5e6015 needs to be added to the supported devices list in esp_flash_registered_chips.

igrr, 2022-05-19:
Hi @DaveBerkeley, according to the ID this seems to be a 2MB flash chip (1 << 0x15 bytes). Indeed it is pretty odd because the [datasheet of the ESP32-C3-32S module](https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s_%E8%8B%B1%E6%96%87.pdf) from Ai-Thinker states that they include a 4MB flash chip.

In any case, the check is probably failing because you have set the flash size larger than 2 MB in the project configuration. Could you please check this? If you are unsure, please attach your `sdkconfig` file. I think if you select 2MB as the flash size, the program will probably work okay using the "generic" driver.

Also please mention which IDF version you are using. You can check this using `git describe --tags` command in IDF directory or by running `idf.py --version` command.

igrr, 2022-05-19:
> I can set back, but I still get the same error.

> E (980) spi_flash: Detected size(2048k) smaller than the size in the binary image header(4096k). Probe failed.

Somehow it seems that the PlatformIO build process still sets the flash size in the app binary image header to 4MB. I'm not sure how this is controlled in PlatformIO, exactly. I'd recommend asking on PlatformIO forums or filing an issue at https://github.com/platformio/platform-espressif32/issues.

igrr, 2022-05-19:
This discussion looks relevant: https://community.platformio.org/t/esp32-flash-size-in-menuconfig-doesnt-work/13961/6

Maybe adding this to the project ini file will help:
```
board_upload.flash_size=2MB
```

closed 2022-06-29 by Alvin1Zhang