espressif/esp-idf #8989: True P2P Hole Punch for IoT to run Without Cloud Premises and or Servers

filzek, 2022-05-19:
In the current days IoT become a role play for smartdevices where data was collect and consumed all over the places.

Using P2P Hole Punch could makes ESP-IDF SDK to be able to allow devices to open the connection to APP side and APP side to device without need to interface to any other servers to makes the data flow.

Images, sensor data and also a VoIP two channel audio to intradevices, intra APP~Device being complete able to be non cloud dependant.

So, this request would be the awesome 2022 feature to be implement over esp-idf esp32 series.

euripedesrocha, 2022-09-06:
Hi @filzek, this feature is in our pipeline, but we'll likely not work on it soon. We welcome any PR and contributions in this area.

filzek, 2024-05-22:
hi @espressif-abhikroy any progress we could test on this?

david-cermak, 2024-05-27:
Hi @filzek 

We haven't planned to work on this feature, yet, but looked around for libraries that could be ported to ESP platform. Here's one that doesn't need any dependencies, is relatively compact and published under MPL2.0 license:
https://github.com/paullouisageneau/libjuice
After taking a closer look, it seems like we only need to provide netif list (similar to linux's `ifaddrs.h`) and `getnameinfo()` implementation (in both cases, a very simplified version could be provided -- 1) we usually want to use only one netif for the actual NAT traversal; 2) could simply supply `inet_ntop()`, no need to provide reverse DNS lookup)

we still don't plan on officially adding this to IDF, in 2024, but the `juice` library looks very interesting and, if you're willing to spend some time on porting it, opening a PR, testing it, we will support the effort and help with certain low level implementations.

david-cermak, 2024-10-30:
FYI: There's been an effort of development of a simplified layer of linux-like implementations of common signalling and network interface abstractions in https://github.com/espressif/esp-protocols/pull/671
This little component opens the door to porting many linux/host-platform libraries, including the aforementioned libjuice.
You can check my initial (very WIP) port in https://github.com/david-cermak/esp-ice (including the [mqtt example](https://github.com/david-cermak/esp-ice/tree/main/examples/demo_2024_10) demonstrated on the last community event)

closed 2025-01-10 by david-cermak

david-cermak, 2025-01-31:
Linking espressif WebRTC solution: https://github.com/espressif/esp-webrtc-solution/tree/main/components/esp_webrtc