espressif/esp-idf #8991: error: 'portTickType' undeclared

cyberumut, 2022-05-19:
i have a problem . 
this problem how to solve ?

`/mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c: In function 'uart_event_task':
/mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c:182:56: error: 'portTickType' undeclared (first use in this function); did you mean 'TickType_t'?
         if(xQueueReceive(uart_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
                                                        ^~~~~~~~~~~~
                                                        TickType_t
/mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c:182:56: note: each undeclared identifier is reported only once for each function it appears in
In file included from /mnt/y/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /mnt/y/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /mnt/y/esp/esp-idf/components/driver/include/driver/uart.h:16,
                 from /mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c:39:
/mnt/y/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:120:25: error: expected expression before 'TickType_t'
 #define portMAX_DELAY ( TickType_t ) 0xffffffffUL
                         ^~~~~~~~~~
/mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c:182:69: note: in expansion of macro 'portMAX_DELAY'
         if(xQueueReceive(uart_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
                                                                     ^~~~~~~~~~~~~
/mnt/y/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:120:38: error: expected ')' before numeric constant
 #define portMAX_DELAY ( TickType_t ) 0xffffffffUL
                                      ^~~~~~~~~~~~
/mnt/c/Users/root/Desktop/espproject/5.0/esp32-test-master/components/test/src/rs485.c:182:69: note: in expansion of macro 'portMAX_DELAY'
         if(xQueueReceive(uart_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
                                                                     ^~~~~~~~~~~~~
esp-idf/test/CMakeFiles/__idf_test.dir/build.make:1262: recipe for target 'esp-idf/test/CMakeFiles/__idf_test.dir/src/rs485.c.obj' failed
make[2]: *** [esp-idf/test/CMakeFiles/__idf_test.dir/src/rs485.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....`

cyberumut, 2022-05-19:
problem solve. 
i do portTickType change to TickType_t

Alvin1Zhang, 2022-05-20:
Thanks for reporting and glad to hear issue is resolved, feel free to reopen.

closed 2022-05-20 by Alvin1Zhang