espressif/esp-idf #8992: Esp32 failed to send one to many using espnow protocol

xiezhoubin, 2022-05-20:


Alvin1Zhang, 2022-05-20:
Seems this is duplicate of https://github.com/espressif/esp-idf/issues/8988, will close https://github.com/espressif/esp-idf/issues/8988 and track in this one. Thanks.

zhangyanjiaoesp, 2022-06-01:
@xiezhoubin I have tested with IDF v4.4.1 locally, ESP32 can send one two many using espnow protocol. 
The error `ESP_ERR_ESPNOW_INTERNAL` show that sending is fail, you can check whether your mac address is correct, or whether your ESP devices are in the same channel.

xiezhoubin, 2022-06-01:
> @xiezhoubin I have tested with IDF v4.4.1 locally, ESP32 can send one two many using espnow protocol. The error `ESP_ERR_ESPNOW_INTERNAL` show that sending is fail, you can check whether your mac address is correct, or whether your ESP devices are in the same channel.

I have checked it and there is no problem. The above Arduino sample code is used. I tried the same code. The package in version 1.0.6 can run normally, and the package in versions 2.0.1, 2.0.2 and 2.0.3 will report errors.

e0v1a, 2022-06-02:
I also encountered the same problem, in the same channel, between two fixed paired devices, this error will appear from time to time, sometimes it will appear every time, sometimes it will swipe the screen crazy, but even in this The communication does not seem to be affected when the error is reported and the screen is swiped (I use him to transmit stateless data, so it does not affect the occasional loss of a few)
And the internal error is a very vague expression that does not give any guidance (such as what is this problem related to? Send frequency? Interrupt?), which makes it impossible for developers to try to solve the problem

zhangyanjiaoesp, 2022-06-07:
@e0v1a @xiezhoubin This issue has been solved in IDF branch 5d95cd58b210523e68fb37e0fe23cd0331ff2585, will update it to Arduino as soon as possible.

zhangyanjiaoesp, 2022-07-12:
@xiezhoubin Have you updated the branch and have a test again?

xiezhoubin, 2022-07-12:
> @xiezhoubin Have you updated the branch and have a test again?

Is it version 2.0.4?

zhangyanjiaoesp, 2022-07-12:
> > @xiezhoubin Have you updated the branch and have a test again?
> 
> Is it version 2.0.4?

yes, please check whether your problem has been solved.

xiezhoubin, 2022-07-12:
The same code in the new version is compilation error.

Version 1.0.6 is OK.
[image]

Compilation error of version 2.0.4.
[image]

xiezhoubin, 2022-07-12:
The problem of compilation error reporting has been solved. Using memcpy requires adding (void*). But after running the code, it can't run, and the serial port output reports an error. There is no problem with the same code in version 1.0.6.
[image]

xiezhoubin, 2022-07-12:
Still not, version 2.0.4, in one to many transmission, will only be sent to the first.
It is recommended to refer to the link test below.
https://randomnerdtutorials.com/esp-now-one-to-many-esp32-esp8266/

Espressif-liuuuu, 2022-07-20:
hi @xiezhoubin, please try to call `esp_now_set_wake_window(65535)` after `esp_now_init()` for all of your esp_now rx devices to see if it could solve your problem. Thanks:)

e0v1a, 2022-07-21:
> hi @xiezhoubin, please try to call `esp_now_set_wake_window(65535)` after `esp_now_init()` for all of your esp_now rx devices to see if it could solve your problem. Thanks:)

It doesn't work, in a master-multiple-slave case, the broadcast data is always only received by the first slave device

zhangyanjiaoesp, 2022-07-21:
I have reproduced the problem locally, and will resolve it ASAP.

zhangyanjiaoesp, 2022-07-27:
@xiezhoubin To solve the problem, please change the code `cfg.cache_tx_buf_num = 1` to `cfg.cache_tx_buf_num = 16`
https://github.com/espressif/arduino-esp32/blob/4b64cde1022975cc8d72c41c500abb3d072762cf/libraries/WiFi/src/WiFiGeneric.cpp#L671
or just delete the code 

> 	if(!WiFiGenericClass::useStaticBuffers()) {
	    cfg.static_tx_buf_num = 0;
            cfg.dynamic_tx_buf_num = 32;
	    cfg.tx_buf_type = 1;
            cfg.cache_tx_buf_num = 1;  // can't be zero!
	    cfg.static_rx_buf_num = 4;
            cfg.dynamic_rx_buf_num = 32;
        }

We will add this change to the office Arduino code ASAP.

SuGlider, 2022-07-27:
@zhangyanjiaoesp @xiezhoubin

All you need to do is:

```c++
 void setup() {
   Serial.begin(115200);

   WiFi.useStaticBuffers(true);  // it will keep sdkconfig default for cache and static TX buffers in WiFi
   WiFi.mode(WIFI_STA);
   
   // .... your code
```

SuGlider, 2022-07-27:
``` WiFi.useStaticBuffers(true);``` will force Arduino to use deafult sdkconfig WiFi setup:

```
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=16
```

Alvin1Zhang, 2022-08-03:
https://github.com/espressif/arduino-esp32/pull/7044 has been merged, feel free to reopen.

closed 2022-08-03 by Alvin1Zhang

lebossejames, 2023-03-05:
Hello,

I have same issue with ESP32-C3 with 2.0.7 library.
Can you help me please?

zhangyanjiaoesp, 2023-03-06:
@lebossejames Please create a new Github issue to describle your problem in details, thanks.