espressif/esp-idf #8993: How to send JSON message in WebSocket's

vinothkrstatiq, 2022-05-20:
https://github.com/espressif/esp-protocols/blob/master/components/esp_websocket_client/examples/main/websocket_example.c

given example websocket_example.c how can i send json message to any WebSocket server,
If you could suggest some examples or methods that how i can implement, it would be helpful.

chegewara, 2022-05-20:
https://github.com/espressif/esp-protocols/blob/master/components/esp_websocket_client/examples/main/websocket_example.c#L122

vinothkrstatiq, 2022-05-22:
@chegewara thanks a lot man 
i also need another help.
How to store received WebSocket data in variable could you give one example code here??

vinothkrstatiq, 2022-05-25:
@david-cermak 
from above
is it cJSON or JSON
eg
cJSON *elem = cJSON_GetArrayItem(root, 0);
JSON *elem = cJSON_GetArrayItem(root, 0);

david-cermak, 2022-05-25:
Thanks for noticing the problem, of course  it is:
```cpp
cJSON *elem = cJSON_GetArrayItem(root, 0);
```

vinothkrstatiq, 2022-05-25:
@david-cermak  it is getting crashed 
assert failed: heap_caps_free heap_caps.c:304 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace:0x4008181e:0x3ffb5f500x40085051:0x3ffb5f70 0x40089f55:0x3ffb5f90 0x40081a6f:0x3ffb60b0 0x40089f85:0x3ffb60d0 0x400d54fe:0x3ffb60f0 0x400d4b3a:0x3ffb6110 0x400e685c:0x3ffb6130 0x40087951:0x3ffb6150
0x4008181e: panic_abort at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/panic.c:402

0x40085051: esp_system_abort at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40089f55: __assert_func at C:/Espressif/frameworks/esp-idf-v4.4/components/newlib/assert.c:85

0x40081a6f: heap_caps_free at C:/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_caps.c:304 (discriminator 1)

0x40089f85: free at C:/Espressif/frameworks/esp-idf-v4.4/components/newlib/heap.c:39

0x400d54fe: cJSON_free at C:/Espressif/frameworks/esp-idf-v4.4/components/json/cJSON/cJSON.c:3109

0x400d4b3a: app_main at C:\Users\Vinoth\Desktop\esp32\json1\build/../main/main.c:20

0x400e685c: main_task at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:129 (discriminator 2)

0x40087951: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 5d261fc7ed972ddf

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694

david-cermak, 2022-05-26:
@vinothkrstatiq The error message is telling you what was wrong:
```
assert failed: heap_caps_free heap_caps.c:304 (heap != NULL && "free() target pointer is outside heap areas")
```
trying to free a variable (`nn`) that was allocated on stack. Unfortunately `cJSON_free()` is not smart enough to tell if the supplied argument actually points to a `cJSON` object.

sorry, haven't posted a complete code. You can even try to compile the below snippet on your host platform, debug it and experiment with cJSON library per your needs.
```cpp
#include <stdio.h>
#include "cJSON/cJSON.h"

int main()
{
	const char *nn = "[{\"id\":\"1\",\"temp\":\"25.4\"},{\"id\":\"2\",\"temp\":\"25.8\"}]";
	cJSON *root = cJSON_Parse(nn);
	for (int i = 0 ; i < cJSON_GetArraySize(root) ; i++) {
	cJSON *elem = cJSON_GetArrayItem(root, i);
		cJSON *id = cJSON_GetObjectItem(elem, "id");
		cJSON *name = cJSON_GetObjectItem(elem, "temp");
		printf("%i id:%s temp:%s\n", i, id->valuestring, name->valuestring);
	}
	cJSON_Delete(root);
	return 0;
}
```

vinothkrstatiq, 2022-05-26:
@david-cermak 
could you try the example below in your side for me it is crashing 
Below is my json string which i m trying
[3,"af2e620d-8571-49f0-b3c2-0144c2b4f9bf",{"currentTime":"2022-05-25T17:38:45.311043","interval":10,"status":"Accepted"}]
/* The example of ESP-IDF
 *
 * This sample code is in the public domain.
 */
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "cJSON.h"

static const char *TAG = "JSON";

void app_main()
{   const char *nn = "[3,\"531531531\",{\"currentTime\":\"2022-05-21T17:02:43.677816\",\"interval\":20,\"status\":\"Accepted\"}]";
   
	cJSON *root = cJSON_Parse(nn);
	for (int i = 0 ; i < cJSON_GetArraySize(root) ; i++) {
	cJSON *elem = cJSON_GetArrayItem(root, i);
		cJSON *id = cJSON_GetObjectItem(elem, "currentTime");
		cJSON *name = cJSON_GetObjectItem(elem, "interval");
		printf("%i id:%s temp:%s\n", i, id->valuestring, name->valuestring);
	}
	cJSON_Delete(root);

}

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d4b41  PS      : 0x00060130  A0      : 0x800e67c7  A1      : 0x3ffb6110
0x400d4b41: app_main at C:\Users\Vinoth\Desktop\esp32\json1\build/../main/main.c:20 (discriminator 3)

A2      : 0x00000000  A3      : 0x3ffaf8c8  A4      : 0x3ffaf89c  A5      : 0x00000000
A6      : 0x3f400120  A7      : 0x3ffe3b80  A8      : 0x800d4b41  A9      : 0x3ffb60f0
A10     : 0x00000000  A11     : 0x3f40305c  A12     : 0x00004e20  A13     : 0x383f8000
A14     : 0x3ffb7214  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400d4b3e:0x3ffb61100x400e67c4:0x3ffb6130 0x40087951:0x3ffb6150
0x400d4b3e: app_main at C:\Users\Vinoth\Desktop\esp32\json1\build/../main/main.c:19 (discriminator 3)

0x400e67c4: main_task at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:129 (discriminator 2)

0x40087951: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 9479df54a85f48d9

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 23:17:42
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0784ch ( 30796) map
I (121) esp_image: segment 1: paddr=00017874 vaddr=3ffb0000 size=0234ch (  9036) load
I (125) esp_image: segment 2: paddr=00019bc8 vaddr=40080000 size=06450h ( 25680) load
I (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=16988h ( 92552) map
I (173) esp_image: segment 4: paddr=000369b0 vaddr=40086450 size=04d14h ( 19732) load
I (181) esp_image: segment 5: paddr=0003b6cc vaddr=50000000 size=00010h (    16) load
I (187) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
I (202) cpu_start: Pro cpu up.
I (202) cpu_start: Starting app cpu, entry point is 0x40080fe4
0x40080fe4: call_start_cpu1 at

chegewara, 2022-05-26:
1. you should check if `id != NULL` and `name != NULL` before you go to print line.
2. You seem to not understand JSON; 
- yes, it is array:
`3,"531531531",{"currentTime":"2022-05-21T17:02:43.677816","interval":20,"status":"Accepted"}]`
- the array size is 3
- array[0] (element 0) is a plain integer equal 3
- array[1] is plain string
- in both cases id and name will be `NULL`, thats why it crash
- the 3rd elemnt is object, so you can try to find values you want

vinothkrstatiq, 2022-05-26:
@chegewara 
Thanks for the suggestions

closed 2022-06-27 by Alvin1Zhang