espressif/esp-idf #8994: esp_ble_gap_disconnect 引起卡死 直至看门狗触发重启

fzboffice, 2022-05-20:
idf commit fe1cf490b9e26689ce6d4b8268c97685e6e73c27

Alvin1Zhang, 2022-05-23:
@fzboffice Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

fzboffice, 2022-05-23:
@Alvin1Zhang My description is wrong, esp_ ble_ gatts_ Close and ESP_ ble_ gap_ Disconnect will not cause jamming, it just will not cause a disconnect event, but the remote device will disconnect.
My device is esp32 and idf commit is c5deb7a9713887263b63549a891282f51a65d167

fzboffice, 2022-05-23:
When I called it many times, the problem was solved

closed 2022-05-23 by fzboffice