espressif/esp-idf #8995: Hi @david-cermak

supradeeplohith, 2022-05-20:
Hi @david-cermak
1. Openssl client is not working can please tell me why? previously i was able to get at least data from last 3 days im not able get the data itself. please help me
2.  as per the template openssl client example template, i have mentioned above , i will get the data as stream from as a form 
of json ,after that  it only gives only data when update in firebase for particular node it will gives..i want same as above explanation application @david-cermak 
3. overall - if any update in firebase for particular node openssl was giving updated data for particular node in json,
4. i request you to please check the code - tell me the reason why this openssl client is not working as it was functioning before, now its completely not working   please refer my serial terminal screen  
[image]

_Originally posted by @supradeeplohith in https://github.com/espressif/esp-idf/issues/8656#issuecomment-1132716570_

david-cermak, 2022-05-20:
Hi @supradeeplohith 

There could be 2 issues:
1) From your screenshot, it looks like the plain TCP connection failed before it tried to establish a TLS handshake. This could be some connectivity problem of your local network. Could you please try to connect to the same endpoint from you PC, e.g.:
```
wget https://<your address and path>
```
or just a simple tcp connect:
```
nc <your address> 443
```

2) Redirection, I've checked the endpoint you mentioned in the linked issue and it redirects to a different URI. Maybe that's the reason why it was originally  *working*... Openssl client example doesn't follow http redirections, unless you implement it manually. You can use `https_client` component, which does support it. Anyway this doesn't seem to you the problem from your log, as you were not able to connect at all, not only on TCP layer.

---
One general comment: Could you please follow the template, when creating a new issue? It would be very helpful for IDF support team and maybe also for users. Thank you!

supradeeplohith, 2022-05-21:
Hi @david-cermak I've checked my internet connection everything looks good David, even i am tried with Openssl_Client() in espidf version 2 ie V.4.3 I've maintained default everything and only Wi-Fi credential in sdk config, I've changed in order to connect to my networ
![1](https://user-images.
[image]
githubusercontent.com/42844966/169638458-c1ccc2fe-a3e9-4ef4-b5fc-f42975ea8439.PNG)

please let me know you comment in this Expecting quick response

david-cermak, 2022-05-23:
Hi @supradeeplohith 

Your connectivity problems are unrelated to the openssl-client example. What I see in all your screenshots is that the TCP connection fails after some 10-20 seconds, which is the default connection timeout. You can try to debug these issue to look where these packets go, so in this case I'd expect you'd see number (~12) of TCP SYN packets retransmitted with no response.
There could be various problems in your network, I cannot help much here, but let me just try to list some:
* your connection might be blocked by some firewall, please contact your network admin to check the settings and the blocked communication. this is likely to happen is some bigger managed, corporate networks, some smart systems might use some AI adaptive learning and when people *play* with that and send multiple invalid packets (e.g. when your IoT app gets stuck in a reset loop and sends various invalid data to the network...) this could be marked as malicious and blocked.
* check the network and it's connectivity manually, join the same AP with your laptop or phone and send some diagnostic commands like `traceroute` or simply `nc` to the endpoint. You can use the resolved IP addresses or try to look it up in your system.
* Try the basic socket example https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_client to connect to your host. Try to resolve the hostname manually and put an IP and port (443) and check if you can simply connect.
* Double check your code and settings. From your last screenshot, it seems like you're connecting the `baidu.com`, but the resolved address geo location points to `Maharashtra`, so unless they have a geo mirror in India, there might be still some wrong hostname in your code and a stale DNS entry?

david-cermak, 2022-05-23:
@supradeeplohith Please stop creating new issues and use a bit more meaningful issue titles and description!
1) Rename this issue to the problem you're currently facing, e.g. "Cannot connect to `https` server with openssl-client example"
2) Close all your remaining issues
3) Follow IDF issue template, assuming this is a bug-report, please use https://github.com/espressif/esp-idf/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

---

About the issue:

> *in the case of previous comment from your end if that would be a network issue i could face isuue with using http_mbedtls also right?

Yes, I agree, this is not a network issue. All these example, that we were talking about `openssl`, `https_request`, `mbedtls` work the same way in the background, so I'd suspect you've made some changes to the openssl_client, e.g. updated the hostname, or something.

I see (from the screenshots and some other issues that you raised), that your IDF version is `v4.3`.
I have checked the original `openssl_client` example from `v4.3.2-574-g5fe7d538a` (5fe7d538a5) and updated per your settings:
```patch
diff --git a/examples/protocols/openssl_client/main/openssl_client_example.h b/examples/protocols/openssl_client/main/openssl_client_example.h
@@ -17,10 +17,14 @@
    the config you want - ie #define OPENSSL_EXAMPLE_TARGET_NAME "www.baidu.com"
    and ie #define OPENSSL_EXAMPLE_TARGET_TCP_PORT 433
 */
-#define OPENSSL_EXAMPLE_TARGET_NAME        CONFIG_TARGET_DOMAIN
+#define OPENSSL_EXAMPLE_TARGET_NAME        "ogweinkling-default-rtdb.firebaseio.com"
 #define OPENSSL_EXAMPLE_TARGET_TCP_PORT    CONFIG_TARGET_PORT_NUMBER
 
-#define OPENSSL_EXAMPLE_REQUEST            "{\"path\": \"/v1/ping/\", \"method\": \"GET\"}\r\n"
+#define WEB_URL "https://ogweinkling-default-rtdb.firebaseio.com/hardware_light_fan/.json"
+#define OPENSSL_EXAMPLE_REQUEST            "GET " WEB_URL " HTTP/1.0\r\n"  \
+    "Host: "OPENSSL_EXAMPLE_TARGET_NAME"\r\n"                                               \
+    "User-Agent: esp-idf/1.0 esp32\r\n"                                    \
+    "\r\n";
 
 #define OPENSSL_EXAMPLE_TASK_NAME        "openssl_example"
```
and everything worked as expected.

> *now i would like u to request to tell me the changes in code, only if update occurs the updated data ( json) with respect to node I should need to get the data

Please check this comment https://github.com/espressif/esp-idf/issues/8656#issuecomment-1132496741 (point 2) from your earlier issue.

david-cermak, 2022-05-25:
Hi @supradeeplohith 

The openssl-API component in IDF was deprecated in `v4.4` and removed in `v5.0`, please see the docs:
* [Removed/deprecated components](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/removed-components.html#removed-or-deprecated-components)

Which means, that you can still use it on `v4.3` and `v4.4`, but please check the support and maintenance period in
* [Support Periods](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods)

So the IDF's `openssl_client` from `v4.4` would be still available and maintained until Aug 2024 (with a deprecation warning though)

closed 2022-06-29 by Alvin1Zhang