espressif/esp-idf #8996: BLE authentication failure crashes ESP32-S3

Huckies, 2022-05-20:
## Environment

- Development Kit: ESP32-S3-DevKit-M
- IDF version:     v4.4.1
- Build System:     idf.py
- Compiler version:     xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     macOS
- Using an IDE?:     VS Code
- Power Supply:     USB

## Problem Description

Example ``esp_hid_host`` crashes when the peripheral fails to input the passkey in time.

### Expected Behavior

Disable current connection and keep scanning.

### Actual Behavior

A call to ``esp_hidh_dev_delay_free()`` corrupts the heap and leads to a panic.

### Steps to reproduce

I choose to run ``esp_hid_host`` on the  esp32-s3 board and leave ``esp_hid_device`` to a esp32-pico-d4 development board. Setting ``io_cap`` in BLE security params to ``ESP_IO_CAP_OUT`` and ``ESP_IO_CAP_IN`` respectively. 

## Debug Logs

```
BLE GAP AUTH ERROR: 0x63

CORRUPT HEAP: multi_heap.c:108 detected at 0x3fca57cc

abort() was called at PC 0x40382075 on core 0
0x40382075: multi_heap_assert at /Users/huckies/esp-idf/components/heap/multi_heap_platform.h:61
 (inlined by) assert_valid_block at /Users/huckies/esp-idf/components/heap/multi_heap.c:106

Backtrace:0x40375b36:0x3fcf2b000x4037cac1:0x3fcf2b20 0x40382412:0x3fcf2b40 0x40382075:0x3fcf2bb0 0x40382141:0x3fcf2bd0 0x40375d69:0x3fcf2bf0 0x40382559:0x3fcf2c10 0x4200be6d:0x3fcf2c30 0x4200bef3:0x3fcf2c50 0x42005f43:0x3fcf2c70 
0x40375b36: panic_abort at /Users/huckies/esp-idf/components/esp_system/panic.c:402

0x4037cac1: esp_system_abort at /Users/huckies/esp-idf/components/esp_system/esp_system.c:128

0x40382412: abort at /Users/huckies/esp-idf/components/newlib/abort.c:46

0x40382075: multi_heap_assert at /Users/huckies/esp-idf/components/heap/multi_heap_platform.h:61
 (inlined by) assert_valid_block at /Users/huckies/esp-idf/components/heap/multi_heap.c:106

0x40382141: multi_heap_free_impl at /Users/huckies/esp-idf/components/heap/multi_heap.c:216
 (inlined by) multi_heap_free_impl at /Users/huckies/esp-idf/components/heap/multi_heap.c:209

0x40375d69: heap_caps_free at /Users/huckies/esp-idf/components/heap/heap_caps.c:340

0x40382559: free at /Users/huckies/esp-idf/components/newlib/heap.c:39

0x4200be6d: esp_hidh_dev_resources_free at /Users/huckies/esp-idf/components/esp_hid/src/esp_hidh.c:660 (discriminator 3)

0x4200bef3: esp_hidh_dev_delay_free at /Users/huckies/esp-idf/components/esp_hid/src/esp_hidh.c:750 (discriminator 4)

0x42005f43: timer_process_alarm at /Users/huckies/esp-idf/components/esp_timer/src/esp_timer.c:360
 (inlined by) timer_task at /Users/huckies/esp-idf/components/esp_timer/src/esp_timer.c:386
```

## Supplementary instruction

After throwing an authentication error in ``ESP_GAP_BLE_AUTH_CMPL_EVT``, it seems that the program entered ``hidh_callback`` with event ``ESP_HIDH_OPEN_EVENT`` which is not what I expected. I think it's the unusually short interval between 'connect' and 'disconnect' (~70ms) caused the re-free of memory and break down the program at last.

esp-cjh, 2022-08-12:
We have found the reason of crash. We will fix it ASAP.

AxelLin, 2023-04-29:
@esp-cjh How is the status?

esp-cjh, 2023-05-23:
@AxelLin This bug has been fixed. Please update to the new version.

AxelLin, 2023-05-23:
@esp-cjh Could you share commit of the fix and close the issue if it's fixed in all impacted branches.

Alvin1Zhang, 2023-05-24:
Thanks for reporting and tracking, fix on master branch is available https://github.com/espressif/esp-idf/commit/88b71f6416b7daac659317b84f91d2be52ff173c, and back ported to other release branches. 

* release/5.0: https://github.com/espressif/esp-idf/commit/c76ecbe6828b9f4a2eccaf82485e2d4171a6aa0d
* release/4.4: https://github.com/espressif/esp-idf/commit/85cc471e8e9cb4b058b9a94a179bfb7ddc0207da 
* release/4.3: https://github.com/espressif/esp-idf/commit/904096432103b8528a04953ce2870da8c38a6192

Feel free to reopen.

closed 2023-05-24 by Alvin1Zhang