espressif/esp-idf #8999: Substantial increase in power consumption when ADC is enabled during light sleep

MarcGeh, 2022-05-21:
## Environment

- Module or chip used: ESP32-PICO-D4
- IDF version v4.4.1
- Power Supply: external 3.3V

## Problem Description

Hey guys. My problem is that I want to understand the power consumption of the ESP32+ULP with the ADC enabled and whether I'm doing something wrong. I created a [test program](https://github.com/MarcGeh/esp32_ulp_test) which has the ULP running based on the [example system/ulp_fsm/ulp_adc](https://github.com/espressif/esp-idf/tree/v4.4.1/examples/system/ulp_fsm/ulp_adc) with some modifications to it. This ULP program wakes up the main CPU every 4 seconds and does some ADC samples once right before the main CPU is woken up.
I tested this ULP program with the main CPU being in deep sleep as well as in light sleep. The current consumption I measured during deep sleep was around an average of 22µA (I'm sure that is an inaccurate reading since I would expect more but lets just say it is pretty low). During light sleep without the ADC enabled I get an average of 1.07mA. Once I also enable the ADC during light sleep I get an average of 2.53mA. So an increase of 1.5mA for running the ADC. What I really would like to understand: How is it possible to run the ADC in deep sleep with this minuscule current consumption while in light sleep it takes 1.5mA to use it? Is this expected behavior, am I configuring something wrong or is this a bug

I'm using an Otii-Arc and I'm measuring the consumption of an ESP32-PICO dev-kit where I removed everything except the ESP32-PICO-D4. Please see the readme of the [test program](https://github.com/MarcGeh/esp32_ulp_test) for the measurement results.

I appreciate any help/explanation. Thanks in advance.

esp-wzh, 2024-08-15:
Sorry for noticing this report so late. On ESP32, due to hardware limitation, it is necessary to keep the RTC_PERIPH power domain powered during lightsleep. Usually it does not bring too much power consumption, but if the ADC power is turned on, things will become worse, the ADC will bring about 1.5 mA of leakage on the RTC_PERIPH power domain, which is exactly the problem you are encountering.

BTW, we do not recommend using ULP during lightsleep, because the HP and ULP processor architectures are different and there is no unified ADC driver, the HP core and LP core operating the ADC peripheral together will cause more problems, and this usage scenario has been banned in IDF after v4.4.6.

It is recommended that for high frequency ADC sampling requirements, you can create a sampling task in HP and cooperate with the esp_pm component to reduce power consumption. For low frequency ADC sampling, you can use deepsleep + ULP ADC to achieve it.

MarcGeh, 2024-08-15:
Thanks for clarifying

closed 2024-08-16 by Alvin1Zhang