espressif/esp-idf #9003: Eclipse plug in an commandline idf do not work parallel on same project

ralphde, 2022-05-22:
Using Eclipse IDF plugin and IDF 4.4

We have build a small application based on the https-server sample project and have added very small functionality to it including tinyUSB stack used for logging through USB.

It all worked fine, compiling debugging etc.

Then we wanted to use flashing through USB and for this it is required to run idf.py dfu and dfu-flash for flashing through USB.

Consequently we opened the corresponding IDF CMD 4.4 from the windows start menu for performing this task.
Running idf.py dfu caused kicking in git and downloading things. Very surprisingly and unexpected. We have been working until now with Eclipse IDE seamlessly and would never expect this behavior.

the dfu file was build after recompiling.

Since this moment nothing worked at all. Not the unchanged sample projects could be compiled in Eclipse nor our own projects.
The message we got was "tool chain version mismatches"

To fix this we updated the IDF to 4.4.1 following the steps given in the documentation. idf.py builds the projects seamlessly.

However, Eclipse Plugin does not compile the project BUT flashes and debugs the project seamlessly.

The compile is ended with a short message "Build not configured correctly" and then flashed and debugged.

It seems that Eclipse plug in and idf.py do not function together on the same project.

What can we do to recover our custom projects? and what has happened?

However, that is not the end of the story. While all sample projects compile in idf.py fine and run after, our custom https-server based project does not run at all. It hits a break point at right after start up.

`
cpu_ll_get_cycle_count()
`

Looking into that, we assume that this may indicate that the ESP goes directly to sleep?

Again, our code was running before without any error or messages. As well this issue appears before any debug or even main() is hit. How to nail this dead end down?

Are we still having version mismatch issues? if so, how can we locate the issue?

kolipakakondal, 2022-05-23:
Hi @ralphde 

Will try to replicate this behavior to check further on this.

Eclipse also provides an [ESP-IDF Terminal](https://github.com/espressif/idf-eclipse-plugin#idfterminal) with all the preconfigured environment variables required to build and flash idf projects. As this will use the existing tools installed based on the idf versions, you might not get into this "toolchain version mismatches" error.  You could run idf.py dfu command directly from ESP-IDF Terminal without leaving the eclipse environment.

"Build not configured correctly" usually appears when the [launch target](https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target) is not configured or some eclipse project configuration files are overwritten from the command line. I think that's what would have happened in your case. To fix this problem, could you try closing the project and re-open it(Right-click on the project and close)?   

Regarding "custom https-server based project does not run at all" - @dobairoland could help on this

ralphde, 2022-05-23:
Hi @kolipakakondal 
before I performed the dfu command I performed idf.py --version in both ESP-IDF Terminal and the Windows start menu IDF CMD. Both reported IDF 4.4 as version.
I expected that the launch target was set, as we used Eclipse only since beginning of the project and there the target is set perfectly. However, in the CMD window it was not set, I had to perform 'idf.py set-target'.
I don't know if that would have been the case using ESP-IDF Terminal, but I would expect so, unless the init of the ESP-IDF Terminal is not taking care of this. Can you disclosure details how Eclipse Plugin sets the launch target and what is the difference to idf.py set-target?
Previous installed sample projects I get not to compile again at all, I have to delete them and install fresh from sample projects. I am very curious why, because I have one custom project based on HTTP-Server sample which I can't get to compile at all now. I would expect that a manual clean by deleting the build folder would reset everything, but no, not at all. Is a python-clean required and what is the difference?
May an 'import existing project' help? e.g. copying the project to somewhere and import it from there into the workspace again?

dobairoland, 2022-05-23:
> Regarding "custom https-server based project does not run at all" - @dobairoland could help on this

Please open a separate issue on that so that the right developer could take a look at it. But please fill out the issue template next time and provide more information so that we could help you.

ralphde, 2022-05-25:
@kolipakakondal well @dobairoland does not seem to be of help with this issue. However, we have the problem that a well working custom project stopped doing so since 
1. we run idf.py dfu from windows CMD which kicked in some updates via GIT which we didn't log what was updated. IDF 4.4 was reported at this time.
2. keeps behaving the same after a full upgrade to IDF 4.4.1 with hitting a system break point at `cpu_ll_get_cycle_count()`  in the startup code before app_main

I am personally would say that must have to do with an incompatible startup code included in the build. The issue appears outside the custom code or components.

Remarkable is that new imported sample projects work fine.

So what went wrong in the installation at calling idf.py dfu ?

I have seen that setting the target in Eclipse Plugin has no effect for idf.py. I have also noticed that setting idf.py set-target has no effect on taget in eclipse plugin.

Could it be possible that using both environment parallel breaks the set-target information so that the wrong startup code is linked?

@kolipakakondal I am still convinced that the source of this issue is coming from the tool set up. Can you please assign someone who just can explain what can cause the startup code landing at `cpu_ll_get_cycle_count() ` please?
and may be how we can debug this?

closed 2022-05-25 by dobairoland