espressif/esp-idf #9004: Blu with Bluedroid not support class of device set

hellolxhlinux, 2022-05-23:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

hellolxhlinux, 2022-05-23:
I need to filter device with peripheral in host ,but device with bluedroid in esp32c3  not support class of device at ble mode ,I dont know why ?

seaskycjh, 2022-06-08:
BLE does not have the concept of cod (class of device), cod can only be used in classic bt. You can set the appearace in advertise data to filter devices more precisely.

closed 2022-07-06 by hellolxhlinux