espressif/esp-idf #9005: ESP32-C3 got "invalid header: 0xffffffff" and cannot erase-flash

AxelLin, 2022-05-23:
ESP32-C3 got "invalid header: 0xffffffff" and cannot erase-flash

- Module or chip used: ESP32-C3 on a customized board
- IDF version: v5.0-dev-3043-g70109e2393
- Build System: idf.py
- Compiler version: riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

It looks like "idf.py app-flash" can successfully download the firmware,
however, monitor shows "invalid header: 0xffffffff".

Also note "idf.py erase-flash" got error:
A fatal error occurred: ESP32-C3 ROM does not support function erase_flash.
esptool.py failed with exit code 2

See below log:

$ idf.py app-flash monitor
Executing action: app-flash
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-C3
Running ninja in directory /home/axel/esp/esp-idf-dev/apps/myproj/build
Executing "ninja app-flash"...
[1/2] cd /home/axel/esp/esp-idf-dev/apps/myproj/build/esp-idf/esptool_py && /home/axel/.espressif/p...ld/partition_table/partition-table.bin /home/axel/esp/esp-idf-dev/apps/myproj/build/myproj.bin
[1/2] cd /home/axel/esp/esp-idf/components/esptool_py && /usr/local/bin/cmake -D IDF_PATH=/home/axel/esp...esp-idf-dev/apps/myproj/build -P /home/axel/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32c3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x110000 myproj.bin
esptool.py v3.3.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:a4:93:28
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00110000 to 0x00214fff...
Erasing flash...
Took 5.33s to erase flash block
Writing at 0x00110000... (0 %)
Writing at 0x00110400... (0 %)
Writing at 0x00110800... (0 %)
Writing at 0x00110c00... (0 %)
Writing at 0x00111000... (0 %)
Writing at 0x00111400... (0 %)
Writing at 0x00111800... (0 %)
Writing at 0x00111c00... (0 %)
Writing at 0x00112000... (0 %)
Writing at 0x00112400... (0 %)
Writing at 0x00112800... (1 %)
Writing at 0x00112c00... (1 %)
Writing at 0x00113000... (1 %)
Writing at 0x00113400... (1 %)
Writing at 0x00113800... (1 %)
Writing at 0x00113c00... (1 %)
Writing at 0x00114000... (1 %)
Writing at 0x00114400... (1 %)
Writing at 0x00114800... (1 %)
Writing at 0x00114c00... (1 %)
Writing at 0x00115000... (2 %)
Writing at 0x00115400... (2 %)
...
Writing at 0x0020f400... (98 %)
Writing at 0x0020f800... (98 %)
Writing at 0x0020fc00... (98 %)
Writing at 0x00210000... (98 %)
Writing at 0x00210400... (98 %)
Writing at 0x00210800... (98 %)
Writing at 0x00210c00... (98 %)
Writing at 0x00211000... (98 %)
Writing at 0x00211400... (98 %)
Writing at 0x00211800... (99 %)
Writing at 0x00211c00... (99 %)
Writing at 0x00212000... (99 %)
Writing at 0x00212400... (99 %)
Writing at 0x00212800... (99 %)
Writing at 0x00212c00... (99 %)
Writing at 0x00213000... (99 %)
Writing at 0x00213400... (99 %)
Writing at 0x00213800... (99 %)
Writing at 0x00213c00... (99 %)
Writing at 0x00214000... (100 %)
Wrote 1065984 bytes at 0x00110000 in 16.0 seconds (533.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Running idf_monitor in directory /home/axel/esp/esp-idf-dev/apps/myproj
Executing "/home/axel/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/axel/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /home/axel/esp/esp-idf-dev/apps/myproj/build/myproj.elf -m '/home/axel/.espressif/python_env/idf5.0_py3.10_env/bin/python' '/home/axel/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c728
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
....
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header:ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c0d4
...

$ idf.py erase-flash
Executing action: erase-flash
Serial port /dev/ttyACM0
Connecting.....
Detecting chip type... ESP32-C3
Running esptool.py in directory /home/axel/esp/esp-idf-dev/apps/myproj/build
Executing "/home/axel/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/axel/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32c3 --no-stub erase_flash"...
esptool.py v3.3.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:a4:93:28
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-C3 ROM does not support function erase_flash.
esptool.py failed with exit code 2

dobairoland, 2022-05-23:
@AxelLin Did you flash the bootloader? `invalid header: 0xffffffff` looks to me that the flash is already/still erased. Please provide the output for `idf.py flash monitor` instead.

BTW, esptool's flasher stub is disabled for ESP32-C3 at the moment. You can see the `--no-stub` arguments in every command. `erase-flash` is not working because of this. You could erase the flash if you connect through UART.

AxelLin, 2022-05-23:
> BTW, esptool's flasher stub is disabled for ESP32-C3 at the moment. You can see the `--no-stub` arguments in every command. `erase-flash` is not working because of this. You could erase the flash if you connect through UART.

Oops, we have set CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y && CONFIG_ESP_CONSOLE_SECONDARY_NONE=y in myproj.
I thought we can use the uart for other purpose.
Did not know it can cuase problem for erase flash because it's fine to download.
Do you have target timeline to support erase-flash?

AxelLin, 2022-05-23:
> @AxelLin Did you flash the bootloader? `invalid header: 0xffffffff` looks to me that the flash is already/still erased. Please provide the output for `idf.py flash monitor` instead.

Now I try build example code (wifi/getting_started/station) and it seems fine.
Maybe it's something wrong in myproj, I will update latter.

dobairoland, 2022-05-23:
> Do you have target timeline to support erase-flash?

No, I'm not aware of any specific deadline for this one. I'm sorry.

AxelLin, 2022-05-24:
@dobairoland 

1. When downloading firmware, it shows:
Flash will be erased from 0x00110000 to 0x00214fff...
Erasing flash...
Took 5.33s to erase flash block
So it can erase flash. but "idf erase-flash" does not work?

2. Is there any other issues without uart?

AxelLin, 2022-05-24:
> @AxelLin Did you flash the bootloader? `invalid header: 0xffffffff` looks to me that the flash is already/still erased. 
Confirm it's because I forgot "flash" before "app-flash".

closed 2022-05-24 by AxelLin

AxelLin, 2022-05-24:
> > Do you have target timeline to support erase-flash?
> 
> No, I'm not aware of any specific deadline for this one. I'm sorry.

Below command works for me:
$ esptool.py -p /dev/ttyACM0 --chip esp32c3 erase_flash
esptool.py v3.3.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:a3:3b:00
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.5s
Hard resetting via RTS pin...

GyroGearl00se, 2024-01-05:
Hey Folks...
I stumbled upon the same issue trying to flash on a "ESP32-C3 Super Mini" -> invalid header: 0xffffffff

I ended up with following steps to get it working:

- ESPHome > Install > Manual download > Modern format
- Use "flash download tool" from Espressif: https://www.espressif.com/en/support/download/other-tools
- Choose as Chiptype: ESP32-C3; work Mode: develop; load Mode: USB
- load you previously exported *-factory.bin into the first field, tick the box and write it to "0x0"
- SPI SPEED: 40MHz, SPI MODE: DIO, Make sure "DoNotChgBin" is ticked.
- Press and Hold the "Boot" button on your device and connect USB cable to your computer. Let go of of the button again :)
- Configure your correct COM-Port. Baudrate: 115200
- Start
- Manually press the reset button on your device
- done! The device appeared as online in my ESPHome Dashboard. From now on everything else can be done wireless.

Hope this helps

SirBramble, 2024-01-17:
@GyroGearl00se Thank you!

Worked for me with one exception. When I connected the Board (GPIO9/Boot high), it kept disconnecting. So I assumed I had to pull GPIO9 low, to get it in to boot loader mode.  This worked to an extent. I managed to communicate with it, but nothing I tried to program it would work.
So I just left GPIO9 high, pressed ERASE and it worked. Got the Firmware running and it showed up in Home Assistant.

Woodpeckercz, 2024-01-18:
> Hey Folks... I stumbled upon the same issue trying to flash on a "ESP32-C3 Super Mini" -> invalid header: 0xffffffff
>
> I ended up with following steps to get it working:
>
> * ESPHome > Install > Manual [...]

Had my Seeeduino ESP32C3 stuck in bootloop. This helped. Thanks

SirBramble, 2024-01-23:
> The same problem with wavshare ESP32 S3 these solutions don't work. The port keeps disconnecting over a second and shows an invalid header: 0xffffffff

Same for me, but I managed to brute force it by pressing ERASE, just after the 'USB disconnect' sound came up. Maybe I got lucky and managed to get just the right time.

Might also be worth mentioning, that I had the firmware selected with start address 0x00 when I pressed ERASE.

Maybe try toggling RESET just before trying to upload/erase.

Trinnss, 2024-01-23:
> Same for me, but I managed to brute force it by pressing ERASE, just after the 'USB disconnect' sound came up. Maybe I got lucky and managed to get just the right time.
>
> Might also be worth [...]

Can u explain a bit more
And what board are u using?

SirBramble, 2024-01-23:
> Can u explain a bit more And what board are u using?

Its a DIY board with a ESP32-C3FH4 (ESP32 C3 with internal Flash). Files are here: [https://github.com/SirBramble/The-Luwu](https://github.com/SirBramble/The-Luwu)

I used the "flash download tool" from espressif ([https://www.espressif.com/en/support/download/other-tools](https://www.espressif.com/en/support/download/other-tools)) and plugged the board in. No bootloader button or anything.
Then I loaded in the Firmware I wanted to flash (generated by Home Assistant) and set the start address to 0x00.
[image]
The board kept disconnecting every second or so (at least Windows told me it did) but after finding and setting the correct Serial Port, I pressed ERASE and it worked. After that I flashed the Firmware with the START button.

The Reason I didn't set the ESP in to BOOT mode with GPIO8 is that the ERASE and START failed. It gave an error telling me It couldn't connect to the ESP.

Trinnss, 2024-01-24:
Thank You

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
01/25/24,
12:03:40 AM

faizannazir, 2024-03-09:
```
E:\Esp_idf_Project\hello_world>idf.py -p COM9 monitor
Executing action: monitor
Running idf_monitor in directory E:\Esp_idf_Project\hello_world
Executing "E:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe E:\Espressif\frameworks\esp-idf-v5.0.5\tools/idf_monitor.py -p COM9 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace E:\Esp_idf_Project\hello_world\build\hello_world.elf --force-color -m 'E:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe' 'E:\Espressif\frameworks\esp-idf-v5.0.5\tools\idf.py' '-p' 'COM9'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x5 (DOWNLOAD(USB/UART0/1))
Saved PC:0x4004d190
waiting for download

```

```

E:\Esp_idf_Project\hello_world>idf.py flash monitor
Executing action: flash
Serial port COM9
Connecting...
Detecting chip type... ESP32-C3
Running ninja in directory E:\Esp_idf_Project\hello_world\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D E:\Esp_idf_Project\hello_world\build\esp-idf\esptool_py && E:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe E:/Espressif/frameworks/esp-idf-v5.0.5/components/partition_table/check_sizes.py --offset 0x8000 partition --type app E:/Esp_idf_Project/hello_world/build/partition_table/partition-table.bin E:/Esp_idf_Project/hello_world/build/hello_world.bin"
hello_world.bin binary size 0x27290 bytes. Smallest app partition is 0x100000 bytes. 0xd8d70 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D E:\Esp_idf_Project\hello_world\build\bootloader\esp-idf\esptool_py && E:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe E:/Espressif/frameworks/esp-idf-v5.0.5/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 E:/Esp_idf_Project/hello_world/build/bootloader/bootloader.bin"
Bootloader binary size 0x4f00 bytes. 0x3100 bytes (38%) free.
[2/3] cmd.exe /C "cd /D E:\Espressif\frameworks\esp-idf-v5.0.5\components\esptool_py && E:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=E:/Espressif/frameworks/esp-idf-v5.0.5 -D SERIAL_TOOL=E:/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Espressif/frameworks/esp-idf-v5.0.5/components/esptool_py/esptool/esptool.py;--chip;esp32c3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=E:/Esp_idf_Project/hello_world/build -P E:/Espressif/frameworks/esp-idf-v5.0.5/components/esptool_py/run_serial_tool.cmake"
esptool.py --chip esp32c3 -p COM9 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.0
Serial port COM9
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x00037fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20224 bytes to 12410...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
CMake Error at run_serial_tool.cmake:58 (message):

  E:/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Espressif/frameworks/esp-idf-v5.0.5/components/esptool_py/esptool/esptool.py;--chip;esp32c3
  failed

FAILED: CMakeFiles/flash E:/Esp_idf_Project/hello_world/build/CMakeFiles/flash
cmd.exe /C "cd /D E:\Espressif\frameworks\esp-idf-v5.0.5\components\esptool_py && E:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=E:/Espressif/frameworks/esp-idf-v5.0.5 -D SERIAL_TOOL=E:/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Espressif/frameworks/esp-idf-v5.0.5/components/esptool_py/esptool/esptool.py;--chip;esp32c3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=E:/Esp_idf_Project/hello_world/build -P E:/Espressif/frameworks/esp-idf-v5.0.5/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:\Esp_idf_Project\hello_world\build\log\idf_py_stderr_output_16204 and E:\Esp_idf_Project\hello_world\build\log\idf_py_stdout_output_16204

```

```

Serial port COM4
Connecting...
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
[... 34 lines cut ...]
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000b2fff...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=4MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 731840 bytes to 446447...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)

C:\Users\HP840G~1\AppData\Local\Temp\arduino\sketches\C5865DAE1F5FE453D8888058EC229443>esptool.py -p COM9 --chip esp32c3 erase_flash
esptool.py v4.5.1
Serial port COM9
Connecting...
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)`

faizannazir, 2024-03-09:
```

C:\Users\HP840G~1\AppData\Local\Temp\arduino\sketches\C5865DAE1F5FE453D8888058EC229443>esptool.py flash_id
esptool.py v4.5.1
Found 2 serial ports
Serial port COM9
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Uploading stub...
Running stub...
Stub running...
Manufacturer: 3f
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...

C:\Users\HP840G~1\AppData\Local\Temp\arduino\sketches\C5865DAE1F5FE453D8888058EC229443>esptool.py chip_id
esptool.py v4.5.1
Found 2 serial ports
Serial port COM9
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: dc:da:0c:8e:a0:bc
Hard resetting via RTS pin...
```

faizannazir, 2024-03-09:
> > > > The same problem with wavshare ESP32 S3 these solutions don't work. The port keeps disconnecting over a second

[image]

byronin, 2024-03-30:
I experienced the same problem as you and managed to solve the issue I had. Actually, the problem was hardware-related. In my own PCB design, I had an opamp connected to the GPIO2 pin of the ESP32, and the issue was resolved once I removed it. I'm not exactly sure which pins are sensitive to pull-up or pull-down, but a similar problem existed with the ESP8266 as well. Of course, the problem I experienced might not be the same as yours. Best of luck.

faizannazir, 2024-03-30:
Try esptool.py flash_id

zizebra, 2024-04-12:
i have been battling with as well. It definitely appears to be a hardware problem. Mine sometimes shows memory as 2MB instead of 4MB.I am beginning to wonder if the crappy flashdisk size problems are part of this. I have 3 or 4 devices that shows this prblem. All the devices were a pain to program right from the start. Other devices you get them, you flash them and you are done.

anoblet, 2024-07-09:
I'm able to erase/flash, though I get a boot loop when I reset. Any suggestions?

colin-purcell, 2025-03-01:
PSA: **Some of these modules are not flashable due to efuses** if you're (like me) getting this error on a unit that's in/from an existing product. I just spent a bunch of time diving into this, so hopefully someone else sees this before they waste their own, especially because it looks like  handful of people are still having issues.

I was trying to flash a Wiz 9290041268 (smart LED light strip/RGB LED strip 15M) which has an esp32-c3-mini-1
In this Wiz product, they've burned an encryption efuse "SPI_BOOT_CRYPT_CNT", and despite that fuse being disable-able if only 1 of the 3 efuses is burned (like mine), writes can be disabled on the efuses (which they were). All of this leads to the following errors with the above linked esp flash tool (which at least was more clear than the 'invalid header' errors I had on boot after trying flashing with other tools):
"14-already flash encrypt or secure boot"
"Please confirm whether the device to be download has enabled flash encryption or secure boot"
"ESP32 efuse secure check fail:Already flash encryption,NOT allow download again!!!"
"ESP32C3 Secure Encrypt check error esp_write_flash"

**If you want to verify the status of efuses** to check if your issue is the same, the Espressif "flash download tool" mentioned above has a "Read Efuse" button in the "chipInfoDump" tab. This dumps the values to a file. You'll have to read the [documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/system/efuse.html) to see if any efuses look problematic. Here is what I found in my own dump for the efuses related to my error: 
SPI_BOOT_CRYPT_CNT ... =  Enable R/- (0b001)
SECURE_BOOT_EN ... = False R/W (0b0)

SECURE_BOOT_EN is set to what I might hope (disabled), but because writes have been disabled on SPI_BOOT_CRYPT_CNT (evidenced by "R/-") I can't burn the next efuse which would set the value to "0b011" which would result in encryption being turned off.
If you're slightly luckier than me and encryption is on but writes are not disabled, you might be able to `pip install esptool` and run `espefuse.py --port /dev/ttyUSB0 burn_efuse SPI_BOOT_CRYPT_CNT` (substituting for your own device)

Because writes are disabled on mine, it does this:
`espefuse.py v4.8.1
Connecting...
Detecting chip type... ESP32-C3

=== Run "burn_efuse" command ===
The efuses to burn:
  from BLOCK0
     - SPI_BOOT_CRYPT_CNT

Burning efuses:

    - 'SPI_BOOT_CRYPT_CNT' (Enables flash encryption when 1 or 3 bits are set and disables otherwise) 0b001 -> 0b011

A fatal error occurred: 	SPI_BOOT_CRYPT_CNT is write-protected. Burn is not possible.`

**Be careful when burning your efuses; burning can not be reversed and may result in an unrecoverable brick of the microcontroller**

My only recourse now seems to be buying and soldering a fresh one that isn't locked-out of flashing like mine... which if I'm honest, I wasn't aware was an option for the esp chips. According to their documentation it's a mature and well supported feature if a vendor cares about it.

favnec5, 2025-05-19:
ok, fixed !

esptool.exe --chip esp32p4 --port COM%com% --baud 46080 write_flash -z 0x002000 firmware.bin

[image]