espressif/esp-idf #9006: LAN8720 no output RMII on gpio 0 and 17

tsctrl, 2022-05-23:
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC]
- Kit version (for WroverKit/DevKitC): v3
- Module or chip used:  [ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // 4.4.1
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
I am unable to get 50MHz  osci signal from gpio0 and gpio17
i can confirm this is working on idf 4.3

[image]

### Expected Behavior

### Actual Behavior
osci display 584mV 0Hz

### Steps to reproduce
hook osci to pin 0 or 17 with RMII output enable

double check my pin connection is correct. anyone can confirm if the RMII working in 4.4.1?
### Code to reproduce this issue

```cpp
res =  ETH.begin(1, -1, 23, 18, ETH_PHY_LAN8720);
```

## Debug Logs

```
D: ETH pins -1, 23 and 18
E (23175) esp.emac: emac_esp32_init(349): reset timeout
E (23176) esp_eth: esp_eth_driver_install(214): init mac failed
[ 20027][E][ETH.cpp:320] begin(): esp_eth_driver_install failed
```

## Other items if possible

kostaond, 2022-05-23:
@tsctrl do you use ESP-IDF as such or Arduino? It seems like you initialize the Ethernet in Arduino style, but configuring it via Menuconfig.

tsctrl, 2022-05-23:
@kostaond, idf with arduino as component. yes probably this issue should be on arduino 

ETH.begin(1, -1, 23, 18, ETH_PHY_LAN8720, ETH_CLOCK_GPIO0_OUT, false);

i change to above init will output 50Mhz but not able go get IP only getting event SYSTEM_EVENT_STA_BEACON_TIMEOUT and followed by SYSTEM_EVENT_AP_STADISCONNECTED event.

1. the menuconfig now seems has no effect.
2. i am not getting ip from the dns

kostaond, 2022-05-23:
It is strange if it works with v4.3 because there is no option to configure EMAC clock during run-time in that version. Configuring EMAC clock was possible only during compile time and hence was not available in Arduino since it uses pre-compiled ESP-IDF.

tsctrl, 2022-05-24:
issue with crystal grounding. close this as issue has been resolve.

closed 2022-05-24 by tsctrl