espressif/esp-idf #9011: External PSRAM | Place task stacks into external memory

DheeranYuvaraj, 2022-05-24:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version : v4.41
- Build System:  CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Issue is, we have maxed out of internal RAM in one of the project and forced to use external PSRAM. We found few options to initialize and use PSRAM as a heap, however we would like to configure and use PSRAM for Stack and Static as well. This will help us to integrate the existing code base seamlessly without the pain of modifying and testing the modules.
We request you to help us to find a solution for this problem.

**As ESP_Sprite said i am reporting this Bug as reference to this topic posted on esp32.com**: [https://www.esp32.com/viewtopic.php?f=2&t=27797](url)

## Excepted behaviour

The option [CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-stack-external-memory) can be used to place task stacks into external memory. In these cases [xTaskCreateStatic()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv417xTaskCreateStatic14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC11StackType_tPC12StaticTask_t) must be used to specify a task stack buffer allocated from external memory, otherwise task stacks will still be allocated from internal memory. This is mentioned in idf document, but not working.

igrr, 2022-05-24:
@DheeranYuvaraj could you please describe the behavior you observe in more details? If the application crashes, please post the log output. Please attach your sdkconfig file as well. If you can narrow down the issue to a particular piece of code in your program, please post that piece of code. Thank you!

DheeranYuvaraj, 2022-05-24:
@igrr Thanks for the reply. 

Yes, the application crashes. Please check the below log. 

Code and Sdkconfig files you can download here: [Files.zip](https://github.com/espressif/esp-idf/files/8761522/Files.zip)

**Log output**

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6624
load:0x40078000,len:14780
load:0x40080400,len:3792
entry 0x40080694
[1B][0;32mI (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader[1B][0m
[1B][0;32mI (27) boot: compile time 14:51:48[1B][0m
[1B][0;32mI (27) boot: chip revision: 1[1B][0m
[1B][0;32mI (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0[1B][0m
[1B][0;32mI (37) boot.esp32: SPI Speed      : 40MHz[1B][0m
[1B][0;32mI (42) boot.esp32: SPI Mode       : DIO[1B][0m
[1B][0;32mI (46) boot.esp32: SPI Flash Size : 2MB[1B][0m
[1B][0;32mI (51) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (56) boot: Partition Table:[1B][0m
[1B][0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (82) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (89) boot: End of partition table[1B][0m
[1B][0;32mI (94) boot_comm: chip revision: 1, min. application chip revision: 0[1B][0m
[1B][0;32mI (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08108h ( 33032) map[1B][0m
[1B][0;32mI (121) esp_image: segment 1: paddr=00018130 vaddr=3ffb0000 size=0234ch (  9036) load[1B][0m
[1B][0;32mI (125) esp_image: segment 2: paddr=0001a484 vaddr=40080000 size=05b94h ( 23444) load[1B][0m
[1B][0;32mI (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=170f4h ( 94452) map[1B][0m
[1B][0;32mI (172) esp_image: segment 4: paddr=0003711c vaddr=40085b94 size=05d0ch ( 23820) load[1B][0m
[1B][0;32mI (183) esp_image: segment 5: paddr=0003ce30 vaddr=50000000 size=00010h (    16) load[1B][0m
[1B][0;32mI (189) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (189) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (203) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (204) cpu_start: Starting app cpu, entry point is 0x4008100c[1B][0m
[1B][0;32mI (0) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (218) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (218) cpu_start: cpu freq: 160000000[1B][0m
[1B][0;32mI (218) cpu_start: Application information:[1B][0m
[1B][0;32mI (222) cpu_start: Project name:     main[1B][0m
[1B][0;32mI (227) cpu_start: App version:      v4.4-dirty[1B][0m
[1B][0;32mI (232) cpu_start: Compile time:     May 24 2022 14:51:06[1B][0m
[1B][0;32mI (238) cpu_start: ELF file SHA256:  540000de50352db0...[1B][0m
[1B][0;32mI (244) cpu_start: ESP-IDF:          v4.4-dirty[1B][0m
[1B][0;32mI (250) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (257) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (263) heap_init: At 3FFB2DC0 len 0002D240 (180 KiB): DRAM[1B][0m
[1B][0;32mI (269) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (275) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (282) heap_init: At 4008B8A0 len 00014760 (81 KiB): IRAM[1B][0m
[1B][0;32mI (289) spi_flash: detected chip: gd[1B][0m
[1B][0;32mI (292) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (296) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (310) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;32mI (375) Test: FREE HEAP: 292444[1B][0m
[1B][0;32mI (10375) Test: Reservated Stack from 0 to 2710[1B][0m

assert failed: xTaskCreateStaticPinnedToCore tasks.c:675 (xPortcheckValidStackMem(pxStackBuffer))

Backtrace:0x40081846:0x3ffb60f00x40085729:0x3ffb6110 0x4008a691:0x3ffb6130 0x40086c99:0x3ffb6250 0x400d4d7b:0x3ffb6290 0x400e6f30:0x3ffb62c0 0x4008807d:0x3ffb62e0 

ELF file SHA256: 540000de50352db0

Rebooting...`

DheeranYuvaraj, 2022-05-24:
@igrr Thanks for the reply. 

Please check this for log output, code, and sdkconfig. 
[Files.zip](https://github.com/espressif/esp-idf/files/8761925/Files.zip)

igrr, 2022-05-24:
@DheeranYuvaraj from your sdkconfig file:
```
# CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY is not set
```
Which means that `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` is not set, so the stack in external memory is rejected:
```
assert failed: xTaskCreateStaticPinnedToCore tasks.c:675 (xPortcheckValidStackMem(pxStackBuffer))
```
Please try enabling `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY`.

DheeranYuvaraj, 2022-05-25:
@igrr Thanks for the information. After enabling  `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY.` I got the below log output. Can u confirm for my task stack is allocated between `3f80086c to 3f818f0c `in PSRAM correct? 

**Log output:**

> [1B][0;32mI (228) psram: This chip is ESP32-D0WD[1B][0m
[1B][0;32mI (230) spiram: Found 64MBit SPI RAM device[1B][0m
[1B][0;32mI (230) spiram: SPI RAM mode: flash 40m sram 40m[1B][0m
[1B][0;32mI (233) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[1B][0m
[1B][0;32mI (241) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (244) cpu_start: Starting app cpu, entry point is 0x40081208[1B][0m
[1B][0;32mI (0) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (1136) spiram: SPI SRAM memory test OK[1B][0m
[1B][0;32mI (1144) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (1144) cpu_start: cpu freq: 160000000[1B][0m
[1B][0;32mI (1144) cpu_start: Application information:[1B][0m
[1B][0;32mI (1147) cpu_start: Project name:     main[1B][0m
[1B][0;32mI (1152) cpu_start: App version:      v4.4-dirty[1B][0m
[1B][0;32mI (1157) cpu_start: Compile time:     May 24 2022 17:15:58[1B][0m
[1B][0;32mI (1163) cpu_start: ELF file SHA256:  a02129cce4a41f48...[1B][0m
[1B][0;32mI (1169) cpu_start: ESP-IDF:          v4.4-dirty[1B][0m
[1B][0;32mI (1175) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (1182) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (1188) heap_init: At 3FFB2E80 len 0002D180 (180 KiB): DRAM[1B][0m
[1B][0;32mI (1194) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (1201) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (1207) heap_init: At 4008F73C len 000108C4 (66 KiB): IRAM[1B][0m
[1B][0;32mI (1214) spiram: Adding pool of 4095K of external SPI memory to heap allocator[1B][0m
[1B][0;32mI (1223) spi_flash: detected chip: gd[1B][0m
[1B][0;32mI (1226) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (1230) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (1245) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;32mI (1254) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[1B][0m
[1B][0;32mI (1294) Test: FREE HEAP: 4449443[1B][0m
[1B][0;32mI (6304) Test: Reservated Stack from 3f80086c to 3f818f0c[1B][0m

DheeranYuvaraj, 2022-05-26:
@igrr Some more information. 

`[1B][0;32mI (1294) Test: FREE HEAP: 4449443[1B][0m`
`[1B][0;32mI (1294) Test: Reservated Stack from 3f80086c to 3f83d8fc[1B][0m`
`[1B][0;32mI (1334) Test: FREE HEAP: 4199359[1B][0m`

This is my final log output. Before allocating PSRAM size 4449443 Bytes. After allocating it changed to 4199359 Bytes. So a total of 250084 Bytes was allocated in PSRAM for my task. Means 250.084KB.

igrr, 2022-05-26:
I've added the second "FREE HEAP" log line to the code you have shared, but wasn't able to reproduce the issue. The behavior looks as expected:

<details><summary>code:</summary>

```cpp
    size_t heap_before = esp_get_free_heap_size();
    ESP_LOGI(TAG, "FREE HEAP: %d", heap_before);

    xStack = heap_caps_calloc(1, sizeof(StackType_t)*STACK_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT);
    ESP_LOGI(TAG, "Reserved Stack from %x to %x", (uint)xStack, ((uint)xStack) + STACK_SIZE);
    
    size_t heap_after = esp_get_free_heap_size();
    ESP_LOGI(TAG, "FREE HEAP: %d, delta: %d", heap_after, heap_before - heap_after);
    
    xTaskCreateStatic(test_task, "test", sizeof(StackType_t)*STACK_SIZE, NULL, 6, xStack, &xTaskBuffer);
```
</details>

<details><summary>output:</summary>

```cpp
I (1290) Test: FREE HEAP: 4450575
I (1290) Test: Reserved Stack from 3f80086c to 3f802f7c
I (1290) Test: FREE HEAP: 4440483, delta: 10092
```
</details>

The stack size you have requested was 10000 bytes, so the delta looks about right.

If you aren't getting the same result, please post the updated code to reproduce the issue.

DheeranYuvaraj, 2022-05-26:
@iggr in my new code i have requested 250000 bytes, so i got a delta value 250084. 

As you said i am getting the same result. 

Thanks for guiding and making it work. 

Closing now.

closed 2022-05-26 by DheeranYuvaraj

ftab, 2023-12-01:
@igrr I am trying to enable CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY but I cannot without also selecting SPIRAM_USE_MALLOC. I run out of memory when I enable SPIRAM_USE_MALLOC because things then tend to prefer the internal SRAM, despite me turning on all the switches that say "allocate from PSRAM". Is there a way to allocate task stacks to external memory while leaving the allocation method set to heap_caps_malloc only?