espressif/esp-idf #9015: Core dump: Stack overflow in BTU_TASK

ztefanjo, 2022-05-25:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v4.4.1
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows

## Problem Description

We occasionally get core dumps indicating stack overflow in the BTU_TASK. In one occasion, we got the [attached dump](https://github.com/espressif/esp-idf/files/8768442/core-dump_51.1_Wifi-BLE-App_master_build-220519_194119_220524-162701.elf.txt) that gives some more information. This issue has previously been mentioned in [issue 5371](https://github.com/espressif/esp-idf/issues/5371), but as that issue does not describe the recursive calls in its original description, I am also creating this new issue. 

### Expected Behavior
No core dumps.

### Actual Behavior
Core dump: [core-dump_51.1_Wifi-BLE-App_master_build-220519_194119_220524-162701.elf.txt](https://github.com/espressif/esp-idf/files/8768442/core-dump_51.1_Wifi-BLE-App_master_build-220519_194119_220524-162701.elf.txt)
Never-ending call chain:
```
#60 0x4204fde4 in bta_gattc_sm_execute (p_clcb=0x3de0345c, event=<optimized out>, p_data=0x3de13de8) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:298
#61 0x4205e307 in bta_gattc_pop_command_to_send (p_clcb=0x3de0345c) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1575
#62 0x4205f3c2 in bta_gattc_write (p_clcb=0x3de0345c, p_data=<optimized out>) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1226
#63 0x4204fde4 in bta_gattc_sm_execute (p_clcb=0x3de0345c, event=<optimized out>, p_data=0x3de13de8) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:298
#64 0x4205e307 in bta_gattc_pop_command_to_send (p_clcb=0x3de0345c) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1575
#65 0x4205f674 in bta_gattc_write_cmpl (p_clcb=0x3de0345c, p_data=<optimized out>) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1358
#66 0x4205f89a in bta_gattc_op_cmpl (p_clcb=0x3de0345c, p_data=0x3de13a54) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1479
#67 0x4204fde4 in bta_gattc_sm_execute (p_clcb=0x3de0345c, event=<optimized out>, p_data=0x3de13a54) at D:/DevOps/agt19/_work/mower/Ext/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:298
```

### Steps to reproduce

Uncertain at this point, as this does not occur every time and the core dump has been harvested from our "test fleet" after it occurred.
[Issue 5371](https://github.com/espressif/esp-idf/issues/5371) tells that it happens when calling esp_ble_gattc_write_char_descr after ESP_GATTC_REG_FOR_NOTIFY_EVT, which we indeed are doing without delay (but from other task than the event callback).

### Work-around

There is a work-around mentioned in [issue 5371](https://github.com/espressif/esp-idf/issues/5371). To be checked if that helps in our case as well. Is there any better work-around recommended? 2 seconds is a long time to wait, do you have another recommendation?

ztefanjo, 2022-06-29:
Update:
Unfortunately, the work-around mentioned in [issue 5371](https://github.com/espressif/esp-idf/issues/5371) does not seem to help. We sporadically get some variants of this core dump, either with bta_gattc_read or bta_gattc_write in the call-chain.
Interestingly, the BTC_TASK seems to be calling btu_task_post when this error happens. This might very well be the trigger. See attached dumps. This is one of the most frequent core dumps that we get from our test fleet, so a fix or work-around would be appreciated.

[core-dump_220622-132340_51.2_Wifi-BLE-Dongle-App_H994143_220620-092040.elf.txt](https://github.com/espressif/esp-idf/files/9009002/core-dump_220622-132340_51.2_Wifi-BLE-Dongle-App_H994143_220620-092040.elf.txt)
[core-dump_51.1_Wifi-BLE-App_24.26_220627-140519.elf.txt](https://github.com/espressif/esp-idf/files/9009003/core-dump_51.1_Wifi-BLE-App_24.26_220627-140519.elf.txt)

ztefanjo, 2022-09-15:
@Weijian-Espressif : In our case, we already await the event as you suggest. But other things might happen in parallel for other simultaneous connections. Thereby my previous question.

Weijian-Espressif, 2022-09-15:
@ztefanjo  Please help to provide detailed test log and test steps. It would be better if you provide reproducible code, which will help us to quickly locate the problem

ztefanjo, 2022-09-16:
@Weijian-Espressif :
As I wrote in the top, these dumps are harvested from our test-fleet hours after the error actually occurred, so it is unfortunately very hard for us to know exactly how to reproduce. I am however attaching a couple of more coredumps to this issue, if it could help. Perhaps they are due to the same error or related to other errors, but they are all BLE-related. So if there is a rule that one should not do multiple requests to esp_ble_* simultaneously, that could perhaps explain some of these dumps.
[BLE-related coredumps.zip](https://github.com/espressif/esp-idf/files/9583767/BLE-related.coredumps.zip)
Interestingly, two of the attached dumps are happening directly after esp_pm_lock_acquire has returned (we call it at UART activity and release it on a timer to prevent light sleep during UART communication). I don't know if this is a coincidence or if it could be related to the coredump.

ztefanjo, 2022-09-19:
@Weijian-Espressif :
I wrote another issue related to the parallel execution of BLE (here disconnect in parallel with write): https://github.com/espressif/esp-idf/issues/9808. Probably not the same as this issue, but as said, also perhaps related to parallel execution of BLE. 
So the question from before stands:
_Is it allowed to do e.g. client and server requests in parallel? Is it allowed to do requests for different connections in parallel? Or is the IDF limited to one esp_ble call at a time?_

Weijian-Espressif, 2022-09-19:
@ztefanjo  I am fixing the bug of btu_task overflow, may need your help to test the fixed code later.

ztefanjo, 2022-09-19:
@Weijian-Espressif : Thanks, good to hear! If you supply a patch on v4.4.2, we can deploy it in our test fleet and let it run for a while.

Weijian-Espressif, 2022-09-19:
`is it allowed to do e.g. client and server requests in parallel? `
yes

`Is it allowed to do requests for different connections in parallel?`

yes

What I mean is that for the operation of the same connection, it is recommended to wait for the callback after the api call before calling the next one

Weijian-Espressif, 2022-09-20:
[bugfix.patch.zip](https://github.com/espressif/esp-idf/files/9603852/bugfix.patch.zip)
@ztefanjo Please help to test the patch in the attachment, I tried to reproduce this problem according to the coredump file, but it failed, so I didn't find the root cause, I can only guess where it might be based on the code

Weijian-Espressif, 2022-09-29:
@ztefanjo  Is there any update?

ztefanjo, 2022-09-29:
@Weijian-Espressif : We have integrated the patch, but it takes a while before all devices are updated. I will keep you informed :-)

ztefanjo, 2022-09-30:
@Weijian-Espressif : Unfortunately, it seems as this is happening also after applying the patch, see attached dump.
There is a small possibility that the crash happened on a previous SW and was read out from the device after update, but the analyzed core dump looks healthy, so that is an unlikely explanation.

[core-dump_51.1_Wifi-BLE-App_43.1_220929-171546.elf.txt](https://github.com/espressif/esp-idf/files/9682539/core-dump_51.1_Wifi-BLE-App_43.1_220929-171546.elf.txt)

I will continue to keep you updated.

Weijian-Espressif, 2022-09-30:
Another question, for the same connection, when using `read_char` or `read_descr`, you need to wait for the read event to come up before calling the next read. Is your code written like this?

ztefanjo, 2022-09-30:
@Weijian-Espressif : This is tested on the test fleet during normal usage. It is very hard to say exactly what happened at the time of the Core dump. But devices are typically both sending and receiving, potentially on multiple connections at the same time.

read_char/read_descr is only used directly after connection to read the name of the peer. So I can't see in the code that we call these multiple times **for the same connection** before we receive the read event. It can however happen that they are called in parallel when having several connections. But that should be OK if I understood you right.

There may however be other actions happening in parallel with a read for the same channel. That is something that we are now trying to solve.

Will you create a new patch to handle also the read case?

ztefanjo, 2022-09-30:
@Weijian-Espressif : Do you think it would make any difference if we changed the priority of the FreeRTOS timer task (Tmr Svc) to a prio higher than btController? I can see in the dump that it has just dispatched a timer event when the error happens.

Or another theory: I see that in the client case, we are reading device name triggered by ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT **or if esp_ble_gap_update_conn_params times out after three seconds (no _EVT received)**. Could this explain the error? We should perhaps close the connection on timeout instead? Or use a longer timeout?

ztefanjo, 2022-10-03:
@Weijian-Espressif : We have not got any response on https://github.com/espressif/esp-idf/issues/9808 yet, but based on this, do you still think it is safe to e.g. close one connection while e.g. reading/writing on another connection?

Weijian-Espressif, 2022-10-08:
What is the priority of all the tasks you create? @ztefanjo

Weijian-Espressif, 2022-10-08:
You should not modify the priority of the task in the IDF. The event of parameter update will also be triggered if it times out. You don't need to use a timer to monitor this event. If there is a lot of data queued, the parameter update may be delayed.

Weijian-Espressif, 2022-10-09:
[diff-2.patch.zip](https://github.com/espressif/esp-idf/files/9741128/diff-2.patch.zip)
Please help to test the patch in the attachment

Weijian-Espressif, 2022-10-09:
It's strange that QA also has similar case tests, but the problem you described did not occur. When using our API normally, the errors should not be triggered. I'm not sure if your code is different from ours. If you can provide test code , we will locate the problem faster.

ztefanjo, 2022-10-10:
@Weijian-Espressif : It is not that easy to provide test code as we do not know ourselves exactly what triggers the error. All I can say is that it happens more or less daily on our test fleet of ~30 devices that are running 24/7.
We are currently rolling out a change to do more things related to BLE sequentially (waiting for event responses). We will see if that improves the situation.

ztefanjo, 2022-10-10:
I will see to that your new patch is rolled out as well as quickly as possible.

ztefanjo, 2022-10-10:
@Weijian-Espressif :
Here is a complete list of the threads:
24 ipc1
24 ipc0
23 wifi
23 btController
22 esp_timer
22 hciT
20 sys_evt
20 BTU_TASK
19 BTC_TASK
18 tiT
10 Tmr Svc
5 UartEventTask
4 UdpRxTask
3 app_main
1 app_low
0 IDLE
0 IDLE

The main thread is returning after initialization and therefore not included in the list.

Alvin1Zhang, 2023-05-24:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/4a3319d381789a11f463d8062934e384006de97b, and already back ported to v5.0, v4.4 and v4.3. Feel free to reopen.

closed 2023-05-24 by Alvin1Zhang