espressif/esp-idf #9018: Instructions for musl

ksmq, 2022-05-25:
It's been a bit long since I first installed the toolchain, but I wanted to verify everything works as expected. If there's been an update affecting the issue, I'll be happy to look into it and update the information here accordingly.

The problem is that the xtensa toolchain needs qsort_r which is not provided by musl, and they don't plan to either as far as I can tell. See https://github.com/elogind/elogind/issues/83 for something similar. The fix is simply to compile this qsort_r separately then link it to a few of the binaries using patchelf:

```
cd ~/.compat
gcc -O2 -c -fPIC qsort_r_missing.c      
gcc -shared -fPIC qsort_r_missing.o -o qsort_r_missing.so    
    
cd ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin
patchelf --print-needed xtensa-esp32-elf-ld    
patchelf --print-rpath xtensa-esp32-elf-ld    
    
patchelf --add-needed qsort_r_missing.so xtensa-esp32-elf-ld    
patchelf --set-rpath /home/kdev/.compat xtensa-esp32-elf-ld    
    
patchelf --add-needed qsort_r_missing.so xtensa-esp32-elf-ld    
patchelf --set-rpath /home/kdev/.compat xtensa-esp32-elf-ld    
    
patchelf --add-needed qsort_r_missing.so xtensa-esp32-elf-objdump    
patchelf --set-rpath /home/kdev/.compat xtensa-esp32-elf-objdump  
```
(apologies for the specific paths)

Although I'm sure most people using musl know what they're doing, it might be helpful to add instructions somewhere "official" or as part of the install script even?

The .c file in question can be found online under a GPL licence - I've attached the one I used. I had to change the suffix to .txt because github :facepalm:, as always, please verify the file yourself before compiling/running anything. Hope this helps.

Cheers

[qsort_r_missing.txt](https://github.com/espressif/esp-idf/files/8769634/qsort_r_missing.txt)

igrr, 2022-05-25:
Hi @ksmq, it looks like musl-1.2.3 has recently added `qsort_r` support (https://musl.libc.org/releases.html). Would you consider upgrading musl as a solution?

ksmq, 2022-05-26:
Ah, good to know, thanks for pointing it out! That is a indeed the best solution, unfortunately it seems not all musl distros have switched to the 1.2 branch, but that's their own shortcoming :pensive: 

Cheers!

closed 2022-05-26 by ksmq