espressif/esp-idf #9019: Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

cyberumut, 2022-05-25:
my function falls into core panic why ? 
test.c
254:void testnt(void* self, int offset, int32_t val)
255:{
256:  *(int32_t*)(((uint8_t*)self) + offset) = val;
257:}

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40152899  PS      : 0x00060130  A0      : 0x800d4eb0  A1      : 0x3ffba940  
0x40152899: setInt at /home/rootx/esp/umut/main/test.c:256

A2      : 0x00000008  A3      : 0x00000004  A4      : 0x00000010  A5      : 0x00000001  
A6      : 0x00000004  A7      : 0x3ffb6ee8  A8      : 0x00000077  A9      : 0x3ffba920  
A10     : 0x3ffbabb4  A11     : 0x00000000  A12     : 0x3ffb6f14  A13     : 0x00000000  
A14     : 0x00000004  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x40089c08  LEND    : 0x40089c24  LCOUNT  : 0xffffffff  
0x40089c08: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x40089c24: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

igrr, 2022-05-25:
Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited, it seems that the `self` pointer is NULL, and the code is attempting to write to address 0x00000008 (`EXCVADDR` in the panic handler output.)

cyberumut, 2022-05-25:
> Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited, it seems that the `self` pointer is NULL, and the code is attempting to write to address 0x00000008 (`EXCVADDR` in the panic handler output.)

thank u for answer 
this function how to fix ?

igrr, 2022-05-25:
I think you need to make sure you don't call this function with `self` being a NULL pointer.

Dazza0, 2022-07-08:
Closing this issue as it seems like the error is caused by a NULL pointer. Please feel free to reopen if necessary.

closed 2022-07-08 by Dazza0