espressif/esp-idf #9022: a2dp_sink: Unable to connect from iOS disabling SSP

pad52, 2022-05-25:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4.1-185-gc5deb7a971
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Using a2dp_sink example, disabling CONFIG_BT_SSP in order to obtain Fixed PIN makes the ESP not connectable from iPhone 13

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. copy esp-idf/examples/bluetooth/bluedroid/classic_bt/a2dp_sink
2. idf.py menuconfig --> Disable Secure Simple Pairing
3. compile & Flash.
4. Try to connect with an iPhone with iOS

BetterJincheng, 2022-05-26:
@pad52 We have tested iPhone13 and iPhone8p and it just worked fine. Input "1234" and connect -> ok. Please check it again. 
[image]

pad52, 2022-05-26:
It was clearly a problem of my iPhone, I tested with another one and it works as expected. Thank you very much for your time and sorry for bothering.

closed 2022-05-26 by pad52

BetterJincheng, 2022-05-26:
@pad52 You are welcome. Turn to us any time.