espressif/esp-idf #9029: CMUX: Protocol mismatch: Missed trailing SOF, recovering

Ganesh-Pod-Point, 2022-05-26:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32E]
- IDF version (run ``git describe --tags`` to find it):
    v4.4
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [USB]

## Problem Description

I am using one of the example code to test the simple CMUX from this [link](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/simple_cmux_client)  and have not change any code from that example but i found an error
`"W (7666) CMUX: Protocol mismatch: Missed trailing SOF, recovering..."`

Can anyone explain how to fix this problem.

david-cermak, 2022-05-26:
Hi @Ganesh-Pod-Point 

This warning says that an unexpected characters or protocol appeared between the devkit and modems. This typically happens when we miss some characters from UART ISR (and when we're testing more intense communication, e.g. OTA). This is a warning, so the protocol should recover for OTA to continue and eventually succeed.
This could be avoided by placing the UART's ISR to RAM
```
UART_ISR_IN_IRAM=y
```

Ganesh-Pod-Point, 2022-05-26:
Hello @david-cermak 
I enabled to UART's ISR to RAM but still the same problem. I do not have problem testing the example `[pppos_client](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/pppos_client)`.

david-cermak, 2022-06-06:
@Ganesh-Pod-Point There's a bug in `esp_modem` CMUX protocol implementation (likely to be seen on devices that use longer payloads). The WIP fix is available here: https://github.com/espressif/esp-protocols/pull/53
What kind of device/modem are you using?

> I do not have problem testing the example [pppos_client]

That example doesn't use CMUX mode.

Ganesh-Pod-Point, 2022-06-06:
I am using the SIMCOM A7600E modem. No `pppos_client` does not have CMUX mode, just tested if my modem works properly.

Alvin1Zhang, 2022-07-11:
Thanks for reporting, feel free to reopen.

closed 2022-07-11 by Alvin1Zhang

DarshanDodal, 2022-08-18:
I am using it for connecting to AWS IoT using Mqtts. I am facing the same problem. It never recovers.