espressif/esp-idf #9032: Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

f-hoepfinger-hr-agrartechnik, 2022-05-26:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

f-hoepfinger-hr-agrartechnik, 2022-05-26:
all Debug Log: 

```
Executing action: monitor
Running idf_monitor in directory /home/franz/git/hr/2/Sample/bleprph_wifi_coex
Executing "/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/franz/esp-idf-v4.4/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 --print_filter  /home/franz/git/hr/2/Sample/bleprph_wifi_coex/build/bleprph_wifi_coex.elf -m '/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/franz/esp-idf-v4.4/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 15:01:49
I (27) boot: chip revision: 3
I (30) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00177000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18440h ( 99392) map
I (132) esp_image: segment 1: paddr=00028468 vaddr=3ffbdb60 size=05644h ( 22084) load
I (140) esp_image: segment 2: paddr=0002dab4 vaddr=40080000 size=02564h (  9572) load
I (145) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=9a42ch (631852) map
I (377) esp_image: segment 4: paddr=000ca454 vaddr=40082564 size=1bc38h (113720) load
I (424) esp_image: segment 5: paddr=000e6094 vaddr=50000000 size=00010h (    16) load
I (439) boot: Loaded app from partition at offset 0x10000
I (439) boot: Disabling RNG early entropy source...
I (450) psram: This chip is ESP32-D0WD
I (451) spiram: Found 64MBit SPI RAM device
I (451) spiram: SPI RAM mode: flash 40m sram 40m
I (456) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (463) cpu_start: Pro cpu up.
I (467) cpu_start: Starting app cpu, entry point is 0x40081348
0x40081348: call_start_cpu1 at /home/franz/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (1360) spiram: SPI SRAM memory test OK
I (1368) cpu_start: Pro cpu start user code
I (1369) cpu_start: cpu freq: 160000000
I (1369) cpu_start: Application information:
I (1372) cpu_start: Project name:     bleprph_wifi_coex
I (1377) cpu_start: App version:      a51a1fc-dirty
I (1383) cpu_start: Compile time:     May 26 2022 16:42:56
I (1389) cpu_start: ELF file SHA256:  52d248c444207d76...
I (1395) cpu_start: ESP-IDF:          v4.4-dirty
I (1401) heap_init: Initializing. RAM available for dynamic allocation:
I (1408) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1414) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1420) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1426) heap_init: At 3FFC8F78 len 00017088 (92 KiB): DRAM
I (1433) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1439) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1446) heap_init: At 4009E19C len 00001E64 (7 KiB): IRAM
I (1452) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1461) spi_flash: detected chip: generic
I (1465) spi_flash: flash io: dio
[... 316 lines cut ...]
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00177000
I (82) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18440h ( 99392) map
I (131) esp_image: segment 1: paddr=00028468 vaddr=3ffbdb60 size=05644h ( 22084) load
I (140) esp_image: segment 2: paddr=0002dab4 vaddr=40080000 size=02564h (  9572) load
I (144) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=9a42ch (631852) map
I (376) esp_image: segment 4: paddr=000ca454 vaddr=40082564 size=1bc38h (113720) load
I (423) esp_image: segment 5: paddr=000e6094 vaddr=50000000 size=00010h (    16) load
I (438) boot: Loaded app from partition at offset 0x10000
I (438) boot: Disabling RNG early entropy source...
I (450) psram: This chip is ESP32-D0WD
I (451) spiram: Found 64MBit SPI RAM device
I (451) spiram: SPI RAM mode: flash 40m sram 40m
I (455) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (463) cpu_start: Pro cpu up.
I (466) cpu_start: Starting app cpu, entry point is 0x40081348
0x40081348: call_start_cpu1 at /home/franz/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.

```

mahavirj, 2022-05-27:
@f-hoepfinger-hr-agrartechnik 

Could you please check by increasing `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to say 256? Default value for this config is 16384. If you keep value as 0 then it shall redirect all potential allocations with default capabilities to external SPIRAM and that can result in this issue. Of-course, relevant software layers should ensure that allocations that require internal memory must ensure appropriate capabilities while allocating the resource, but we can fix that once you confirm the problem.

JannesBrands, 2022-05-31:
After setting `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to 256, it seems to work normally as expected. 

We will take a closer look in our continues process.

Kind regards
Jannes

closed 2022-07-29 by espressif-bot