espressif/esp-idf #9034: use_global_ca_store does not work for MQTT client and OTA updates

kotyara12, 2022-05-26:
I am using multiple TLS connections with ESP32:
1. mqtt client
2. https get and post requests (sending data to third party services)
3. ota updates from my hosting

Almost all of these connections use the same root certificate - ISRG X1.
If I specify the same certificate in the field esp_http_client_config_t.cert_pem = ...
for all three connections, then everything works.

But! A little digression from the main topic
With a "bad" connection (when the ping is very large and the connection often "breaks"), memory leaks very often occur, namely when the connection is restored. Analysis of the backtrace on restart caused by a memory allocation failure showed that the failure occurs when allocating memory for a new TLS connection. Of course, this does not prove anything that there is a problem, but I have analyzed "my" code for 10 circles and do not see a problem yet. I strongly suspect that the problem is somewhere in mbedTLS.

But back to the main topic. At some point I decided to try using global_ca_store. After all, all connections use the same root certificate.
To my regret, the ESP-IDF documentation almost does not cover this point, it is completely unclear to me how to CORRECTLY use these functions. There are no examples either. Decided to act on a whim. What I've done:

At the beginning of the program I added the following code:
`bool initTlsGlobalCAStore()
{
  esp_err_t err = esp_tls_init_global_ca_store();
  if (err != ESP_OK) {
    rlog_e(logTAG, "Failed to init global CA store");
    return false;
  };

  err = esp_tls_set_global_ca_store((const unsigned char*)isrg_root_x1_pem_start, isrg_root_x1_pem_end-isrg_root_x1_pem_start);
  if (err != ESP_OK) {
    rlog_e(logTAG, "Failed to load certificate for ISRG X1");
    return false;
  };

  return true;
};
`

Then, before establishing each HTTPS or MQTT or OTA connection, I changed the settings:
`
  cfgHttp.transport_type = HTTP_TRANSPORT_OVER_SSL;
  // cfgHttp.cert_pem = (char*)isrg_x1_pem_start;
  cfgHttp.use_global_ca_store = true;
`

That is, I commented out the pointer to the certificate and allowed the use of the global store.
Am I acting right?

Now the crux of the problem.

1. There are no problems with "simple" HTTPS connections - everything works fine, data is sent as it should and where it should be.

2. OTA flatly refused to work. In the esp_https_ota_begin code, I saw a check:
`
    if (!ota_config->http_config->cert_pem) {
        ESP_LOGE(TAG, "Server certificate not found in esp_http_client config");
        *handle = NULL;
        return ESP_ERR_INVALID_ARG;
    }
`
It turns out that OTA can not use the global store in any way ?????

3. MQTT seems to start working, but when trying to connect, the mbedTLS library gives an error (unfortunately, I did not save it) and the connection is not established.

As luck would have it, I didn’t really check and at once lost control of 5 remote devices at once, which I can’t manage and update now. I'm tearing my hair out.

ESP-YJM, 2022-05-27:
@kotyara12 Could you tell me what esp-idf version and commit you used.

mahavirj, 2022-05-27:
@kotyara12 

Example of using global ca store:

https://github.com/espressif/esp-idf/blob/179ea878c747519138b40f387b80f30dd72097f0/examples/protocols/https_request/main/https_request_example_main.c#L187-L201

For this particular use-case, I would recommend that you try using certificate bundle instead. Please refer to its usage at:

https://github.com/espressif/esp-idf/blob/179ea878c747519138b40f387b80f30dd72097f0/examples/protocols/https_request/main/https_request_example_main.c#L166-L175

kotyara12, 2022-05-27:
@[ESP-YJM](https://github.com/ESP-YJM), https://github.com/espressif/esp-idf/issues/9034#issuecomment-1139218733

I use PlatformIO, it manages versions itself. If you believe the file c:\Users\kotyara12\.platformio\packages\framework-espidf\version.txt then 4.3.2

@[mahavirj](https://github.com/mahavirj), https://github.com/espressif/esp-idf/issues/9034#issuecomment-1139269156

> Example of using global ca store:

Is it necessary to add a certificate to the repository before directly accessing the site and delete it immediately after????
I assumed that if you add the certificates necessary to the store once when starting the microcontroller, and then just access this store every time you access the site.

> For this particular use-case, I would recommend that you try using certificate bundle instead. Please refer to its usage at:

Will this mechanism work for https ota and mqtt clent? The question is precisely this. For normal http requests, and so everything works

mahavirj, 2022-05-27:
> Is it necessary to add a certificate to the repository before directly accessing the site and delete it immediately after????
I assumed that if you add the certificates necessary to the store once when starting the microcontroller, and then just access this store every time you access the site.

No. Only requirement here is that global ca store is setup with API `esp_tls_set_global_ca_store` before starting any TLS connection. After that any TLS connection configured with `use_global_ca_store = true` shall use certs from same store.

> Will this mechanism work for https ota and mqtt clent?

Yes. But I will recommend using latest ESP-IDF v4.4.1 release here. Previous v4.3 release branch may need few patches to be cherry-picked for this. Here is example for OTA updates with certificate bundle:

https://github.com/espressif/esp-idf/blob/179ea878c747519138b40f387b80f30dd72097f0/examples/system/ota/simple_ota_example/main/simple_ota_example.c#L94-L96

kotyara12, 2022-05-27:
Thank you very much, I will test it

mahavirj, 2022-05-27:
>  OTA flatly refused to work. In the esp_https_ota_begin code, I saw a check:

Probably you need to cherry-pick fix from https://github.com/espressif/esp-idf/commit/48987ad9df359f27ec88df7b020c942d1b069ab2 on v4.3 branch for this (linking https://github.com/espressif/esp-idf/issues/5504)

kotyara12, 2022-05-27:
> But I will recommend using latest ESP-IDF v4.4.1 release here. Previous v4.3

Please tell me how to put version 4.4 in PlatformIO? I'm not very good at this, alas.
At the moment I only have version 4.3.0 available in the platform manager in PlatformIO home and I don't know how to download the newer one

mahavirj, 2022-05-27:
> Please tell me how to put version 4.4 in PlatformIO? I'm not very good at this, alas.

Sorry I am not sure about this. Please check on PlatformIO forum. Alternatively, you can use ESP-IDF documentation pointers for getting started.

kotyara12, 2022-05-27:
> or this particular use-case, I would recommend that you try using certificate bundle instead

THANK YOU SO MUCH FOR THE HINT!
I got acquainted with the documentation - bungle certificates is a great option. Alas, I didn't know about it. Now, I guess, it will not be necessary to worry about the expiration dates of root certificates.

closed 2022-05-27 by kotyara12

riwalker, 2025-02-06:
is there an example for MQTT with CONFIG_EXAMPLE_USE_CERT_BUNDLE ?
assume i add just the  intermediate crt ? or client crt ? or we need to add the root CA crt ?