espressif/esp-idf #9035: [Question] [Security] Can a simple buffer overflow attack decrypt all firmware with esp_partition_read()?

chipweinberger, 2022-05-26:
From the docs, any flash can be automatically decrypted on read when using `esp_partition_read()`.

Why should code be able to decrypt instruction code? Im not sure a reason for it. Data, yes. But instructions?

Ideally instruction decryption  would be done entirely in hardware with no ability to call decryption from software.

When I read this in the docs, my first thought was "ah, an attacker just needs to find a buffer overflow with `esp_partition_read()` to read the entire flash contents.

Are there any protections against this?

negativekelvin, 2022-05-26:
There is only one flash encryption engine so yes it is possible to dump the whole flash.  One example is you need to read code to calculate the hash of an app partition.  The checks in esp_partition_read do seem to be missing an explicit check on size argument.  If you can find a buffer overflow to overwrite the call stack you can probably do anything.

chipweinberger, 2022-05-26:
Thanks for the answer!

That's a pretty unfortunate limitation (and vulnerability...). 😔 

For most attackers is pretty easy to find some out of bounds writes, use after free, etc, bugs in a complex C application. And it only takes 1 bug, in any version of the app you ever shipped.

Allowing `esp_read_flash()` to decrypt instruction code just makes extracting app firmware too easy!

I'd gladly take the tradeoff of not being able to calculate app hashes (and just store a copy of the app hash in data, like we already have!).

Request - Can new revisions of the ESP32 please remove the ability to decrypt instructions from software?

Thanks for your consideration.

chipweinberger, 2022-05-26:
@igrr

negativekelvin, 2022-05-27:
There may be something to be done with the MPU especially in newer chips to provide more protection

igrr, 2022-05-27:
It is indeed possible to protect the code section from reading, allowing only instruction fetch (starting from esp32-s2.) However compilers also insert literal values (constants) into code sections, so in the process of normal code execution the CPU often has to read such data stored in the code sections. If the instruction set allowed us to place these constants "far away" from the code itself, it would have been possible to move the constants into a separate readable section, which would allow read-protecting the code section. There are some limits on how far this section can be (and hence on the size of code section), given the number of bits which can be used for PC-relative addressing.

Out of curiosity, do you know of an example of a system (embedded or otherwise) which has this kind of code readback protection?

chipweinberger, 2022-05-27:
That's great news that the hardware potentially already ready!!

What is preventing us from modifying the compiler? That seems like a pretty doable change.

> do you know of an example of a system (embedded or otherwise) which has this kind of code readback protection?

No, I'm not familiar with enough processors to say, but it's similar to the idea behind DEP (Data Execution Prevention) in modern OS's.

filzek, 2022-05-30:
Its impossible to protect any system that has instrutions to on the fly decrypt any part of flash content.

As an attack goes good, taking control of execution address could be possible and flasible, but, take in mind that you must know exact the part in memory where the flash read decrypt is allocated to invoke the function, so, its not so simple.

Also, a buffer overflow must be achieved to open a space to call some other function to act.

To clear the full understand the ESP-IDF decrypt is part of work process so, its impossiple to disable it, even if espressif just close to the public the use of such function it will remain used internaly. So, barely impossible to protect.

Even if this kind of function is used by hardware crypto calls it is possible to to a bufferunderan to move pointers set to clear the registry and open the hardware to be invoked to another part of the code and move the cache to be pointed to another pointer.

So, its an endeless discussion that takes nowhere!

chipweinberger, 2022-05-30:
Thanks for your input.

> Its impossible to protect any system that has instrutions to on the fly decrypt any part of flash content.

Ideally, these instructions should not exist in the first place. Firmware decryption would happen automatically in HW without software control.

> As an attack goes good, taking control of execution address could be possible and flashible, but, take in mind that you must know exact the part in memory where the flash read decrypt is allocated to invoke the function, so, its not so simple.

Most simply, an attacker can just try every address until it succeeds. If it crashes, just reboot.

> Also, a buffer overflow must be achieved to open a space to call some other function to act.

Something along those lines, yes. 

> To clear the full understand the ESP-IDF decrypt is part of work process so, its impossiple to disable it, even if espressif just close to the public the use of such function it will remain used internaly. So, barely impossible to protect.

Right, but why does Espressif need this function? Is it really necessary? 

> Even if this kind of function is used by hardware crypto calls it is possible to to a bufferunderan to move pointers set to clear the registry and open the hardware to be invoked to another part of the code and move the cache to be pointed to another pointer.

Not sure I follow. But ideally this would be entirely in hardware.

In short, exposing esp_flash_read function to decrypt is a god send to a hacker. Yes, they need to gain local code execution (easy). Exposing this function is just asking to have your firmware extracted.

Just for some inspiration of what hackers are capable of. A turing complete software CPU written into ram using a corrupt PDF file: https://googleprojectzero.blogspot.com/2021/12/a-deep-dive-into-nso-zero-click.html

chipweinberger, 2022-07-02:
Bump. 

Just curious if this might be worked on / studied more.

Would be great to have flash encryption be impervious to software exploits.

chipweinberger, 2024-03-09:
Someone used this idea (using custom code to decrypt with esp_partition_read())  to hack the flash encryption!

> As a bonus round, [courk] then [decided to speed up](https://courk.cc/esp32-c3-c6-fault-injection) the painfully slow process of recovering the encryption keys by fault injection, which just requires the first 128 bytes (one block) on the -C3 and -C6 parts. Using a voltage fault injection the Secure Boot feature is bypassed. The essential idea is that through a buffer overflow custom code can be run, which dumps the entire Flash content. After demonstrating this and report it to Espressif, an [advisory was published](https://www.espressif.com/sites/default/files/advisory_downloads/AR2023-007%20Security%20Advisory%20Concerning%20Bypassing%20Secure%20Boot%20and%20Flash%20Encryption%20using%20CPA%20and%20FI%20attack%20on%20ESP32-C3%20and%20ESP32-C6%20EN.pdf) that notes that there’s no defense against this SCA and fault injection attack, other than using an ESP32 part that has internal Flash and no access to the QSPI bus from the outside.

https://hackaday.com/2024/01/15/breaking-the-flash-encryption-feature-of-espressifs-microcontrollers/

filzek, 2024-03-09:
The problem will only be solved with a unique key per flash, so this way, doing such work will do nothing in the end.

Paired keys to cloud and Mac, True Flash I'd, and MD5 from the read of readable refuses blocks will be enough to keep the healthy of a secure connection to the cloud and assure the software boundaries still intact.

What I see now is that the always secure problem is a flawless design if the software is now well designed as well.l, so, the manufacture process also relying on just hardware will always be a mess.

chipweinberger, 2024-03-09:
A unique key per flash doesn't help much once they've decrypted all your firmware and can create clones of your device.

filzek, 2024-03-09:
Sure, this is why you can use alternative approaches, like used in PS5 and other protected devices.

FW decryption can lead to a lot of secure flaws over time, so a correct security design shall be taken into consideration prior to anything else.

There is a primary security mistaken in the esp32 and others MCU design that doesn't have a hardcoded ID, mac address are also used as boundaries for the key to encrypt derivation, but this is not the correct approach.

PS5 uses 4 techniques, each flash used in the motherboard have theirs unique ID registered, the manufacturer, the id's of the MCU in the border, those all hardwares creates the main part of the salt to the RSA/Encryption keys to run the firmware and also to receive the firmware and secure connect to the cloud.

When the firmware got decrypted in some consoles, they had the maximum possibility to just do it in a unique device and not in all devices, and this is the main intention for the hardware contention losses, lower as possible the damage taken in the production environment.

So, a simple FW decryption becomes useless to any kind of network attack or intellectual property broken down, the device become just an hardware that can be used but not to create a harsh into the production environment.

mahavirj, 2024-03-11:
> Someone used this idea (using custom code to decrypt with esp_partition_read()) to hack the flash encryption!

Adding some details here:

For IDF case, the 2nd stage bootloader executes entirely from the internal memory of the target (to avoid any TOCTOU attacks). ROM loader must first verify the bootloader image and then load it into the internal memory through staging buffer. So, if we look at this part then entire bootloader contents must be treated as data itself, and loaded through the memory mapped cache (for decryption purpose) to internal memory. The attack also re-utilized same flash cache mapping (setup by the ROM loader) and it was able to read (decrypt) the bootloader contents.

> Why should code be able to decrypt instruction code? Im not sure a reason for it. Data, yes. But instructions?

Requirement is a valid one after successful setup of the application. However, early boot sequence or even the code that executes from internal memory does have a dependency on the decryption engine. Probably, hardware specific control for selective decryption could help to meet the requirement here. But for the early boot sequence part, we are still brain-storming on few countermeasures to reduce the attack surface.

dzungpv, 2025-07-09:
So what is the most security chip now? After read the https://courk.cc/breaking-flash-encryption-of-espressif-parts and his laser one https://courk.cc/rp2350-challenge-laser. Seem now impossible to prevent such attach on current generation ESP32 chip.

filzek, 2025-07-09:
@chipweinberger — thank you for confirming what Courk's research independently demonstrated: Flash Encryption and Secure Boot on ESP32 revisions (V‑3, C3, C6) can be compromised via side-channel and fault injection, resulting in full plaintext firmware recovery at scale.

This isn’t just a firmware misconfiguration issue—it’s a hardware-bound vulnerability. The AES core, boot ROM logic, and glitch-resilience design can be exploited using low-cost power traces or voltage glitches.

A robust secure deployment must ensure:

1. Unique hardware-rooted keys (eFuses + flash UID) tied to identity.

2. Disable UART reflashing and redundant boot paths.

3. Implement tamper-detection in firmware and restrict public bootloader access.

4. Cloud-side validation of firmware authenticity counters duplication.

In short: ESP32 provides tools (Secure Boot, Flash Encryption), but real resilience demands tying hardware identity, fault detection, and cloud attestation into a secure ring—otherwise clones and leaks remain real threats.