espressif/esp-idf #9036: I2C driver slave mode is missing critical features

kelnos, 2022-05-26:
This is a follow-up to #2202.  I've looked at the I2C driver in the forthcoming ESP-IDF v5, and it looks like there is no change from v4.4.  The issue is that it's not possible to implement an I2C slave with the current information that is exposed by the driver.  It is possible that manually implementing the I2C ISR would allow me to fix these deficiencies; however a) I should not have to do that myself, and b) I can't seem to find documentation as to how the ISR works, and what the various events mean with respect to what's happened on the wire.

The main issues with the current API are:

* The slave cannot tell when the master sends a start, repeated-start, or stop condition.
* The slave cannot tell when the master sends an ACK/NACK.
* The slave cannot read the low bit of the master-sent address field, which indicates if the master wants to read from or write to the slave.

Some more detail on these: if we look at the SMBus protocol definition, we can see that there are different ways of sending data back and forth.  The master can write or read a byte, word, or block of data (among other things).  The slave needs to be able to know if/when the master sends a stop condition in order to tell which is which.  A write-byte looks like this:

```
START -> (SLAVE_ADDRESS << 1) | WR -> ACK -> COMMAND_BYTE -> ACK -> DATA_BYTE -> ACK -> STOP
```
(In this case, the two `ACK`s are sent back by the slave, while the rest are sent from the master to the slave.  The `COMMAND_BYTE` bit often specifies a "register" to read or write on the slave side.)

However, a write-word looks like this:
```
START -> (SLAVE_ADDRESS << 1) | WR -> ACK -> COMMAND_BYTE -> ACK -> DATA_BYTE -> ACK -> DATA_BYTE -> ACK -> STOP
```
(Again the `ACK`s are sent back by the slave, while the rest are sent from the master to slave.)

Note that the slave is notified of completion by the use of the `STOP` condition.  Without being able to read the `STOP` in my slave code, I cannot tell if the master is intending to write one or two bytes.  There are more complicated examples; in particular both the SMBus "process call" and "block read" operations require notification of the "repeated start" condition bit being sent by the master.

You'll also note the `WR` bit added to the slave address.  I cannot tell if the master wants to read from or write to the slave without knowing the value of that bit.  The current I2C slave API does not seem to give me access to that information.

I see that #2096 has been stuck waiting on CLA signing for a long time.  I am frankly not sure if that new driver solves the problems I've listed here (I know it solves some _other_ problems, though).  I seems like it's designed to be a higher-level driver that (from what I can tell) more or less implements the SMBus register protocol, which I think is useful, but might be a bit too inflexible as the "default" driver.  Perhaps @loboris can chime in and clarify.

Ideally what I would like to see is, instead of just a simple `i2c_slave_read_buffer()`/`i2c_slave_write_buffer()` type API, something that allows me to consume I2C "events", probably via an ISR since timing is so important when it comes to I2C.  If we go a more low-level route, we'd probably want an event for pretty much everything:
* Start condition
* Repeated start condition
* Stop condition
* ACK/NACK
* Address + rd/wr bit
* Data byte
I don't love this, because it feels like there'd be a lot of overhead to get an event per byte, essentially.

I do also like the idea of a higher-level "driver" that implements the SMBus protocol and hides a lot of the details from the developer.  Something like that could be presented in the form of a vtable that the developer would fill in to provide handlers for the various SMBus operations (read byte, write byte, read block, process call, etc.), with automatic handling of register reads/writes.  But I don't think this should be the only interface presented; some people will want to implement their own non-SMBus protocol, or something else like PMBus or IPMI.

kelnos, 2022-05-27:
Digging through the I2C register descriptions, I found
```
/* I2C_SLAVE_RW : RO ;bitpos:[1] ;default: 1'b0 ; */
/*description: when in slave mode  1: master read slave  0: master write slave.*/
#define I2C_SLAVE_RW  (BIT(1))
#define I2C_SLAVE_RW_M  (BIT(1))
#define I2C_SLAVE_RW_V  0x1
#define I2C_SLAVE_RW_S  1
```

That seems like it would be helpful for determining if the master wants to do a read or write.  But I can't seem to find anything that can tell me about START/RSTART/STOP condition.

kelnos, 2022-05-27:
After examining `i2c_dev_t->slave_rw` in the I2C ISR, it doesn't seem like we see everything.  For example, for a SMBus read, the master will send the slave address twice, once first with the rw bit set to "write", and then again with the rw bit set to "read".  But the ISR only sees it getting set to "read".  It's possible that the transition does happen, but the original "write" state doesn't trigger an interrupt, so by the time the ISR runs, it's already seen two of them?

I also found that we get an interrupt when we see a `STOP` condition, which is useful.  The ISR doesn't seem to handle this right now, but it was simple to add.

(I'm working on a `i2c_slave_wait_event()` type API as a proof of concept just to see what information I can get out of the ISR.)

kelnos, 2022-05-27:
Frustratingly, I can't seem to get interrupts on `START` (or `RSTART`), even if I enable `I2C_TRANS_START_IN_ENA_M`.  Perhaps this only works in master mode (though master mode appears not to enable it either), or, from the docs, maybe this only does something when the ESP32 _sends_ a `START`, not when it receives one (which seems not all that useful to me)?

So I'm becoming a bit worried that it may just not be possible to implement a proper I2C slave on ESP32.

kelnos, 2022-05-27:
I've been doing some questionable things, like setting GPIO interrupts on the SDA and SCL lines so I can detect `START`, and then wait another 8 clock cycles to check if the hardware finds our device address, and the read/write bit.  This actually mostly works, though there are some odd cases where I miss a `START`, and I don't know why yet.

At this point, I'm running into more of the problems @loboris was hitting, mainly the issue where the master is requesting a byte from us, but our TX FIFO is empty, and it's seemingly impossible to provide data to the TXFIFO fast enough, so the master ends up getting garbage data.  The possible solutions I've come up with for this aren't really working reliably, and make for a really awkward API anyway.  I'm beginning to think that this whole "generic I2C slave driver" thing just isn't particularly practical, and @loboris's approach of doing something higher-level and essentially implementing a SMBus "memory" driver is the right approach.  This is unfortunate, but I think might be necessary.

Even then, there are still things that are probably difficult to implement with how this hardware works.  For example, the SMBus "quick" command consists of just:
```
START -> (SLAVE_ADDRESS << 1) | WR -> ACK -> STOP
```
The idea here is that it's a super simple request to the slave that doesn't require the master to send any data or the slave to reply with any data.  The only bit of information the slave gets is the state of the `WR` bit in the address field.  As it stands now, it's still really hard to notify an application of this; AFAICT my ugly hack of setting interrupts on the SDA and SCL lines and counting clock cycles is the only way to do it (and then unreliably), unless I'm missing something.

I'm getting pretty close to giving up here.  My original need for this was to emulate an I2C fan controller IC, because nearly all of the parts I'm finding suitable for my project are out of stock and have 1+ year lead times right now.  But I'm thinking it might be simpler to give up on this idea with the ESP32 and instead buy one of the less-suitable, but still probably workable, fan controller ICs I've found that are actually in stock.

igrr, 2022-06-14:
Hi @kelnos, I think the feature you are looking for is not available in the ESP32 I2C hardware. In I2C slave mode, the program running on the ESP32 can't "react" to the address specified in the command. Data to be sent to the master should be loaded into the FIFO in advance, before the command is received. This means that you can't implement an SMBus slave using an ESP32.

Starting from the ESP32-S2, it is possible to receive an interrupt when I2C controller in slave mode matches the address, or when the TX FIFO is empty. The hardware will start clock stretching when this happens and the software has time to handle the interrupt and put the required data into the TX FIFO. There is no support for this mode in ESP-IDF I2C slave driver, though. Support for this was recently implemented in the Arduino project (https://github.com/espressif/arduino-esp32/pull/5746), and we will consider adding this feature when reworking the I2C driver in ESP-IDF.

kelnos, 2022-06-14:
Ah,  I see.  I guess ESP32 just isn't that useful as an I2C slave device, more it was a "prototype" that was fleshed out with the S2.  Thank you for the reply!

closed 2022-06-14 by kelnos

vihanb, 2023-02-14:
@kelnos One year later, I am encountering this same situation now. What approach did you end up taking?

finger563, 2023-06-16:
@igrr When is the planned i2c driver rewrite which would support this functionality on supported hardware?

schveiguy, 2024-09-23:
FYI, I have a completely rewritten slave driver that I've used in an internal project, which handles mostly the things that need to work. I also am working with SMBus on top of it.

See my slave driver here: https://github.com/schveiguy/espi2cslave/

I do not handle nack/ack differences, but maybe it can be added. The key thing is to properly implement the clock stretching, and the interrupt handling.