espressif/esp-idf #9037: Parameter checking in esp_gap_ble_api.c, esp_ble_gap_update_conn_params() is wrong.

felixcollins, 2022-05-26:
## Environment
IDF 4.4.1

## Problem Description

By Inspection, the parameter checking in esp_gap_ble_api.c, esp_ble_gap_update_conn_params seems to be wrong.

line 146
`
if (ESP_BLE_IS_VALID_PARAM(params->min_int, ESP_BLE_CONN_INT_MIN, ESP_BLE_CONN_INT_MAX) &&
        ESP_BLE_IS_VALID_PARAM(params->max_int, ESP_BLE_CONN_INT_MIN, ESP_BLE_CONN_INT_MAX) &&
        ESP_BLE_IS_VALID_PARAM(params->timeout, ESP_BLE_CONN_SUP_TOUT_MIN, ESP_BLE_CONN_SUP_TOUT_MAX) &&
        (params->latency <= ESP_BLE_CONN_LATENCY_MAX || params->latency == ESP_BLE_CONN_PARAM_UNDEF) &&
        ((params->timeout * 10) >= ((1 + params->latency) * ((params->max_int * 5) >> 1))) && params->min_int <= params->max_int) {
`
The `latency`  parameter is used in the final check even if it is set to `ESP_BLE_CONN_PARAM_UNDEF`. This can make the check fail when the callers intention is to not update that parameter.

The same is true for the other parameters. The `ESP_BLE_IS_VALID_PARAM` macro returns true if they are set to `ESP_BLE_CONN_PARAM_UNDEF`.  This problem seems to be widespread.

### Expected Behavior

In the case that `ESP_BLE_CONN_PARAM_UNDEF` is passed in, instead of using the passed parameter in the check, the currently set value should be used.

Additionally, the documentation and comments should provide the conditions under which the parameters are valid.

zhp0406, 2023-11-17:
In the latest version, we have fixed the issue by addressing the bug and added checks for connection parameters. Thank you for reporting the issue.

zhp0406, 2023-11-17:
This is the fixed ID:
Commit ID: bd2bde144883fbc81e73a4c432c194396d98efe0

closed 2023-11-21 by Indastri