espressif/esp-idf #9039: mtime incorrect in fatfs during daylight saving time

lbernstone, 2022-05-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      Custom, but N/A
- Module or chip used:  ESP32-C3, likely all
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         Arduino
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-linux-amd64
- Operating System:     Linux
- Using an IDE?: Yes , arduino just for convenience.  No Arduino code included in example
- Power Supply:         USB

## Problem Description

When a timezone is set with DST, and time is set inside DST, file modified time (mtime) is 3599 seconds off.  Weird number, maybe something is supposed to be one hour diff, but isn't multiplied?

### Expected Behavior
System time and file written time should be identical

### Actual Behavior

```
22:29:11.888 -> systime: 1653638041
22:29:12.154 -> filetime: 1653641640
```

### Steps to reproduce

```
#include <vfs_fat_internal.h>
static wl_handle_t hndWl = WL_INVALID_HANDLE;

void setup() {
  Serial.begin(115200);
  delay(2000);
  struct timeval tv = {1653638041, 0};
  settimeofday(&tv, NULL);
  setenv("TZ", "MST7MDT,M3.2.0,M11.1.0", 1);
  tzset();
  Serial.printf("systime: %d\n", time(NULL));
  const esp_vfs_fat_mount_config_t conf = {
      .format_if_mount_failed = true,
      .max_files = 4,
      .allocation_unit_size = CONFIG_WL_SECTOR_SIZE
  };
  esp_vfs_fat_spiflash_mount("/fatfs", "ffat", &conf, &hndWl);
  FILE *tfile = fopen("/fatfs/test", "w");
  fprintf(tfile, "hello");
  fclose(tfile);
  struct stat info;
  stat("/fatfs/test", &info);
  Serial.printf("filetime: %d\n", info.st_mtime);
}
void loop() {delay(-1);}
```

igrr, 2022-05-27:
Thanks for reporting this @lbernstone. I vaguely remember there being some discussion on this before, and some reason why the filesystem works the way it does, but I can't find it right now. We'll investigate this.

The difference is probably indeed one hour (3600 seconds), but FATFS keeps time at 2-second precision, so instead of mtime=1653641641 (exactly 3600 seconds difference) you got 1653641640.

lbernstone, 2022-05-27:
Only thing I see in the history is this https://github.com/espressif/esp-idf/issues/1369

closed 2022-06-18 by espressif-bot