espressif/esp-idf #9040: usb-cdc overwrites its own buffer during write, hence data arrives truncated

moefear85, 2022-05-27:
## Environment

- Module or chip used:  esp32-s2
- IDF version: affects all versions from atleast v4.3.1 to latest master
- Build System:   idf.py
- Operating System:  Linux
- Using an IDE?: VSCode IDF Extension
- Power Supply:      external 3.3V

## Problem Description

When calling tinyusb_cdcacm_write_queue(), then the data sent out is always the last portion of whatever buffer was given, and it depends on the configured cdc tx fifo size in sdkconfig.
### Expected Behavior

That when sending a buffer larger than what tinyusb_cdcacm_write_queue() can handle, that it only write as much as it can, and return that value, so another call can be placed later at proper offsets, until the entire buffer is sent.

### Actual Behavior

The function correctly reports the size it accepted & sent, but instead of taking the first n bytes of the buffer, it takes the last n bytes. This means if the buffer in sdkconfig is chosen large enough, such as 1024 bytes, one might never notice this problem. But at smaller sizes such as 64 bytes, it leads to stream corruption (truncation of beginnings).

### Steps to reproduce

1. Any sample project that tests tinyusb_cdcacm_write_queue(), as long as the size of the buffer passed is larger than the tx FIFO size chosen in sdkconfig (CFG_TUD_CDC_TX_BUFSIZE).

### Cause of Problem

The structure "tu_fifo_t" used to hold the outgoing data contains a boolean called "overwritable". The default value seems to be true. Hence when tud_cdc_n_write() is called, and the source buffer is larger than the ringbuffer in "tu_fifo_t", it will wrap around and overwrite itself, so that if the ringbuffer size is m and the passed buffer size is n, then the first (n-m) bytes get lost/overwritten, and only the last n bytes are saved and thus later flushed/written out.

### Workaround

I manually set the value of "overwritable" to false before doing any sending, and this solves the problem. No matter how small CFG_TUD_CDC_TX_BUFSIZE (even a value of 1 works), and no matter how large the buffer I pass to tinyusb_cdcacm_write_queue(), it will only queue up to the first n bytes of the passed buffer, correctly returning the size it queued, so that later calls at proper offsets will correctly queue & send out the right offsets.

moefear85, 2022-06-01:
yeah my general thinking was placing a menuconfig constraint to precent bewilderment when someone experiments with the value. as for the tx path, the solution in my experience is to simply set overwritable to false during cdc driver install.

but honestly, there are issues open since 2019 for which PRs have been submitted that haven't been merged yet. I again urge espressif to fix the uart BREAK issue. it is raised as an event immediately upon arrival, thus jumping lines relative to preexisting buffered data, which makes it unusable in most cases despite being important to properly delimit frames of any protoc sent over serial.

tore-espressif, 2022-06-02:
About the TX path:

> the data sent out is always the last portion of whatever buffer was given

Will fix. Thanks for letting us know.

> it no longer automatically flushes when necessary

I could reproduced this too. For buffer sizes < 64 this condition is never met https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L174

I'll publish PR in upstream TinyUSB project.

moefear85, 2022-06-02:
thx i'll close this

Alvin1Zhang, 2022-06-06:
Thanks for reporting, reopen until the fix is available on GitHub. Thanks.

moefear85, 2022-06-17:
an issue to consider:

after applying the mentioned workaround, i realize under heavy writing in both directions to usb, then despite using 10ms delays between tinyusb_cdcacm_write_queue and tinyusb_cdcacm_write_flush, the usb subsystem still hangs after random intervals and doesn't recover. The issue doesn't occur without the patch, but then corruption does. I also can't avoid having to call flush(), because if I don't, then the next queue operation would return immediately with a value of 0, hence any send loop I'm using to retry unsent parts of data, will trigger the watchdog.

closed 2022-06-18 by espressif-bot