espressif/esp-idf #9041: esp-idf-v5 breaks ability to flash target chips using an esp32-s2 as a uart-bridge

moefear85, 2022-05-27:
## Environment

- Module or chip used:  esp32-s2
- IDF version (run ``git describe --tags`` to find it): esp-idf-v5 (all commits affected)
- Build System:      idf.py
- Operating System:    Linux
- Using an IDE?: VSCode + ESP-IDF Extension
- Power Supply:  external 3.3V

## Problem Description

A while back I wrote my own tool to use an esp32-s2 as a serial flasher for other esp modules. I recently realized it compiles but doesn't work with esp-idf-v5. I even tried the project "esp-usb-bridge" which is published by espressif itself. It too works against esp-idf-v4.4 but not against esp-idf-v5.
I tried latest master as well as the tag "v5.0-dev". Neither of them work with either project. This along with further analysis seems to hint this isn't a bug/mistake, rather a silent change between esp-idf-v4 and esp-idf-v5.

### Expected Behavior

I expect after placing a target module into download-mode, that it respond to esptool packets.

### Actual Behavior

esptool packets are successfully forwarded by the flasher module to the target module. But responses from the target module are silently ignored by the flasher module, not detected at all by the uart subsys, hence not forwarded to esptool over usb.

### Steps to reproduce

1. compile esp-usb-bridge against esp-idf-v5 (any commit) and try to flash any target esp module using it.

### Code to reproduce this issue

see the project "esp-usb-bridge".

## Other items if possible

These unexpected changes (along with others that don't need mentioning) don't seem coincidental, and they really discourage one from contributing and sharing back with the community.

dobairoland, 2022-05-31:
Thank you @moefear85 for letting know about this. This cannot be reproduced in every cases. We are trying to find  and understand the root cause....

Duplicate issue: https://github.com/espressif/esp-usb-bridge/issues/5

moefear85, 2022-06-07:
@dobairoland 

Further testing, with a completely different project, has revealed the following:

There is definately a considerable data corruption bug in the uart subsystem in idf-v5. It is not present in v4.4. It is most likely the root cause of the above mentioned problem, as well as further corruption in other projects. But the issue is very specific, which is perhaps why you were not able to reproduce it.

It affects only esp-idf-v5, and only manifests itself at speeds above 115200, and with all likelyhood only affects uart 1 and possibly uart 2 on the esp32 and esp32-s2. I can't test uart 0, because I always use that to output logging info, but that works flawlessly even at 2073600 baud. But this might be masking the problem, because even with uart 1, I realize, if I connect it to my PC, then I see all the communication I am expecting (although I can't verify if it is correct byte for byte), but when I connect two esp chips together, then the corruption is clearly detectable. So it might even be so specific that it only affects two esp chips connected together.

Whatever you changed in the uart subsystem between v4.4 and v5, it has corrupted it in certain areas.

I will try later to create a specific testing code to detect/demonstrate the problem then upload it. For now, trust me, this is not a bug anywhere in my app, it is in esp-idf.

dobairoland, 2022-06-08:
> Can you atleast verify that you are aware that there is a problem? This is not clear from your comment

I think currently only ESP-USB-Bridge developers looked at it. Only one of us was able to reproduce it. We are attempting to narrow down the root cause so we could report it for the chip driver development team. If you would happen to have a minimal example for reproduction then please share it - it could speed up the process.

erhankur, 2022-06-09:
@moefear85 In the uart_config, can you set `.source_clk = UART_SCLK_DEFAULT` and try again? e.g. 
``` 
    uart_config_t uart_config = {
        .baud_rate = config->baud_rate,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_DEFAULT,
    };
   ```

moefear85, 2022-06-09:
@ erhankur

thx, UART_SCLK_DEFAULT resolved the issue.

I can swear I had already tried all possible values for this, but perhaps at the time I had fiddled around with too many other settings that eventually masked its effect/success. It's working flawlessly, as is esp-uart-bridge as well as my own flashing tool.

So if one doesn't set UART_SCLK_DEFAULT, then it will take a default value of 0, which won't work in idf5, but will work in idf4.

Still, assuming one uses a wrong value of 0, then despite not being able to communicate properly, when a UART_DATA event is generated, "event.size" is still supposed to have an acceptable/reasonable value. But I realized sporadically (in a repetitive fashion), a UART_DATA event is generated, but where *event.size* has the value of 1073422464, which is bogus (it by far exceeds the chosen buffer size, around 2048 in my case). So the value is always either a correct/reasonable value, or 1073422464 (which might be some error code, but it's not documented, and event.size has a type of size_t). This is still a bug that needs to be fixed, since it was throwing off and crashing my code until I caught it through
>assert(event.size<=CONFIG_UART_BUF_SIZE)

. Correct behavior would be either not to raise a UART_DATA event in the first place, or have event.size return 0.

closed 2022-06-09 by moefear85