espressif/esp-idf #9044: (NimBLE) Example for L2CAP COC Communication

mickeyl, 2022-05-28:
(I'm on an ESP32-C3)

I'd like to see an example for using an ESP32C3 via BLE L2CAP connection oriented channel.

I've augmented the standard `blehr` configuration with

CONFIG_BT_NIMBLE_L2CAP_COC_MAX_NUM=1

and enhanced the sample with 

```
rc = l2cap_create_server(0x235, 500, _l2cap_evt_handler, NULL);
assert(rc == 0);
```

but apparently the l2cap event handler callback does not get called.

Is anything else necessary to get l2cap channel communication working?

IshaESP, 2022-11-07:
Hi @mickeyl, we have added an example for this under the name - ble_l2cap_coc.

mickeyl, 2022-11-07:
@IshaESP Great, thanks a lot, I will have a look! I have since implemented my first bits and they seem to work fine and stable although I have problems with memory management (IMHO the NIMBLE memory pool functions are not very well documented), so I'm looking forward to checking out your example.

mickeyl, 2022-11-22:
I wonder why you don't never release any mbuf instances (e.g., in `BLE_L2CAP_EVENT_COC_DATA_RECEIVED`). If I do that, I quickly run out of memory.

rahult-github, 2022-11-24:
Hi @mickeyl ,  

mbuf mangement ( allocation / free ) is taken care of by the underlying stack. If an application needs to use the data received from stack via events , it is advisable to make a copy of it and store if it needs to be used later on. Once the control goes back to stack ( say after the event posting functionality is over), then stack would taking the decision of freeing up the mbuf that was allocated by it.

mickeyl, 2022-11-24:
@rahult-github Very interesting. Like I said, this doesn't go with my findings. If remove e.g. the `os_mbuf_free_chain` in https://github.com/mickeyl/esp-nimble-cpp/blob/25d23f92270d007ce0635f71cd81b3881658f113/src/NimBLEL2CAPService.cpp#L131, I get out-of-memory within short time. But anyways, if that's an issue, then it's another issue. Thanks for the example code, I'll close this here.

closed 2022-11-24 by mickeyl

nebkat, 2024-03-15:
Could we get a similar example for Bluedroid LE, or does it function using the Bluetooth Classic L2CAP API?