espressif/esp-idf #9045: `esp_wifi_get_config` yields `wifi_config_t` with wrong channel value

chenlijun99, 2022-05-28:
## Environment

- Development Kit: ESP32-S2-Saola-1
- Module or chip used:  ESP32-S2-WROVER
- IDF version: both `v4.4.1` and `v5.0-dev-3055-g179ea878c7` have this problem
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

`esp_wifi_get_config` returns wrong value for the field `sta.channel` in `wifi_config_t`.

### Expected Behavior

`sta.channel` value read using `esp_wifi_get_config` should be the same that has been set using `esp_wifi_set_config`.

### Actual Behavior

They are different.

Interestingly, using `v4.4.1` the value that is read back is 63, while using `v5.0-dev-3055-g179ea878c7` the value that is read back is 0.

### Steps to reproduce

1. Set a STA config with  channel `1` using `esp_wifi_set_config`
2. Read it back using `esp_wifi_get_config`
3. The value read back is not the same value that has been set

### Code to reproduce this issue

https://gist.github.com/chenlijun99/2500115661b3ecf241c18ab49f0252d3

Alvin1Zhang, 2022-08-12:
Thanks for reporting, sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/804bdc1721eb8cbb24be6ab7bd585b1a83c38b97, we are also back porting to other release branches. 

Will update further if any progress. Thanks.

AxelLin, 2022-08-12:
@Alvin1Zhang 
Are you sure 804bdc1 is related to this issue?

closed 2023-04-24 by Alvin1Zhang