espressif/esp-idf #9046: WiFi STA in LR mode finds also APs that use "normal" WiFi" when scanning

chenlijun99, 2022-05-28:
Maybe a more general question is: can LR mode also receive frames from normal "WiFi"?
I'm know almost zero about RF. I wonder if this is the expected behaviour or I am configuring the LR mode wrong.

zhangyanjiaoesp, 2022-05-30:
Do you set the LR only mode ( `esp_wifi_set_protocol(ESPNOW_WIFI_IF, WIFI_PROTOCOL_LR) )`) or bgnLR mode (`esp_wifi_set_protocol(ESPNOW_WIFI_IF, WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N|WIFI_PROTOCOL_LR) )`)

chenlijun99, 2022-05-30:
I set the LR only mode.`esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR)`

chenlijun99, 2022-05-30:
I have also used Wireshark to sniff packets. Since I'm performing active scan, the STA should send probe requests.
Using LR only mode `esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR)`, I can't see any probe requests with Wireshark. This makes sense since LR mode is proprietary of ESP and I think the probes are being sent using WiFi LR modulation. OTOH, it seems that the LR only mode STA is still able to receive beacons from "normal WiFi".

If I use "normal WiFi", `esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B)`, then I am also able to see the probe requests of the STA.

zhangyanjiaoesp, 2022-05-30:
1. In LR only mode, ESP STA send the probe requests using the LR rate, this can't be parsed by the Wireshark, so you can't sniff them.
2. In LR only mode, ESP STA can receive beacons from "normal WiFi", this is correct, but it can't connect to it.

chenlijun99, 2022-05-30:
Thank you! I'm asking because of my lack of knowledge of RF. So the answer to "can LR only mode also receive frames from normal WiFi?" is YES. Correct?

zhangyanjiaoesp, 2022-05-30:
yes

chenlijun99, 2022-05-30:
Perfect. Thank you for your help!

closed 2022-05-30 by chenlijun99