espressif/esp-idf #9047: Failed to install ESP-IDF 5.0 because of 403 Forbidden error

mike-2020, 2022-05-28:
When I run install script on "master", it failed to download the file below because server resturns "403 Forbidden" error

Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt
Destination: /home/mike/.espressif/espidf.constraints.v5.0.txt.tmp
WARNING: Download failure HTTP Error 403: Forbidden
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt to /home/mike/.espressif/espidf.constraints.v5.0.txt.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/home/mike/SDKs/esp-idf/tools/idf_tools.py", line 2476, in <module>
    main(sys.argv[1:])
  File "/home/mike/SDKs/esp-idf/tools/idf_tools.py", line 2468, in main
    action_func(args)
  File "/home/mike/SDKs/esp-idf/tools/idf_tools.py", line 1971, in action_install_python_env
    constraint_file = get_constraints(idf_version)
  File "/home/mike/SDKs/esp-idf/tools/idf_tools.py", line 1889, in get_constraints
    raise DownloadError()
__main__.DownloadError

FanhuaCloud, 2022-05-29:
Yes, I have the same problem

dobairoland, 2022-05-30:
Hi @mike-2020, @FanhuaCloud. Please try again. It was an issue with mirroring between servers and it has been fixed. We are sorry for the inconvenience.

closed 2022-05-30 by dobairoland

FanhuaCloud, 2022-05-30:
> Hi @mike-2020, @FanhuaCloud. Please try again. It was an issue with mirroring between servers and it has been fixed. We are sorry for the inconvenience.

Thanks, it works fine now.

MichaelWu-ESP, 2022-12-06:
> I'm seeing this issue again now - unable to complete the download : Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt

I can see it's working fine, pls try again.

timsterc, 2022-12-06:
Yep, working now.  Thanks!

yusuhua, 2023-09-07:
I'm seeing this issue again now - unable to complete the download :
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt

MichaelWu-ESP, 2023-09-13:
@yusuhua where did you access this? I tried from several overseas servers, working fine.

tianrking, 2023-10-15:
still fail

dalbert2, 2025-04-30:
URL:
  https://components-file.espressif.com/components/espressif/esp_modem.json

  Status code: 403 Forbidden

MichaelWu-ESP, 2025-04-30:
> URL: https://components-file.espressif.com/components/espressif/esp_modem.json
> 
> Status code: 403 Forbidden

Fixed now, pls try again.

dalbert2, 2025-04-30:
Working, thank you!