espressif/esp-idf #9048: Convserver issue in Eclipse after update from IDF 4.4 to 4.4.1 'error retrieving output from JSON configuration server'

ralphde, 2022-05-28:
## Environment
- Development Kit:     all
- Module or chip used:  ESP32-WROVER-I
- IDF version (run ``git describe --tags`` to find it):
    // v4.4.1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [indows
- Using an IDE?: Eclipse 2021-09
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

We upgraded from 4.4 to 4.4.1 and since we observing that our custom project deriving from HTTPS simple server example does not run anymore as smoth as it did with IDF 4.4. 

Trying to nail this down we looked at the original sample project we took as base to build our application to see if it runs as well not as expected any more.

However, it even does not compile any more and we can not open the sdkconfig file.

We get the error 'error retrieving output from JSON configuration server'

and the console output looks like this
`Executing action: confserver
Running ninja in directory c:\esp-ws-202109-17\simple\build
Executing "ninja confserver"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
-- Checking Python dependencies...
Python requirements from C:\_esp-idf-v4.4\requirements.txt are satisfied.
-- Project sdkconfig file C:/esp-ws-202109-17/simple/sdkconfig
Loading defaults file C:/esp-ws-202109-17/simple/sdkconfig.defaults...
CMake Error at C:/_esp-idf-v4.4/tools/cmake/crosstool_version_check.cmake:37 (message):
  

  Toolchain:
  C:/Users/ralph/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe,
  crosstool-ng version esp-2021r2-patch2 doesn't match supported version
  esp-2021r2-patch3

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.

Call Stack (most recent call first):
  C:/_esp-idf-v4.4/components/esp_common/project_include.cmake:7 (crosstool_version_check)
  C:/_esp-idf-v4.4/tools/cmake/build.cmake:324 (include)
  C:/_esp-idf-v4.4/tools/cmake/build.cmake:537 (__build_process_project_includes)
  C:/_esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp-ws-202109-17/simple/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
C:\Users\ralph\.espressif\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SC:\esp-ws-202109-17\simple -BC:\esp-ws-202109-17\simple\build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1`

The given fixes do not work. Deleting the build and component folder does not fix it either. Project does not compile.
Can anyone enlighten us what is going on and how to fix? Because any other project we can open the sdkconfig

We would like to understand what is going on here
 
Here is the sdkconfig
`#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
[... 1362 lines cut ...]
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_USB_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
`

kolipakakondal, 2022-05-30:
Hi @ralphde 

I was able to resolve this issue after running `idf.py fullclean` on the project and I have followed the below steps.

1. Select the project
2. From the toolbar, click on the `Open terminal` icon
3. Select ESP-IDF Terminal
4. Run `idf.py fullclean`
5. Re-build the project using eclipse toolbar build icon
6. Launch sdkconfig editor

[image]

mbratch, 2022-05-30:
@kolipakakondal I have the exact same problem, but your solution didn't work for me. There is no "Open Terminal" on my toolbar I could not find "Open Terminal" anywhere when right-clicking the project, or in the menu tree, either.

What I tried instead then was open an ESP-IDF 4.4.1 power shell terminal, cd to my project, and did an `idf.py fullclean` there. However, I still get the same error in Eclipse. I can build using the command line in the power shell terminal, `idf.py build`. From then on, incremental builds work in Eclipse. But I would like to find a solution to the Eclipse issue. I can see in the Preferences that the new "patch3" tools don't show up in the directory list for the Core Build Toolchain. Eclipse has this as read only and I cannot change it. I can "add" a user Toolchain path, but I cannot prioritize it so it doesn't seem to recognize it. I tried r3einstalling the ESP-IDF tools from the Eclipse Espressif menu but that didn't change anything.

Every time I upgrade my ESP-IDF, it takes a few days to get Eclipse working again. It seems each time it's a slightly different problem.

ralphde, 2022-05-30:
@mbratch 
[image]
you don't have this icon? is the only way I know to open the various terminals.

mbratch, 2022-05-30:
@ralphde oh my bad. Yeah I found it. The icons are tiny. 📦 

Regardless, I did run `idf.py fullclean` from an ESP-IDF terminal and it did not fix my problem, unfortunately. Somehow, Eclipse is locked into using the "patch2" compiler and the latest one ESP-IDF tools want is "patch3". I've tried all sorts of ways to coax it but no luck so far. If this defect gets closed, I'll probably raise a new one.

ralphde, 2022-05-30:
@kolipakakondal @mbratch  I agree, however, what I have not tested is if it is the same issue when you use the command line sdkconfig utility. I would expect so but need to be tested. may be not, but would tell if that is a Eclipse issue or IDF issue I am just too busy right now to do it. Glad that you have a broken project too (sorry:)

ralphde, 2022-05-30:
and I even have deleted the component and build folder manually. No success.