espressif/esp-idf #9049: install.sh Error

pylover, 2022-05-28:
OS: Ubuntu 22.04 LTS
Commit: 179ea878c747519138b40f387b80f30dd72097f0

`install.sh` script raised this error during the python virtual environment creation:

```
Installing Python environment and packages
pip 22.1.1 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Creating a new Python environment in /home/vahid/.espressif/python_env/idf5.0_py3.10_env
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: 'override.py'
Consider using the `--user` option or check the permissions.

RuntimeError: failed seed with code 1
Traceback (most recent call last):
  File "/home/vahid/esp32/esp-idf/tools/idf_tools.py", line 2476, in <module>
    main(sys.argv[1:])
  File "/home/vahid/esp32/esp-idf/tools/idf_tools.py", line 2468, in main
    action_func(args)
  File "/home/vahid/esp32/esp-idf/tools/idf_tools.py", line 1958, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'virtualenv', '--python', '/usr/bin/python3', '--seeder', 'pip', '/home/vahid/.espressif/python_env/idf5.0_py3.10_env']' returned non-zero exit status 1.
```

pylover, 2022-05-28:
Fixed by:

```diff
diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index 8d7092e953..0296856655 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -1242,7 +1242,7 @@ def get_python_exe_and_subdir() -> Tuple[str, str]:
         subdir = 'Scripts'
         python_exe = 'python.exe'
     else:
-        subdir = 'bin'
+        subdir = 'local/bin'
         python_exe = 'python'
     return python_exe, subdir
 
@@ -1953,7 +1953,7 @@ def action_install_python_env(args):  # type: ignore
         info('Creating a new Python environment in {}'.format(idf_python_env_path))
         virtualenv_options = ['--python', sys.executable]
         if with_seeder_option:
-            virtualenv_options += ['--seeder', 'pip']
+            virtualenv_options += ['--seeder', 'app-data']
 
         subprocess.check_call([sys.executable, '-m', 'virtualenv',
                                *virtualenv_options,
```

moefear85, 2022-05-29:
are you sure its necessary? maybe you're not starting with a clean environment? I freshly installed ubuntu 22.04 and latest master just yesterday and everything is working fine.

pylover, 2022-05-29:
I tested on a fresh environment, and also deleted the `$HOME/.espressif` folder.  I'm wondering about the `$ENV/local/bin` folder instead of the `$ENV/bin`.

 Some users have the same issue with python's venv too: https://unix.stackexchange.com/questions/689319/how-to-prevent-pip-from-installing-to-venv-local-bin

dobairoland, 2022-05-30:
Hi @pylover.

Could you please run the same command manually and share the output?

```sh
/usr/bin/python3 -m virtualenv --python /usr/bin/python3 --seeder pip /home/vahid/.espressif/python_env/idf5.0_py3.10_env
```

BTW, this part of the code will be updated on the next sync to Github. The built-in `venv` module will be used on Linux instead of the `virtualenv` package.

pylover, 2022-05-30:
This is output of the command above:

```
  WARNING: No metadata found in /home/vahid/.local/lib/python3.10/site-packages
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/usr/local/lib/python3.10/dist-packages/distutils-precedence.pth'
Consider using the `--user` option or check the permissions.

RuntimeError: failed seed with code 1

```

dobairoland, 2022-05-31:
It looks like in your case it tries to install things globally. I haven't seen this issue before and cannot reproduce it. I also got a report from a colleague that this is working on Ubuntu 22.04 LTS. It must be something with your setup.

Also if you try the latest master then this command is not used anymore.

pylover, 2022-05-31:
Thanks for the time.

Currently, my `esp-idf` is working well. 

FYI, I faced this error within a fresh environment, but I'll try to test the latest master on weekend.

closed 2022-06-22 by dobairoland

zingo, 2022-10-20:
I got this when installing v4.4.2 switch to clone "latest" (e.g. without -b v4.4.2 ) then install passed.
This was on pop!_os 22.04

dobairoland, 2022-10-25:
Do you mean the following?
```
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: 'override.py'
Consider using the `--user` option or check the permissions.
```

I would say that it is more likely that https://github.com/espressif/esp-idf/issues/9931 is happening for you, for which we have just included a fix a couple of days ago. v4.4.2 is affected.