espressif/esp-idf #9050: Ubuntu 22.04 /export.sh >> This script should be sourced, not executed:

politsin, 2022-05-28:
----------------------------- Delete below -----------------------------

I try to set up tools on linux https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

## Environment
- IDF version: v5.0-dev-3055-g179ea878c7
- Build System: idf.py
- Operating System: Ubuntu 22.04

## Problem Description
Can't install tools

### Steps to reproduce

1. step1
2. ...

## Logs:
```sh
$ ./install.sh esp32
***************
Collecting idna<4,>=2.5
  Downloading https://dl.espressif.com/pypi/idna/idna-3.3-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 KB 2.2 MB/s eta 0:00:00
Building wheels for collected packages: esp-coredump
  Building wheel for esp-coredump (setup.py) ... done
  Created wheel for esp-coredump: filename=esp_coredump-1.2-py3-none-any.whl size=33210 sha256=14c2e9e1c983ad9558965f93dff78561ab83b83b179988c53c87dbbe4aa1e5ff
  Stored in directory: /home/s/.cache/pip/wheels/4f/f5/71/86f358ec7c80f7a918207bac4cb447a7274f2ae29aa7448a50
Successfully built esp-coredump
Installing collected packages: reedsolo, pyserial, pygdbmi, pyelftools, bitstring, urllib3, tqdm, six, setuptools, pyyaml, pyparsing, pycparser, kconfiglib, idna, future, freertos_gdb, contextlib2, construct, colorama, click, charset-normalizer, certifi, schema, requests, ecdsa, cffi, requests-toolbelt, cryptography, idf-component-manager, esptool, esp-coredump
  Attempting uninstall: setuptools
    Found existing installation: setuptools 62.1.0
    Uninstalling setuptools-62.1.0:
      Successfully uninstalled setuptools-62.1.0
Successfully installed bitstring-3.1.9 certifi-2022.5.18.1 cffi-1.15.0 charset-normalizer-2.0.12 click-8.0.4 colorama-0.4.4 construct-2.10.68 contextlib2-21.6.0 cryptography-36.0.2 ecdsa-0.17.0 esp-coredump-1.2 esptool-3.3.1 freertos_gdb-1.0.0 future-0.18.2 idf-component-manager-1.1.0 idna-3.3 kconfiglib-14.1.0 pycparser-2.21 pyelftools-0.27 pygdbmi-0.9.0.2 pyparsing-3.0.9 pyserial-3.5 pyyaml-6.0 reedsolo-1.5.4 requests-2.27.1 requests-toolbelt-0.9.1 schema-0.7.5 setuptools-62.3.2 six-1.16.0 tqdm-4.64.0 urllib3-1.26.9
WARNING: You are using pip version 22.0.4; however, version 22.1.1 is available.
You should consider upgrading via the '/home/s/.espressif/python_env/idf5.0_py3.10_env/bin/python -m pip install --upgrade pip' command.
All done! You can now run:

  . ./export.sh
----------------------------------------
$ chmod +x export.sh
$ ./export.sh
This script should be sourced, not executed:
. ./export.sh
```

politsin, 2022-05-28:
Same result on Debian
```sh
s@debian:~/esp/esp-idf$ python --version
Python 3.9.2
s@debian:~$ mkdir -p ~/esp
s@debian:~$ cd ~/esp
s@debian:~/esp$ git clone --recursive https://github.com/espressif/esp-idf.git
s@debian:~/esp$ cd ~/esp/esp-idf
s@debian:~/esp/esp-idf$ ./install.sh esp32
***
WARNING: You are using pip version 22.0.4; however, version 22.1.1 is available.
You should consider upgrading via the '/home/s/.espressif/python_env/idf5.0_py3.9_env/bin/python -m pip install --upgrade pip' command.
All done! You can now run:

  . ./export.sh

s@debian:~/esp/esp-idf$ ./export.sh
-bash: ./export.sh: Access denied
s@debian:~/esp/esp-idf$ chmod +x ./export.sh
s@debian:~/esp/esp-idf$ ./export.sh
This script should be sourced, not executed:
. ./export.sh
s@debian:~/esp/esp-idf$

```

atanisoft, 2022-05-28:
It looks like you missed the "." at the start of the command, in the docs it is `. ./export.sh` and you are executing `./export.sh`. The `.` is shorthand for `source` and likely the docs should be updated to be `source ./export.sh` to be explicit.

politsin, 2022-05-28:
Thank you! That helps!
This is the first time I've seen a command in linux look like this `. ./export.sh`

closed 2022-05-28 by politsin

chegewara, 2022-05-28:
It is short version of `source ./export.sh`