espressif/esp-idf #9051: free() error

cyberumut, 2022-05-28:
my code 

```
char ipadress[15];
	char ipadresslast[15];
	
	char port[25];

char* ipadressx=malloc(20);
			char* portx=malloc(20);
	
		snprintf(ipadress,sizeof(ipadress),"iBMTAi%d",33);
		ipadressx = ipadress;
		snprintf(port,sizeof(port),"PBMTAi%d",33);
		portx = port;

free(ipadressx);
free(portx);

```
free() line have a error 
abort() was called at PC 0x40082ce6 on core 1

how to free char * ? 
memory bloats if i don't use free()

chegewara, 2022-05-28:
The problem is not with free, the problem is with your code.
`ipadressx = ipadress;`
ipadressx is no longer pointer to memory allocated with malloc, it is now pointer to first element of `ipadress`, which cant be free, because it is local variable and will be released on function exit.
The same situation is with `portx = port;`.
What you can do is to remove `char ipadress[15];`  or malloc and free from your code.

cyberumut, 2022-05-29:
> The problem is not with free, the problem is with your code. `ipadressx = ipadress;` ipadressx is no longer pointer to memory allocated with malloc, it is now pointer to first element of `ipadress`, [...]

thank u problem  solved.

closed 2022-05-30 by cyberumut