espressif/esp-idf #9053: Undefined member 'ssize' in struct FATFS

markuspi, 2022-05-28:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    179ea878c747519138b40f387b80f30dd72097f0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]

## Problem Description

The recently added commit 22b46580f8a1ed63ae04574520882851611e062f changed the file `components/fatfs/vfs/vfs_fat.c` and added a reference to struct member `fs->ssize;`. This struct member is only conditionally defined in `components/fatfs/src/ff.h`:

```c
#if FF_MAX_SS != FF_MIN_SS
	WORD	ssize;			/* Sector size (512, 1024, 2048 or 4096) */
#endif
```

When `FF_MAX_SS == FF_MIN_SS`, this leads to a compilation error

### Expected Behavior

Compilation without error

### Actual Behavior

Compilation error

### Steps to reproduce

1. Manually install esp-idf as described in [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html)
2. Copy hello_world example project
3. Save this [sdkconfig](https://gist.github.com/markuspi/123fb1ea13ddfa44c63330639356122d) in project dir
4. `idf.py build`

### Code to reproduce this issue

default example project, but custom [sdkconfig](https://gist.github.com/markuspi/123fb1ea13ddfa44c63330639356122d)

note: this sdkconfig is based on https://github.com/feelfreelinux/cspot/tree/master/targets/esp32 . I just removed the custom partition table for this PoC. The issue happens regardless of the partition table.

## Debug Logs

```
/home/user/esp/esp-idf/components/fatfs/vfs/vfs_fat.c: In function 'esp_vfs_fat_info':
/home/user/esp/esp-idf/components/fatfs/vfs/vfs_fat.c:230:48: error: 'FATFS' {aka 'struct <anonymous>'} has no member named 'ssize'; did you mean 'csize'?
         *out_total_bytes = total_sectors * fs->ssize;
                                                ^~~~~
                                                csize

```

## Other items if possible

- [x] sdkconfig file: see above

igrr, 2022-05-29:
@markuspi Thank you for reporting this regression, we'll fix it soon!

Alvin1Zhang, 2022-07-11:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/0b0befc2a6451d83b529a0af103245796236d1d8, feel free to reopen.

closed 2022-07-11 by Alvin1Zhang