espressif/esp-idf #9055: Hard hang when establishing an HTTPS connection with use_global_ca_store enabled (ESP-IDF 4.3.2)

kotyara12, 2022-05-29:
Good day everyone!

How to properly use esp_http_client_config_t cfg.use_global_ca_store = true;

As far as I understand, global CA storage is designed to store several certificates, which can then be used by one or more https requests.

How I did:

1. When the application is launched on the MCU, the following code is executed (error handling is omitted here):
```
  extern const char isrg_root_x1_pem_start[]        asm("_binary_isrg_root_x1_pem_start");
  extern const char isrg_root_x1_pem_end[]          asm("_binary_isrg_root_x1_pem_end"); 
  extern const char digi_cert_pem_start[]         asm("_binary_digi_cert_pem_start");
  extern const char digi_cert_pem_end[]           asm("_binary_digi_cert_pem_end"); 
  extern const char api_telegram_org_pem_start[]  asm("_binary_api_telegram_org_pem_start");
  extern const char api_telegram_org_pem_end[]    asm("_binary_api_telegram_org_pem_end"); 

esp_tls_init_global_ca_store();
esp_tls_set_global_ca_store((const unsigned char*)isrg_root_x1_pem_start, isrg_root_x1_pem_end-isrg_root_x1_pem_start);
esp_tls_set_global_ca_store((const unsigned char*)digi_cert_pem_start, digi_cert_pem_end-digi_cert_pem_pem_start);
esp_tls_set_global_ca_store((const unsigned char*)isrg_root_x1_pem_start, isrg_root_x1_pem_end-isrg_root_x1_pem_start);
```

Then, when the connection is established, I set a flag (i do the same for HTTPS, MQTT and OTA):

```
esp_http_client_config_t cfgHttp;
...
cfgHttp.use_global_ca_store = true;
```

That is, I initialize the global CA store only once at startup and only once add THREE certificates to this store AT ONCE, which are used by the program for all secure connections throughout its "life"

-------------------

However, in the examples, they act differently: the store is initialized and ONE certificate is added immediately before connecting: [https://github.com/espressif/esp-idf/bl ... #L187-L201](https://github.com/espressif/esp-idf/blob/179ea878c747519138b40f387b80f30dd72097f0/examples/protocols/https_request/main/https_request_example_main.c#L187-L201)

I don't see the point in this at all! Why do this? In this case, it's easier to attach the certificate directly "as usual"

-------------------

I've run into the following issues. I have 10 devices running on ESP32, nine on ESP32-WROOM32x, one on ESP32-WROWER that has no shortage of RAM. On some of them (ESP32-WROOM32D or ESP32-WROOM32U), everything seems to work fine.

Problem number one. But on some, including WROWER, periodically (but not every time, with a variable interval - from 1 minute to several hours) HEAP_ALLOCATION_FAILED occurs and the device reboots (I deliberately turned on the reboot in case of a memory allocation error). At the same time, the logs show that some function requires more than 512 bytes of RAM. This happens, apparently, when establishing an HTTPS connection, namely when accessing the Telegram API.

Problem number two. Sometimes, when calling esp_http_client_perform() , the device began to "hang", and in such a way that all threads / tasks freeze, even those that do not touch https in any way

At the same time, there are no problems on another exactly the same device. The configuration is the same, compared line by line.

-------------------

In this regard, questions:

1. In general, can you add several certificates to the store at once? If not, what is the deeper meaning for its use????

2. Is use_global_ca_store safe for multi-threaded use? I strongly suspect that "hard" freezes occur due to the fact that two threads access the storage at once (for example, the send queue in telegram and mqtt)

There are almost no examples for use_global_ca_store in "these Internets of yours", and the help does not explain anything. How to use it correctly is not clear.

What do you advise?

-------------------

PS: I'm aware that there is also crt_bundle_attach, but I can't use it yet, because on version 4.3 it does not work for https, and 4.4 is not available to me yet (I use PlatformIO, and they are in no hurry). And I don't see any point in dragging a bunch of useless certificates into memory that I will never use. Three is enough for me.

kotyara12, 2022-05-31:
Addition.

Blocking shared access to the global CA store using a semaphore seems to have solved the problem, there are no freezes. Presumably this may be the reason. While I'm testing.

hmalpani, 2022-06-01:
@kotyara12,

>     1. In general, can you add several certificates to the store at once? If not, what is the deeper meaning for its use????

Yes, we can add multiple certificates to the store at once. The way you are doing it seems fine to me.

>     2. Is use_global_ca_store safe for multi-threaded use? I strongly suspect that "hard" freezes occur due to the fact that two threads access the storage at once (for example, the send queue in telegram and mqtt)

I don't think that it might be the reason. As of now I am unable to reproduce the issue. I am using [https_request](https://github.com/espressif/esp-idf/blob/179ea878c747519138b40f387b80f30dd72097f0/examples/protocols/https_request/main/https_request_example_main.c) example. I updated the code a little to have two tasks which are using global CA store for https request. I have also initialized global CA once in main().

Can you share your code or some code to reproduce this issue?