espressif/esp-idf #9056: E (167) flash HPM: Flash high performance mode hasn't been executed successfully

FanhuaCloud, 2022-05-29:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S3FN8
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-3115-g73be74fd7
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:    Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes,VSCode
- Power Supply:        USB

## Problem Description
```
E (167) flash HPM: Flash high performance mode hasn't been executed successfully
```
### Expected Behavior
The FLASH inside S3FN8 is detected as gd25, which should be able to open successfully
### Actual Behavior
failed.
### Steps to reproduce
1. set Flash SPI speed to 120Mhz
2. Build, Run
[image]

## Debug Logs

```
I (167) flash HPM: Enabling high speed mode for chip GD
E (167) flash HPM: Flash high performance mode hasn't been executed successfully
W (170) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
```

ginkgm, 2022-05-30:
Hi @FanhuaCloud ,
High performance mode is optional feature. See: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_optional_feature.html#high-performance-mode

Not all official module support that. As for GD25, due to its support, there are some technical difficulities to fix. We are not sure if it can be finally supported at this moment. If you do want to use 120MHz, please contact Espressif Business team to get stable supply for the types that supports high performance mode.

ginkgm, 2022-05-30:
Hi @FanhuaCloud ,
High performance mode is optional feature. See: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_optional_feature.html#high-performance-mode

Not all official module support that. As for GD25, due to its support, there are some technical difficulities to fix. We are not sure if it can be finally supported at this moment. If you do want to use 120MHz, please contact Espressif Business team (https://www.espressif.com.cn/en/contact-us/sales-questions) to get stable supply for the types that supports high performance mode.

FanhuaCloud, 2022-05-30:
> Hi @FanhuaCloud , High performance mode is optional feature. See: [...]

Thank you for your reply.

closed 2022-05-30 by FanhuaCloud