espressif/esp-idf #9057: httpd server stops working after a while

cyberumut, 2022-05-29:
IDF:4.3
ESP32-C3

error : 
` httpd_txrx: httpd_sock_err: error in recv : 104`

slows down after a while and is not reachable at all. 
i using : 
`  httpd_resp_set_hdr(req, "Connection", "close");

	httpd_resp_send(req, outbuf, HTTPD_RESP_USE_STRLEN);
httpd_resp_send(req, NULL, 0);
httpd_sess_trigger_close(req->handle, httpd_req_to_sockfd(req));
    return ESP_OK;
`
The web page is constantly making requests with ajax, but after 15 minutes it starts to slow down.
where is the problem ?

ESP-YJM, 2022-05-30:
Hi @cyberumut, could you help enable socket debug in `idf.py menuconfig->Components config->LWIP->Enable LWIP Debug->Enable socket debug`. I think it maybe caused by memory not enough, so you maybe need add some memory print.

cyberumut, 2022-05-30:
> 

`lwip_send(18, data=0x3ffd0de0, size=63, flags=0x0)
lwip_send(18) err=0 written=63
lwip_send(18, data=0x3f401b18, size=10, flags=0x0)
lwip_send(18) err=0 written=10
lwip_send(18, data=0x3f4909b0, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_send(18, data=0x3f401b10, size=5, flags=0x0)
lwip_send(18) err=0 written=5
lwip_send(18, data=0x3f490750, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_send(18, data=0x3f490750, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_send(18, data=0x3fffb330, size=676, flags=0x0)
lwip_send(18) err=0 written=676
lwip_send(18, data=0x3ffd0de0, size=61, flags=0x0)
lwip_send(18) err=0 written=61
lwip_send(18, data=0x3f401b18, size=10, flags=0x0)
lwip_send(18) err=0 written=10
lwip_send(18, data=0x3f4909b0, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_send(18, data=0x3f401b10, size=5, flags=0x0)
lwip_send(18) err=0 written=5
lwip_send(18, data=0x3f490750, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_send(18, data=0x3f490750, size=2, flags=0x0)
lwip_send(18) err=0 written=2
lwip_sendto(17, data=0x3fffb2d0, short_size=12, flags=0x0 to=127.0.0.1 port=32768
lwip_accept(14)...
lwip_accept(14) returning new sock=19 addr=192.168.1.63 port=64431
lwip_select(20, 0x3fffbac0, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=14 ready for reading
lwip_selscan: fd=16 ready for reading
lwip_selscan: fd=18 ready for reading
lwip_selscan: fd=19 ready for reading
lwip_select: nready=4
lwip_recvfrom(16, 0x3fffba80, 12, 0x0, ..)
lwip_recvfrom_udp_raw[UDP/RAW]: top sock->lastdata=0x0
lwip_recvfrom_udp_raw[UDP/RAW]: netconn_recv err=0, netbuf=0x3fff4900
lwip_recvfrom_udp_raw: buflen=12
lwip_recvfrom_udp_raw(16):  addr=127.0.0.1 port=51710 len=12
lwip_close(18)
lwip_recvfrom(19, 0x3ffd0de0, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=0, pbuf=0x3fff2db8
lwip_recv_tcp: buflen=406 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff2db8
lwip_recvfrom(19):  addr=192.168.1.63 port=64431 len=128
lwip_recvfrom(19, 0x3ffd0e48, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff2db8
lwip_recv_tcp: buflen=278 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff2db8
lwip_recvfrom(19):  addr=192.168.1.63 port=64431 len=128
lwip_recvfrom(19, 0x3ffd0ec8, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff2db8
lwip_recv_tcp: buflen=150 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff2db8
lwip_recvfrom(19):  addr=192.168.1.63 port=64431 len=128
lwip_recvfrom(19, 0x3ffd0f48, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff2db8
lwip_recv_tcp: buflen=22 recv_left=128 off=0
lwip_recv_tcp: deleting pbuf=0x3fff2db8
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=-7, pbuf=0x0
lwip_recvfrom(19):  addr=192.168.1.63 port=64431 len=22
I (127661) GET: MAX BINARY INPUT = 5

lwip_send(19, data=0x3ffd0de0, size=63, flags=0x0)
lwip_send(19) err=0 written=63
[... 112 lines cut ...]
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=0, pbuf=0x3fff5a18
lwip_recv_tcp: buflen=406 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff5a18
lwip_recvfrom(18):  addr=192.168.1.63 port=64432 len=128
lwip_recvfrom(18, 0x3ffd0e48, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff5a18
lwip_recv_tcp: buflen=278 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff5a18
lwip_recvfrom(18):  addr=192.168.1.63 port=64432 len=128
lwip_recvfrom(18, 0x3ffd0ec8, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff5a18
lwip_recv_tcp: buflen=150 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff5a18
lwip_recvfrom(18):  addr=192.168.1.63 port=64432 len=128
lwip_recvfrom(18, 0x3ffd0f48, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff5a18
lwip_recv_tcp: buflen=22 recv_left=128 off=0
lwip_recv_tcp: deleting pbuf=0x3fff5a18
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=-7, pbuf=0x0
lwip_recvfrom(18):  addr=192.168.1.63 port=64432 len=22`

server.c 6489 line :
httpd_resp_send(req, NULL, 0);

cyberumut, 2022-05-30:
i deleted "httpd_resp_send(req, NULL, 0);" 

lwip_recvfrom(19, 0x3ffd0de0, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=0, pbuf=0x3fff513c
lwip_recv_tcp: buflen=405 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff513c
lwip_recvfrom(19):  addr=192.168.1.63 port=65508 len=128
lwip_recvfrom(19, 0x3ffd0e49, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff513c
lwip_recv_tcp: buflen=277 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff513c
lwip_recvfrom(19):  addr=192.168.1.63 port=65508 len=128
lwip_recvfrom(19, 0x3ffd0ec9, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff513c
lwip_recv_tcp: buflen=149 recv_left=128 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fff513c
lwip_recvfrom(19):  addr=192.168.1.63 port=65508 len=128
lwip_recvfrom(19, 0x3ffd0f49, 128, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fff513c
lwip_recv_tcp: buflen=21 recv_left=128 off=0
lwip_recv_tcp: deleting pbuf=0x3fff513c
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=-7, pbuf=0x0
lwip_recvfrom(19):  addr=192.168.1.63 port=65508 len=21
lwip_send(19, data=0x3ffd0de0, size=63, flags=0x0)
lwip_send(19) err=0 written=63
lwip_send(19, data=0x3f401b18, size=10, flags=0x0)
lwip_send(19) err=0 written=10
lwip_send(19, data=0x3f4909b0, size=2, flags=0x0)
lwip_send(19) err=0 written=2
lwip_send(19, data=0x3f401b10, size=5, flags=0x0)
lwip_send(19) err=0 written=5
lwip_send(19, data=0x3f490750, size=2, flags=0x0)
lwip_send(19) err=0 written=2
lwip_send(19, data=0x3f490750, size=2, flags=0x0)
lwip_send(19) err=0 written=2
lwip_send(19, data=0x3fffb1b0, size=103, flags=0x0)
lwip_send(19) err=0 written=103
lwip_sendto(17, data=0x3fffb160, short_size=12, flags=0x0 to=127.0.0.1 port=32768
lwip_accept(14)...
lwip_accept(14) returning new sock=18 addr=192.168.1.63 port=65509
lwip_select(20, 0x3fffb420, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=14 ready for reading
lwip_selscan: fd=16 ready for reading
lwip_selscan: fd=18 ready for reading
lwipE (42531) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (42531) task_wdt:  - IDLE1 (CPU 1)
E (42531) task_wdt: Tasks currently running:
E (42531) task_wdt: CPU 0: IDLE0
E (42531) task_wdt: CPU 1: httpd
E (42531) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401789E3:0x3FFB0690 0x40082311:0x3FFB06B0 0x401EABD3:0x3FFD2180 0x40178FBE:0x3FFD21A0 0x4008CE81:0x3FFD21C0 0x4008E9B1:0x3FFD21E0
0x401789e3: task_wdt_isr at /home/rootx/esp/esp-idf/components/esp_common/src/task_wdt.c:187

0x40082311: _xt_lowint1 at /home/rootx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x401eabd3: esp_pm_impl_waiti at /home/rootx/esp/esp-idf/components/esp32/pm_esp32.c:484

0x40178fbe: esp_vApplicationIdleHook at /home/rootx/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x4008ce81: prvIdleTask at /home/rootx/esp/esp-idf/components/freertos/tasks.c:3386 (discriminator 1)

0x4008e9b1: vPortTaskWrapper at /home/rootx/esp/esp-idf/components/freertos/xtensa/port.c:169

E (42531) task_wdt: Print CPU 1 backtrace

Backtrace:0x40084109:0x3FFB0C90 0x40082311:0x3FFB0CB0 0x4010A0C2:0x3FFFAE20 0x4010AEC6:0x3FFFAE40 0x4010B642:0x3FFFAE60 0x4008A2AA:0x3FFFAE80 0x4008939D:0x3FFFAEA0 0x4008942C:0x3FFFAEC0 0x40089AD7:0x3FFFAEF0 0x400F2D35:0x3FFFAF20 0x400F89A6:0x3FFFAF50 0x400ED9BA:0x3FFFB260 0x401B2923:0x3FFFB2B0 0x401B4936:0x3FFFB310 0x4010BE9B:0x3FFFB3A0 0x40179518:0x3FFFB410 0x4017962E:0x3FFFB450 0x4008E9B1:0x3FFFB470
0x40084109: esp_crosscore_isr at /home/rootx/esp/esp-idf/components/esp32/crosscore_int.c:77

0x40082311: _xt_lowint1 at /home/rootx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4010a0c2: uart_tx_char at /home/rootx/esp/esp-idf/components/vfs/vfs_uart.c:160 (discriminator 1)

0x4010aec6: uart_write at /home/rootx/esp/esp-idf/components/vfs/vfs_uart.c:217

0x4010b642: esp_vfs_write at /home/rootx/esp/esp-idf/components/vfs/vfs.c:422 (discriminator 4)

0x4008a2aa: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/stdio.c:94

0x4008939d: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c:224

0x4008942c: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fflush.c:278

0x40089ad7: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251

0x400f2d35: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:403

0x400f89a6: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)

0x400ed9ba: printf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/printf.c:56 (discriminator 5)

0x401b2923: lwip_selscan at /home/rootx/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:1930

0x401b4936: lwip_select at /home/rootx/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:2055

0x4010be9b: esp_vfs_select at /home/rootx/esp/esp-idf/components/vfs/vfs.c:987 (discriminator 15)

0x40179518: httpd_server at /home/rootx/esp/esp-idf/components/esp_http_server/src/httpd_main.c:196

0x4017962e: httpd_thread at /home/rootx/esp/esp-idf/components/esp_http_server/src/httpd_main.c:249 (discriminator 15)

0x4008e9b1: vPortTaskWrapper at /home/rootx/esp/esp-idf/components/freertos/xtensa/port.c:169

_selscan: fd=19 ready for reading
lwip_select: nready=4
lwip_recvfrom(16, 0x3fffb3e0, 12, 0x0, ..)
lwip_recvfrom_udp_raw[UDP/RAW]: top sock->lastdata=0x0
lwip_recvfrom_udp_raw[UDP/RAW]: netconn_recv err=0, netbuf=0x3fff52b4
lwip_recvfrom_udp_raw: buflen=12
lwip_recvfrom_udp_raw(16):  addr=127.0.0.1 port=59232 len=12
lwip_close(19)

ESP-YJM, 2022-05-30:
For the watchdog, it maybe due to log print, you can disable the log print what i told you in https://github.com/espressif/esp-idf/issues/9057#issuecomment-1140746934.
And for err `104` means the `RST`, the peer send `RST` to close the connection. If possible, you can capture the packet in your PC client.

cyberumut, 2022-05-30:
> For the watchdog, it maybe due to log print, you can disable the log print what i told you in [#9057 (comment)](https://github.com/espressif/esp-idf/issues/9057#issuecomment-1140746934). And for err `104` means the `RST`, the peer send `RST` to close the connection. If possible, you can capture the packet in your PC client.

i changed cpu mhz 160 to 240 problem solved wtf :)=

closed 2022-05-30 by cyberumut

ESP-YJM, 2022-05-31:
@cyberumut Change CPU to 240M indeed improve the speed of handle HTTP request, that HTTP server task will not occupy too much time to make IDLE task detected watchdog timeout. Maybe errno `104` is that client wait too much time to receive the data and send RST to close the connection. Although increasing the CPU frequency can avoid the problem, I think it's just a workaround.

AxelLin, 2022-05-31:
> i changed cpu mhz 160 to 240 problem solved wtf :)=

I don't get it.
How do you set ESP32-C3 to 240 MHz?